libusb/hidapi #6: Continuous integration

z3ntu, 2019-06-05:
I'm working on CI scripts for:
* Windows (msbuild, cygwin) using Appveyor
* Windows (mingw) using Fedora on builds.sr.ht (couldn't get mingw to work properly on appveyor)
* macOS using Travis CI
* FreeBSD using builds.sr.ht (OpenBSD would also be possible there)
* Linux (probably Alpine Linux and Arch Linux) also using builds.sr.ht

This makes sure at least that hidapi builds for most/all supported platforms.

| Target | Duration |
|--------|----------|
|msbuild|20 sec|
|cygwin|1 min 45 sec|
|mingw|70 sec|
|freebsd|55 sec|
|macos|54 sec|
|alpine|todo|
|arch|todo|

Qbicz, 2019-06-05:
That's very interesting, please share the results when you have the CI setup ready.

Which of these platforms can be integrated with GitHub, e.g. to run when a pull-request is opened?
Also, would it be possible to download artifacts from them? (ideally shared libraries for Windows and Mac)

z3ntu, 2019-06-05:
You can see the results for all the platforms listed above at https://github.com/z3ntu/hidapi/commits/ci . All of them are fully integrated into GitHub ;)

Artifacts are definitely possible on AppVeyor, but I'm not sure if Travis supports artifacts natively. But honestly, it's really easy to compile hidapi on macOS, so I wouldn't actually bother with that. Windows, sure, it's a pain otherwise.

Qbicz, 2019-06-05:
Sounds good!

What about attaching the Windows DLL to the next release? Do you think it would be a good idea? (I can provide MSVC-built library). Then having the artifacts is not essential.

z3ntu, 2019-06-05:
I also have https://github.com/signal11/hidapi/pull/324 on my CI branch which is needed for the Windows build. 

I'll work on the Windows dll artifacts tomorrow!

Youw, 2019-06-05:
I'd suggest switching to CMake instead of having MSVC2015/2017/2019/etc. projects.
Personally I'm using CMake for all my projects on all platforms (Windows/Linux/macOS/cross-aarch64).

z3ntu, 2019-06-06:
Windows DLL artifacts now work: https://ci.appveyor.com/project/z3ntu/hidapi/builds/25095767/job/q9378cp84do5qs49/artifacts

Youw, 2019-06-06:
I'd say having only a DLL as an artifact is not enough.
Perfect case, if we have .dll, .lib and .pdb - that's very common for Windows.

z3ntu, 2019-06-06:
https://ci.appveyor.com/project/z3ntu/hidapi/build/job/f2h18kwiw3kqo2qx/artifacts Better?

Youw, 2019-06-06:
I don't believe `hidtest.exe` is a required artifact for this library.

z3ntu, 2019-06-06:
No it isn't but I think it doesn't hurt including that there.

jdk, 2019-06-06:
I’m heavily in favor of leaving it, otherwise we should include it as part of the release downloads for convenience.

Youw, 2019-06-06:
Yeah, actually, not having those in the release downloads - is my bigger concern.

Qbicz, 2019-06-13:
@z3ntu this is great work, thanks! I downloaded the Windows artifacts and tested. It looks like the hidapi.dll is built as 32-bit, despite of name of the build `arch=x64`. It cannot be loaded with 64-bit Python, but works with 32-bit Python.

The reason I think is that you build there a 64-bit Cygwin version and 32-bit MSVC version, but only the MSVC DLL is copied as artifact. This would be really great to have 64-bit artifact available as well.

Qbicz, 2019-06-13:
If this would be helpful, I have 64-bit target added to MSVC project here (apart from update to VIsual Studio 2017) https://github.com/Qbicz/hidapi-1/commit/a0a00ca76bbc6a2bad1b235f23c96fac6853c30b

Qbicz, 2019-06-30:
For me we can close this, as https://github.com/libusb/hidapi/pull/34 was successful.

One thing that could be added is 64-bit Windows artifacts, @z3ntu will you have time for this? Or should we create a separate task to be picked up later by someone?

z3ntu, 2019-06-30:
Create a new issue for that anyways, but I'll have a look at it tomorrow

closed 2019-06-30 by z3ntu