libusb/hidapi #11: Update README.txt/AUTHORS.txt

Youw, 2019-06-06:
- add notice about moving under @libusb/hidapi;
- update links if there are any;
- consider Markdown?

todbot, 2019-06-06:
+1 on Markdown

z3ntu, 2019-06-10:
Notes from #17 that should be fixed as a follow-up:

> * the text is now rather lengthy in comparison to original, which is more compressed. I feel the markdown version is overwhelming, what about having some table of contents in the beginning?
> * Title of library is broken down into 2 lines, with "X" from OS X in second line. What about using ## heading which would results in smaller letters?

Qbicz, 2019-07-23:
Fixed by #19

closed 2019-07-23 by Qbicz

Youw, 2019-07-23:
AUTHORS.txt hasn't been updated yet, aren't we going to?

Qbicz, 2019-08-02:
@Youw if you have idea how AUTHORS should now look like, please go ahead with PR ;)

Youw, 2019-08-02:
I'll think of something when I'm back from a family trip. (Somewhere next week.)