libusb/hidapi #24: How to build for Android

z3ntu, 2019-06-17:
```
$ ~/Android/Sdk/ndk-bundle/ndk-build
Android NDK: APP_PLATFORM not set. Defaulting to minimum supported version android-16.    
/mnt/hdd/Android/Sdk/ndk-bundle/build/core/build-binary.mk:672: Android NDK: Module hidapi depends on undefined modules: usb1.0    
/mnt/hdd/Android/Sdk/ndk-bundle/build/core/build-binary.mk:685: *** Android NDK: Aborting (set APP_ALLOW_MISSING_DEPS=true to allow missing dependencies)    .  Stop.
```
```
$ env APP_ALLOW_MISSING_DEPS=true ~/Android/Sdk/ndk-bundle/ndk-build
Android NDK: APP_PLATFORM not set. Defaulting to minimum supported version android-16.    
/mnt/hdd/Android/Sdk/ndk-bundle/build/core/build-binary.mk:672: Android NDK: Module hidapi depends on undefined modules: usb1.0    
/mnt/hdd/Android/Sdk/ndk-bundle/build/core/build-binary.mk:672: Android NDK: Module hidapi depends on undefined modules: usb1.0    
/mnt/hdd/Android/Sdk/ndk-bundle/build/core/build-binary.mk:672: Android NDK: Module hidapi depends on undefined modules: usb1.0    
/mnt/hdd/Android/Sdk/ndk-bundle/build/core/build-binary.mk:672: Android NDK: Module hidapi depends on undefined modules: usb1.0    
[arm64-v8a] Compile        : hidapi <= hid.c
/home/luca/Dev/hidapi/android/jni/../../libusb/hid.c:47:10: fatal error: 'libusb.h' file not found
#include <libusb.h>
         ^~~~~~~~~~
1 error generated.
make: *** [/home/luca/Dev/hidapi/android/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o] Error 1
```

With the following modifications to the Android.mk file, the build succeeds (using `~/Android/Sdk/ndk-bundle/ndk-build`):
```diff
diff --git a/android/jni/Android.mk b/android/jni/Android.mk
index 527b43f..32e3442 100644
--- a/android/jni/Android.mk
+++ b/android/jni/Android.mk
@@ -1,8 +1,16 @@
-LOCAL_PATH:= $(call my-dir)
+MY_LOCAL_PATH:= $(call my-dir)
+
+LOCAL_PATH := $(MY_LOCAL_PATH)
 
 HIDAPI_ROOT_REL:= ../..
 HIDAPI_ROOT_ABS:= $(LOCAL_PATH)/../..
 
+include /tmp/libusb/android/jni/libusb.mk
+
+# hidapi
+
+LOCAL_PATH := $(MY_LOCAL_PATH)
+
 include $(CLEAR_VARS)
 
 LOCAL_SRC_FILES := \

```
```
[armeabi-v7a] Install        : libhidapi.so => libs/armeabi-v7a/libhidapi.so
```

See also https://developer.android.com/ndk/guides/android_mk#my-dir

Youw, 2019-06-18:
> `include /tmp/libusb/android/jni/libusb.mk`

I'm not familiar with `.mk` and Android build system, but I'm almost sure, that there has to be a better way specifying dependency, instead of hard-coding it.

cnlohr, 2019-08-23:
I am struggling with the same issue and a complete dirth of any information surrounding linking jni's to jni's on the internet at all.  Nothing I do seems to let hidapi find libusb1.0.  I perpetually see Android NDK: Module hidapi depends on undefined modules: usb1.0  .  If anyone can shed light on the "proper" way to build the Android NDK module, please let me know!

TheBricktop, 2019-09-14:
thanks for posting this, it should be added to master.

Youw, 2019-09-14:
Again, hard-coding full system-specific paths is definitely not something we should have in master.
I'll have Android project to play soon, and will try to find elegant solution.

Ghostbird, 2021-11-24:
Because I didn't really understand this fully at first, I worked it out in a script:
https://gist.github.com/Ghostbird/566fbf92000b615934c0af8c6069ec94

The point that I entirely missed was that I had to clone libusb/libusb too and point to its `Android.mk` file.
Note: The script does everything it also clones _this_ repository, which is more than you need if you already have this cloned.

Youw, 2023-03-12:
At this point I think it should be enough to say that CMake build works perfectly fine on Android (I'm using it in my personal projects).

For that two things needed:
1) General CMake+Android [instruction](https://developer.android.com/ndk/guides/cmake);
2) libusb for Android [build](https://github.com/libusb/libusb/blob/master/android/README).

As for `Android.mk` present in the repo - time to deprecated it.
Those who really need it - would know how to use it.
All others should probably use CMake build. Gradle (as native build tool for Android) supports CMake too.

closed 2023-03-12 by Youw