libusb/hidapi #32: Enforce coding style

z3ntu, 2019-06-20:
I've already seen some places where it's "tab 4x space tab", so some automated code formatter would be good. 

I'm using clang-format for my own projects, which is very configurable; but something like astyle is also not bad :)

z3ntu, 2019-06-20:
I've come up with the following `.clang-format` file that keeps changes *relatively* small:

```
---
AlignAfterOpenBracket: DontAlign
BraceWrapping:
#  AfterExternBlock: true
  AfterFunction: true
BreakBeforeBraces: Custom
ColumnLimit: 0
# BeforeHash is in clang 9 which should be used
IndentPPDirectives: AfterHash
IndentWidth: 4
SpaceAfterCStyleCast: true
TabWidth: 4
UseTab: Always
...
```
`AfterExternBlock` would indent after the `extern "C"` blocks which would match with the code in `hidapi.h` but not the code in the implementations, so I would disable that wrapping. (Also, if we don't want AfterExternBlock, we can set `BreakBeforeBraces` to some preset that matches our style instead of defining it manually)

For `IndentPPDirectives` (indent preprocessor directives), there's `BeforeHash` but only in clang 9 (which should be released about September 2019) which matches the style in hidapi.

For `AlignAfterOpenBracket` it's also a mixed case, we can really consider changing that (so setting it to `Align` = leaving the directive out)

I also didn't find a clang-format option to do this (so making a preprecessor block indent the code in it):
```
int myint;

#if MYDEFINE
    int myint2;
#endif
```

But still with that, a bunch of formatting (especially comments it seems) has to be done manually.

Youw, 2023-03-12:
I think this is something we can consider/apply for v1.0