libusb/hidapi #35: Add CI build badges

Qbicz, 2019-06-27:
Blocked by https://github.com/libusb/hidapi/pull/34

Badges look like this: https://builds.sr.ht/~qbicz/hidapi-1/archlinux.yml

z3ntu, 2019-07-13:
We can use

[image]
```
[![builds.sr.ht status](https://builds.sr.ht/~qbicz/hidapi.svg)](https://builds.sr.ht/~qbicz/hidapi?)
```
now. This should display a "combined view" of the different builds for hidapi (so if a build for e.g. FreeBSD fails, this badge should change). I'm unsure how (failing) pull requests will affect this badge though.

Youw, 2019-07-13:
it seems that is shows the status of the last build
branch-specific badges are not available yet (at least as per this discussion): https://todo.sr.ht/~sircmpwn/builds.sr.ht/137

it worth to try, but if I need to guess - I'd say, that failed PR will make a failed badge

Youw, 2019-07-13:
one more note:

Travis supports per-branch badges, like this:
[image]
`[[image]](https://travis-ci.org/libusb/hidapi)`

Qbicz, 2019-07-14:
This will be perfect to use last master build, I'm also fine with having a separate badge for each platform (I have impression the project look more stable this way). I guess we can integrate the combined badge and change it later if we see it is problematic.

Qbicz, 2019-07-15:
We can also add AppVeyor badge when https://github.com/libusb/hidapi/pull/37 is available.

PR #46 (2019-07-25): README: add CI build badges
Show status of test builds on the project GitHub page.
Fixes #35.
commit 651150d8 (2019-07-25): README: add CI build badges

Show status of test builds on the project GitHub page.
Fixes #35.
closed 2019-07-25 by Qbicz