libusb/hidapi #47: AppVeyor project under libusb organisation

Youw, 2019-07-17:
AppVeyor the build configuration is working, but currently working under @Youw personal configuration: https://ci.appveyor.com/project/Youw/hidapi

@LudovicRousseau, can you create a project as libusb organisation project?
To do so:

Sign in as GitHub Organisation to https://ci.appveyor.com
Projects -> New Project -> GitHub -> libusb/hidapi
All projects settings are fine to be default except one:
[image]

---
Original message: https://github.com/libusb/hidapi/pull/37#issuecomment-512439210
As discussed - to be managed in August.

LudovicRousseau, 2019-07-29:
I can't sign in to appveyor as the libusb organisation. I can only sign in as "LudovicRousseau".

Youw, 2019-07-29:
Hm, what about creating a project on behalf of libusb?

Well, we managed to do it for Travis. I can't say for sure (as I never tried it myself), but would be great to make the same for AppVeyor too.

Youw, 2019-07-30:
Just played with a test organisation/repo, searched some documentation, and here what I've found:
https://www.appveyor.com/docs/team-setup/#setting-up-appveyor-account-for-github-organization

Basically we need to create a new Appveyor account and name it as "libusb". I could do it myself, but I think better if we keep ownership to such accounts to someone who is libusb admin.

LudovicRousseau, 2019-07-30:
I don't plan to be a single point of failure for the hidapi project.
I will not create a "libusb" AppVeyor account just for hidapi.

Youw, 2019-07-30:
OK, then I see 2 options here:
1) Leave it as is, and have my own AppVeyor account for hidapi;
2) Create that account, share access to few persons among the organisation (not to have a single point of failure), and use it for libusb project too, since [libusb uses AppVeyor](https://ci.appveyor.com/project/dickens/libusb) too, but under _dickens_ account;

@dickens do you have an opinion on this?

z3ntu, 2019-07-30:
Imo even if the owner of whatever CI account disappears, it's really easy to use any other account to run the appveyor/builds.sr.ht builds with, as the configuration is stored in the repo.

tresf, 2020-11-20:
> I don't plan to be a single point of failure for the hidapi project.
> I will not create a "libusb" AppVeyor account just for hidapi.

It wouldn't be a single point of failure once the administrators and team security were setup.  It's a bit convoluted since it requires the initial one-off registration, then requires a new, dedicated security group, but it seems like they've designed it with multiple administrators in mind and the "single point of failure" concern is mitigated once configured.

LudovicRousseau, 2021-03-17:
I tried to create a new appveyor account (libusbadmin) not linked to my github account but I never receive the activation email from appveyor. And I tried different valid emails.

@Youw (or someone else) maybe you can create the appveyor admin account and add my github account as co-administrator.

Youw, 2021-03-17:
I've created a new email address and a new Appveyor account.
Next steps:
1) to add addmin/user accounts to that one;
2) link libusb Gihub organisation to Appveyor.

Right now I'm working to solve some issues regarding 1) on Appveyor side with Appveyor support.

Youw, 2021-03-17:
Everything seems to be fine. New organisation account is created. At this point 2 persons have admin access: me and Ludovic.

PR #253 (2021-03-18): Switch to libusb/hidapi appveyor project for badge
Closes: #47
closed 2021-03-18 by Youw