libusb/hidapi #49: Non-free files?

z3ntu, 2019-07-19:
Debian had code to remove `windows/ddk_build/makefile` and `testgui/mac_support.cpp` from the tarball in their `debian/rules` file

`windows/ddk_build/makefile`:
```
 Copyright (C) Microsoft Corporation 1995, 1996
       All Rights Reserved.
```

`testgui/mac_support.cpp`:
```
Some of this code is from Apple Documentation, most notably
 http://developer.apple.com/legacy/mac/library/documentation/AppleScript/Conceptual/AppleEvents/AppleEvents.pdf 
```
and the pdf (archived at http://web.archive.org/web/20141226100754/http://developer.apple.com/legacy/library/documentation/AppleScript/Conceptual/AppleEvents/AppleEvents.pdf) has a footer `Copyright © 2005, 2007 Apple Inc. All Rights Reserved.`

z3ntu, 2019-07-19:
I've tried removing `testgui/mac_support.cpp` and building the test gui on macOS and it still works fine. So I don't think this file is actually used. After looking around a bit, it seems that this file is unused since https://github.com/libusb/hidapi/commit/bab22b7ac1f2d97290209910fcb6f2cea68ea972
I'll send a PR.

Qbicz, 2019-07-23:
Good catch, thank you!

Qbicz, 2019-08-30:
@z3ntu did we merge it already? Is it https://github.com/libusb/hidapi/commit/a1d49824ea6f8d5fcc90734b11bd934067eb730a? If yes, we can close the issue.

z3ntu, 2019-08-30:
The `windows/ddk_build/makefile` part is still relevant I think

mcuee, 2021-01-26:
Actualy the whole ddk_build directory can be removed now. MinGW and Visual Studio are enough.

FYI: libusb has already removed the ddk_build legacy files.

mcuee, 2021-06-13:
With Windows 7 out of support by Microsoft, I think it does not make sense to keep the ddk build files which is only valid for Windows 7 DDK. New Windows WDK are totally different (based on msbuild).

mcuee, 2021-08-29:
With CMake added, I think we should remove this legacy Windows ddk build scripts.

mcuee, 2021-09-13:
It does not make sense to keep the Windows DDK build scripts now. In fact, it does not work any way.

```
c:\WinDDK\7600.16385.1\bin>setenv.bat c:\WinDDK\7600.16385.1 free x64 WIN7
WARNING: x64 Native compiling isn't supported. Using cross compilers.
Launching OACR monitor

c:\WinDDK\7600.16385.1>cd c:\work\hid\hidapi\windows\ddk_build

c:\work\hid\hidapi\windows\ddk_build>build
BUILD: Compile and Link for AMD64
BUILD: Loading c:\winddk\7600.16385.1\build.dat...
BUILD: Computing Include file dependencies:
BUILD: Start time: Mon Sep 13 12:36:02 2021
BUILD: Examining c:\work\hid\hidapi\windows\ddk_build directory for files to compile.
    c:\work\hid\hidapi\windows\ddk_build - 1 source files (1,265 lines)
BUILD: Saving c:\winddk\7600.16385.1\build.dat...
BUILD: Compiling c:\work\hid\hidapi\windows\ddk_build directory
Configuring OACR for 'root:amd64fre' - <OACR on>
Compiling - hid.c
1>errors in directory c:\work\hid\hidapi\windows\ddk_build
1>c:\work\hid\hidapi\windows\hid.c(84) : error C2059: syntax error : '.'
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2143: syntax error : missing ';' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2143: syntax error : missing ';' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2143: syntax error : missing ')' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2143: syntax error : missing ';' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2065: 'compatible_id' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2100: illegal indirection
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2065: 'compatible_id' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2065: 'compatible_id' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2059: syntax error : ')'
1>c:\work\hid\hidapi\windows\hid.c(472) : error C2143: syntax error : missing ';' before '{'
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2143: syntax error : missing ';' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2143: syntax error : missing ';' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2143: syntax error : missing ')' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2143: syntax error : missing ';' before 'type'
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2065: 'p' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2100: illegal indirection
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2059: syntax error : ')'
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2065: 'p' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2065: 'p' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2065: 'p' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2100: illegal indirection
1>c:\work\hid\hidapi\windows\hid.c(474) : error C2106: '=' : left operand must be l-value
1>c:\work\hid\hidapi\windows\hid.c(477) : error C2065: 'compatible_id' : undeclared identifier
1>c:\work\hid\hidapi\windows\hid.c(482) : error C2043: illegal break
Building Library - .obj\work\hid\hidapi\windows\ddk_build\objfre_win7_amd64\amd64\hidapi.lib
1>link : error LNK1181: cannot open input file '.obj\work\hid\hidapi\windows\ddk_build\objfre_win7_amd64\amd64\hid.obj'
BUILD: Compile errors: not linking c:\work\hid\hidapi\windows\ddk_build directory
BUILD: Finish time: Mon Sep 13 12:36:03 2021
BUILD: Done

    3 files compiled - 12 Warnings - 25 Errors
    1 library built - 1 Error
```

commit 9e66485e (2021-09-18): Remove DDK support files.

Deleted DDK build scripts, support files and documentation for it,
as it wasn't working of-the-box with recent tool-sets from MS anyway.

Still keeping `HIDAPI_USE_DDK` macro in the code,
as it still _can_ be used if the build configuration is setup properly.

Resolves: #49
PR #333 (2021-09-22): Remove DDK support files.
Deleted DDK build scripts, support files and documentation for it,
as it wasn't working of-the-box with recent tool-sets from MS anyway.

Still keeping `HIDAPI_USE_DDK` macro in the code,
as it still _can_ be used if the build configuration is setup properly.

Resolves: #49
commit d4888c5f (2021-09-22): Remove DDK support files. (#333)

Deleted DDK build scripts, support files and documentation for it,
as it wasn't working of-the-box with recent tool-sets from MS anyway.

Still keeping `HIDAPI_USE_DDK` macro in the code,
as it still _can_ be used if the build configuration is setup properly.

Resolves: #49
closed 2021-09-22 by Youw