libusb/hidapi #56: NodeJs library

Gayathri-Rajamanickam-MindTree, 2019-07-25:
Hi Team,

We want to use this library as node module. Having issues in using node-hid library. Can you please let us know if this library has node library?

Regards,
Gayathri

todbot, 2019-07-25:
What is the issue you are having with `node-hid`?  (I am one of the maintainers)

Gayathri-Rajamanickam-MindTree, 2019-07-25:
We are using write method - device.write(data) from node-hid to write into hid device using windows 10. 

We are appending first byte as '4' as per this fix reported here - https://github.com/node-hid/node-hid/issues/187

Following is the example data buffer that we send to the device

4,103,73,72,48,115,67,105,65,103,73,109,108,107,73,106,111,103,73,106,65,121,79,68,89,53,78,71,81,119,76,87,69,53,78,68,89,116,77,84,70,108,79,83,49,104,78,122,82,104,76,84,100,107,89,50,73,119,89,106,81,121,78,122,73

For single write, it works fine. 

But when we write in a loop for about 100/500 times without any delay, it hangs in the write method.

todbot, 2019-07-25:
How fast are you attempting to write data?  How big is the HID report you are attempting to send and what is the size of your device's max packet size?

USB HID can only support one report every 1 millisecond.  At best, and that depends on how many hubs are between the OS and the device and the HID driver on the OS.   In practice it's usually much less.

It would be interesting to see if you could write a native C version of your loop (using `hidapi`, which `node-hid` is just a thin wrapper on) and see if you get the same hangs.

Gayathri-Rajamanickam-MindTree, 2019-07-25:
We have tested hidapi with native C. we are able to reproduce this issue. so the issue seems to be with hidapi. For workaround, we tried to introduce a delay of 20 milliseconds between writes in native C. In that case, it doesn't hang. 

Can you please let us know how to introduce this kind of delay in write operation when using node-hid module?

todbot, 2019-07-25:
That's more of a general Node.js question and depends on how your code is structured.  But I would approach it using [Node Timers](https://nodejs.org/api/timers.html):
```
const reportsToSend = [[]];  // array of byte arrays
const sentReportCount = 0;
const sendRateMillis = 100;  // send report once every 100 milliseconds
const sendTimer = setInterval( function() {
  const nextReport = reportsToSend[currentReport];
  hidDevice.write(nextReport);
  sentReportCount++;
  if( sentReportCount == reportsToSend.length ) {
    clearInterval(sendTimer);
  }
}, sendRateMillis);

closed 2020-06-20 by mcuee