libusb/hidapi #57: Building for android Unity

Albond87, 2019-07-25:
Anyone know how I can use this for an android project, built from Unity? Trying to get [this plugin](https://github.com/Looking-Glass/JoyconLib) to work on android, which it seemingly doesn't at the moment and I think it might be because it uses HIDAPI which isn't compatible with android? Can I use android.mk instead?

Youw, 2019-07-25:
You just use [android/jni/Android.mk](https://github.com/libusb/hidapi/blob/master/android/jni/Android.mk) to build project for Android.

Note, that you also nee libusb1.0 (build or prebuilt).

Albond87, 2019-07-25:
@Youw Thanks for the reply, but could you tell me exactly how to do this in Unity? Sorry, I'm not very experienced with this kind of thing. And which exact things I need to download. I appreciate the help

Youw, 2019-07-25:
Unfortunately I cannot.

This project is about using HID devices, and has nothing to do with Unity or Android development in general.

stackoverflow/reddit/etc. would be a better place to ask this kind of questions.

Albond87, 2019-07-26:
I have posted in some other places but no responses. I guess I just have to keep trying

tuxun, 2019-07-28:
you have to compile libusb for your target cpu type and to link the binary lib in your linker settings. That all I know, i did it in C++, but that didnt solved the permission problem I was facing, now I have an android service made in java which handle the permission.

TheBricktop, 2019-09-14:
> 
> 
> now I have an android service made in java which handle the permission.

Care to share how did You did that?

Chakib-Temal, 2019-11-14:
> you have to compile libusb for your target cpu type and to link the binary lib in your linker settings. That all I know, i did it in C++, but that didnt solved the permission problem I was facing, now I have an android service made in java which handle the permission.

could you share more informations ?

closed 2021-06-10 by mcuee