libusb/hidapi #58: Why did the call to hid_open () fail?

emily-5728, 2019-07-27:
As USB host, am5728 evm board is connected with dlp module through USB HID V1.11.
The test function uses hidtest. in hidapi-master open source Running the test function on the ti am5728 evm board
failed to call hid_open (). Why did the call to hid_open () fail? Do you need to write dlp driver on 5728 single board?
Usb1.0 and udev libraries are currently used. Are there any restrictions on the version of the library? Thank you!
root@am57xx-evm:# ./hid_test
no device
unable to open device
Segmentation fault
root@am57xx-evm:#

The same test function can also correctly read dlp information in ubuntu, as follows:
[root@tl hid_test]# ./hid_test
Device Found
type: 0e0f 0003
path: /dev/hidraw0
serial_number:
Manufacturer: VMware
Product: VMware Virtual USB Mouse
Release: 103
Interface: 0

Device Found
type: 0451 6401
path: /dev/hidraw1
serial_number: LCR2
Manufacturer: Texas Instruments Inc.
Product: DLPC350
Release: 0
Interface: 0

Device Found
type: 0451 6401
path: /dev/hidraw2
serial_number: LCR2
Manufacturer: Texas Instruments Inc.
Product: DLPC350
Release: 0
Interface: 1

hid open dlp ok!
Manufacturer String: Texas Instruments Inc.
Product String: DLPC350
Serial Number String: (76) LCR2
Unable to read indexed string 1
Indexed String 1:
Feature Report

waiting...
waiting...
waiting...

When the dlp module is connected to the am5728 through the USB interface, the following information is printed:
root@am57xx-evm:/lib# [ 4071.898196] usb 1-1.2: new full-speed USB device number 9 using xhci-hcd
[ 4072.005095] usb 1-1.2: New USB device found, idVendor=0451, idProduct=6401
[ 4072.012029] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4072.019560] usb 1-1.2: Product: DLPC350
[ 4072.023426] usb 1-1.2: Manufacturer: Texas Instruments Inc.
[ 4072.029165] usb 1-1.2: SerialNumber: LCR2
[ 4072.067017] hid-generic 0003:0451:6401.000D: hiddev0: USB HID v1.11 Device [Texas Instruments Inc. DLPC350] on usb-xhci-hcd.1.auto-1.2/input0
[ 4072.087137] input: Texas Instruments Inc. DLPC350 as /devices/platform/44000000.ocp/48880000.omap_dwc3_1/48890000.usb/xhci-hcd.1.auto/usb1/1-1/1-1.2/1-1.2:1.1/0003:0451:6401.000E/input/input9
[ 4072.158539] hid-generic 0003:0451:6401.000E: input: USB HID v1.11 Keyboard [Texas Instruments Inc. DLPC350] on usb-xhci-hcd.1.auto-1.2/input1

root@am57xx-evm:~# cat /proc/bus/input/devices
I: Bus=0019 Vendor=0001 Product=0001 Version=0100
N: Name="gpio_keys"
P: Phys=gpio-keys/input0
S: Sysfs=/devices/platform/gpio_keys/input/input1
U: Uniq=
H: Handlers=kbd event0
B: PROP=0
B: EV=100003
B: KEY=16c0 0 0 0

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="palmas_pwron"
P: Phys=palmas_pwron/input0
S: Sysfs=/devices/platform/44000000.ocp/48070000.i2c/i2c-0/0-0058/48070000.i2c:tps659038@58:tps659038_pwr_button/input/input0
U: Uniq=
H: Handlers=kbd event1
B: PROP=0
B: EV=3
B: KEY=100000 0 0 0

I: Bus=0003 Vendor=0451 Product=6401 Version=0111
N: Name="Texas Instruments Inc. DLPC350"
P: Phys=usb-xhci-hcd.1.auto-1.2/input1
S: Sysfs=/devices/platform/44000000.ocp/48880000.omap_dwc3_1/48890000.usb/xhci-hcd.1.auto/usb1/1-1/1-1.2/1-1.2:1.1/0003:0451:6401.0008/input/input6
U: Uniq=LCR2
H: Handlers=sysrq kbd mouse0 event2
B: PROP=0
B: EV=120017
B: KEY=1f0000 0 0 800000 0 e0b0ffdf 1cfffff ffffffff fffffffe
B: REL=3
B: MSC=10
B: LED=1f

There is a usbhid module on am5728 single board, but see hid code, there is no dlp drive, do you need to add dlp driver?
root@am57xx-evm:#
root@am57xx-evm:# lsmod
Module Size Used by
ipv6 317996 26
hid_generic 941 0
usbhid 27328 0
gdbserverproxy 4477 0
cryptodev 36191 1
cmemk 29155 0
xhci_plat_hcd 4469 0
xhci_hcd 102093 1 xhci_plat_hcd
usbcore 191588 3 xhci_plat_hcd,usbhid,xhci_hcd

Add the 99-hid rule file to/ etc/ udev/ rules. d, which is modified to:

HIDAPI/hidraw
KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="003f", MODE="0666"

There are libraries of usb1.0 and udev on the am5728 veneer, as follows:
root@am57xx-evm:/lib# ls -all

lrwxrwxrwx 1 root root 17 Dec 29 2015 libudev.so.0 -> libudev.so.0.13.1
-rwxr-xr-x 1 root root 45928 Dec 29 2015 libudev.so.0.13.1
lrwxrwxrwx 1 root root 16 Oct 8 00:00 libudev.so.1 -> libudev.so.1.6.4
-rwxr-xr-x 1 root root 108816 Oct 7 23:27 libudev.so.1.6.4
lrwxrwxrwx 1 root root 19 Dec 29 2015 libusb-0.1.so.4 -> libusb-0.1.so.4.4.4
-rwxr-xr-x 1 root root 14628 Dec 29 2015 libusb-0.1.so.4.4.4
lrwxrwxrwx 1 root root 19 Dec 29 2015 libusb-1.0.so.0 -> libusb-1.0.so.0.1.0
-rwxr-xr-x 1 root root 81652 Dec 29 2015 libusb-1.0.so.0.1.0

drwxr-xr-x 5 root root 4096 Dec 29 2015 udev
root@am57xx-evm:/lib#

z3ntu, 2019-07-30:
The `hid_test` program shipped with hidapi tries to open a hard-coded USB device, did you change that or are you using the defaults? 

Related: #14

emily-5728, 2019-07-31:
Hi z3ntu，
Thanks for your reply...
The previous problem mentioned that when the dlp4500 module is connected to the am5728 single board through the USB port, the am5728 single board can not recognize the dlp module because the application program of the am5728 single board fails to call the hid_open () function.

Then analyzing the HIDAPI code, the hid_open () function is implemented by calling the hid_enumerate () function. hid_enumerate () needs to open a device of type hidraw, and when the dlp module is connected to am5728 through the USB port, am5728 recognizes the dlp module as a hiddev0, which leads to the failure of the program execution. However, the dlp module is inserted into the Ubuntu, and the dlp module can be correctly identified as the hidraw device on the Ubuntu, and the correct connection can be made.

 

What changes need to be made on the am5728 evm board in order to identify the dlp module as a hidraw device?

 

The dlp4500 module is connected to the am5728 evm veneer through the USB port, and the test process is as follows:

root@am57xx-evm:~# lsusb

Bus 001 Device 004: ID 0451:6401 Texas Instruments, Inc.  ----- can recognize the dlp module

Bus 002 Device 002: ID 0451:8140 Texas Instruments, Inc.

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

Bus 001 Device 002: ID 0451:8142 Texas Instruments, Inc.

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

root@am57xx-evm:~#

root@am57xx-evm:~#

root@am57xx-evm:~# ls /dev | grep usb            

usb

root@am57xx-evm:~# ls /dev | grep hid

root@am57xx-evm:~# ls /dev/usb   ----- The dlp module can be recognized, but it is identified as a hiddev0 device

hiddev0

root@am57xx-evm:~#

root@am57xx-evm:~#

root@am57xx-evm:~#

root@am57xx-evm:~# cat /proc/bus/input/devices

I: Bus=0003 Vendor=0451 Product=6401 Version=0111

N: Name="Texas Instruments Inc. DLPC350"

P: Phys=usb-xhci-hcd.1.auto-1.2/input1

S: Sysfs=/devices/platform/44000000.ocp/48880000.omap_dwc3_1/48890000.usb/xhci-hcd.1.auto/usb1/1-1/1-1.2/1-1.2:1.1/0003:0451:6401.0004/input/input4

U: Uniq=LCR2

H: Handlers=sysrq kbd mouse1 event3

B: PROP=0

B: EV=120017

B: KEY=1f0000 0 0 800000 0 e0b0ffdf 1cfffff ffffffff fffffffe

B: REL=3

B: MSC=10

[... 516 lines cut ...]
Serial Number String: (76) LCR2

Unable to read indexed string 1

Indexed String 1:

Feature Report

[root@tl app_test]# ./hiddev ----- Write test function, open hiddev0 device, can open dlp device.

 

/dev/usb/hiddev0 infor

HIDIOCGVERSION: 1.4

HIDIOCGDEVINFO: bustype=3 busnum=2 devnum=4 ifnum=0

    vendor=0x0451 product=0x6401 version=0x0000

    num_applications=1

HIDIOCGNAME: Texas Instruments Inc. DLPC350

Reports of type Input (1):

todbot, 2019-07-31:
It looks like your USB host’s Linux kernel doesn’t have hidraw support enabled. Try compiling hidapi with Linux libusb instead of hidraw

emily-5728, 2019-07-31:
Hi todbot，
Thanks for your reply, i'll try it.

emily-5728, 2019-08-12:
Hi todbot，
Thanks for your reply, this problem has been resolved。

closed 2019-08-12 by emily-5728

todbot, 2019-08-12:
Hi @emily-5728, what was the solution?

emily-5728, 2019-08-13:
Compiling hidapi with Linux libusb instead of hidraw,arm5728 can correctly identify dlp modules.

jyzhong, 2021-05-13:
> As USB host, am5728 evm board is connected with dlp module through USB HID V1.11.
> The test function uses hidtest. in hidapi-master open source Running the test function on the ti am5728 evm board [...]

Hi, I have the same issue. I get lots of "waiting...".  Read your solution "Compiling hidapi with Linux libusb instead of hidraw". I run the executable file "./hidtest-libusb", I get the same result. Hope get your advice, thanks!

emily-5728, 2021-06-21:
你好：
    不知道你问题解决没有？
    很抱歉，这么久才看到你的邮件。
    dlp的sdk代码存在问题，不能使用hidraw模式，需要使用libusb模式。
    你只需要修改lcr4500.pro文件，将hidapi库的依赖部分去掉，加上libusb的部分。
    编译过程中如果报错，进行相应的修改。

S: ***@***.***:tps659038_pwr_button/input/input0