libusb/hidapi #61: Add Windows quirks to documentation

Hoernchen, 2019-08-03:
Using hidapi on windows is constrained by multiple issues, i.e. the inability to gain exclusive access to many devices (see https://github.com/signal11/hidapi/issues/231) and no access to mouse & keyboard devices (https://github.com/signal11/hidapi/issues/313) . The latter is important since it affects fancy keyboards that offer lighting effects, and the most obvious approach would be to use hidapi, while this actually requires dealing with zadig and libusb.

Things like this should be added to the docs, so Windows users don't need to rediscover these issues all the time.

CalcProgrammer1, 2020-06-03:
I have a modified version of the libusb-backend hidapi.c that compiles on Windows.  It is able to control RGB lighting devices in keyboards and mice if the appropriate interfaces have the WinUSB driver with zadig.  It's basically just a hack to remove the linux-specific stuff in the libusb backend and could use some cleanup, but this method does work if you wish to use the hidapi library in your code and still access these devices in Windows.

https://gitlab.com/CalcProgrammer1/OpenRGB/-/blob/master/dependencies/hidapi/hidapi.c

My OpenRGB project uses the Linux and MacOS native versions of hidapi just fine, but the Windows version does not work for whatever reason.  It detects some devices but can't control them.  If I use my hacked up libusb hidapi.c in place of the released library, it works just fine.

mcuee, 2021-06-13:
Actually I think it may not be a bad idea to have a libusb based Windows backend (using Zadig and WinUSB driver) to take away some of the limitations of the Windows HID driver. Some users may want to use that way.

Youw, 2021-06-14:
Agree, that would be a good option.

The usage of POSIX API would need to be replaced for libusb backend implementation of hidapi (preferably by C11 imho).