libusb/hidapi #62: cython-hidapi

prusnak, 2019-08-10:
Hi!

I am the current maintainer of [cython-hidapi](https://github.com/trezor/cython-hidapi) - a python binding for hidapi. I see you've resurrected the hidapi project and I'd like to ask you whether you would like to take part in maintainership of cython-hidapi package as well (i.e. move the package under your libusb organization where we'd maintain the package together).

prusnak, 2019-08-10:
Maybe we can even move the bindings into the main libusb repo, because the cython-hidapi uses hidapi repo as its submodule, so it makes a lot of sense.

todbot, 2019-08-10:
Hi @prusnak!

I am a fan of keeping the repositories separate, simply because it makes it easier to manage, with only one language's quirks and build process needing to be dealt with. Especially here because of the many languages wrap `hidapi`. 

There's also a discussion as to which Python `hidapi` library to support. I like how `cython-hidapi` builds for all OSes and includes pre-built binaries of its library with a statically-built `hidapi` on PyPi.  I feel this preferable than relying on a "system" install of `hidapi` (I think what `pyhidapi` does)

Youw, 2019-08-10:
Agree with @todbot, better keep it separate

there is also similar discussion at https://github.com/apmorton/pyhidapi/pull/29 with same conclusion

in any case, it is great to know, that there is another project, that has switched to this repository

Qbicz, 2019-08-30:
Agree with @todbot and @Youw.

Thank you for good work being done @prusnak!

closed 2019-08-30 by Qbicz