libusb/hidapi #63: Coverity Scans reports available

LudovicRousseau, 2019-08-11:
A long time ago I registered the hidapi project at Coverity Scans https://scan.coverity.com/faq
The project should be available at https://scan.coverity.com/projects/hidapi?tab=overview

I can add any other member of this hidapi project to have access to the Coverity Scans results. Just connect  on Coverity Scans and ask for an access to the hidapi results.

For now only 2 issues are reported by Coverity Scans.

Youw, 2019-08-11:
Sent request.

LudovicRousseau, 2019-08-11:
Aprouved

LudovicRousseau, 2019-11-01:
Coverity found a new bug in `testgui/test.cpp`:
```
** CID 1019782:  Uninitialized members  (UNINIT_CTOR)
/testgui/test.cpp: 73 in MainWindow::MainWindow()()

________________________________________________________________________________________________________
*** CID 1019782:  Uninitialized members  (UNINIT_CTOR)
/testgui/test.cpp: 73 in MainWindow::MainWindow()()
67      hid_device *connected_device;
68      size_t getDataFromTextField(FXTextField *tf, char *buf, size_t len);
69      int getLengthFromTextField(FXTextField *tf);
70     
71     
72     protected:
>>>     CID 1019782:  Uninitialized members  (UNINIT_CTOR)
>>>     Non-static class member "connected_device" is not initialized in this constructor nor in any functions that it calls.
73      MainWindow() {};
74     public:
75      MainWindow(FXApp *a);
76      ~MainWindow();
77      virtual void create();
78      
```

Youw, 2019-11-02:
Thanks for catching it.

This is just a test app, not a part of any critical implementation.
Will have a fix for it as soon as someone have a chance to get his hands to it.

mcuee, 2023-05-02:
https://scan.coverity.com/projects/hidapi?tab=overview
Looks like there are two outstanding issues.

Youw, 2023-05-02:
I think those are outdated.
We don't have automatic build submissions (yet). Probably a good idea to setup some.

LudovicRousseau, 2023-05-03:
I submitted a new scan with the current source code a few hours ago.
The results are not yet available. Strange.

LudovicRousseau, 2023-05-03:
The status is still " Last Build Status: Running. Your build is currently being analyzed ".
Maybe the server is overloaded.

mcuee, 2023-05-04:
Indeed it seems the server is overloaded. It still shows "Last Build Status: Running. Your build is currently being analyzed".

mcuee, 2023-05-09:
@LudovicRousseau 

Looks like the submission failed. Please help to submit a new build. Thanks.

LudovicRousseau, 2023-05-09:
New build submitted.
The coverity results are now visible.

Youw, 2023-05-09:
Thanks.
This time we did found some valid issues (error handling mostly).

JoergAtGithub, 2023-05-09:
I see only Linux files listed. Were the Windows files also scanned?

LudovicRousseau, 2023-05-09:
@JoergAtGithub I did a GNU/Linux build only. I do not use Windows.
Feel free to submit a Windows build.

Youw, 2023-05-14:
I've prepared an on-demand Github Actions job to run and submit Coverity report for 3 platforms (Windows, Linux, macOS).
But it is not exactly usable yet: each submission overrides the previous one, so even if the job is executed - the results only for one platform will be available (whichever runs last).

Splitting into 3 different jobs for each platform is not exactly a solution, since still only one report will be available at the time, and what I want to achieve - a combiner report for all HIDAPI sources for all platforms, to track all issues alltogether on Coverity Web UI.

I'll try to get it done some time this week.

Meanwhile I've manually executed all jobs on all platforms against current master and created tickets for each issue found by Coverity.

Youw, 2023-05-21:
Since #557 is merged, we have a single multi-platform job to submit Coverity scans for all platforms of HIDAPI. (With fixed issues I've described in my prev message.)

#558 to not forget about those checks.

That being said, I think we can safely close this ticket.

closed 2023-05-21 by Youw