libusb/hidapi #64: Please add link to release DLL to readme

Zaaphod, 2019-08-20:
In the DOWNLOAD section of the readme on the main page for libusb/hidapi here: [https://github.com/libusb/hidapi](https://github.com/libusb/hidapi) PLEASE add that the latest release version of hidapi can be downloaded from:
[https://github.com/libusb/hidapi/releases](https://github.com/libusb/hidapi/releases)

I did not want to download the entire repository and figure out how to build it myself, I just wanted the last stable release so I could use it with easyhid for python under windows, and I could not figure it out at all.

I have been looking for this for DAYS it does not show up on ANY google searches when you do a search for HIDAPI.DLL and it's impossible for anyone who just needs the latest release dll to find it.

Finally someone was kind enough to point me to the link, if not I would still not be able to find it.. searching for it doesn't work because it's in a zip file.

Youw, 2019-08-20:
If you're searching for pre-compiled version of library - looking at `releases` page on github would be a logical first place, where to look at :)

Try it yourself with some other libraries sometime.

I don't thing adding link to `releases` to README necessary.

Zaaphod, 2019-08-20:
It would have absolutely never occurred to me to look there.. it's one of those things.. once you know it, it seems obvious, I'll keep it in mind, I'm just hoping to save others the frustration I've been having.... Thanks again for the help!!

Zaaphod, 2019-08-20:
maybe just put "HIDAPI.DLL release"  somewhere on the [https://github.com/libusb/hidapi/releases](https://github.com/libusb/hidapi/releases) page.. at least then I would have found it with a google search.

Youw, 2019-08-20:
We could add this message to release description:
```
Includes pre-compiled:
- hidapi.dll
- hidapi.lib
For x86 and x64 Windows.
```

@Qbicz what do you think?

Qbicz, 2019-08-21:
@Youw your suggestion looks reasonable. I totally agree that we should make this released file searchable!

Qbicz, 2019-08-21:
I updated the release description, hope it will help future seekers.

closed 2019-08-21 by Qbicz