libusb/hidapi #68: Test GUI build error on windows

gandomi, 2019-08-29:
Hi!

I want to build the library with Test GUI using Cygwin on windows. But after adding `--enable-testgui` argument to `./configure` and run `make` it gives me many `undefined reference to ...` errors. But without `--enable-testgui` argument the library is built successfully.
the whole `make` output command is as follow:
```
make  all-recursive
make[1]: Entering directory '/cygdrive/d/hidapi-master'
Making all in windows
make[2]: Entering directory '/cygdrive/d/hidapi-master/windows'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/cygdrive/d/hidapi-master/windows'
Making all in hidtest
make[2]: Entering directory '/cygdrive/d/hidapi-master/hidtest'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/cygdrive/d/hidapi-master/hidtest'
Making all in testgui
make[2]: Entering directory '/cygdrive/d/hidapi-master/testgui'
  CXXLD    hidapi-testgui.exe
test.o: In function `FX::FXMainWindow::FXMainWindow()':
/cygdrive/d/hidapi-master/testgui/./../../hidapi-externals/fox/include/FXMainWindow.h:46: undefined reference to `FX::FXTopWindow::FXTopWindow()'
/cygdrive/d/hidapi-master/testgui/./../../hidapi-externals/fox/include/FXMainWindow.h:46:(.text+0x27): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXTopWindow::FXTopWindow()'
test.o: In function `MainWindow::handle(FX::FXObject*, unsigned int, void*)':
/cygdrive/d/hidapi-master/testgui/test.cpp:113: undefined reference to `FX::FXMetaClass::search(unsigned int) const'
/cygdrive/d/hidapi-master/testgui/test.cpp:113:(.text+0x7f): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXMetaClass::search(unsigned int) const'
test.o: In function `MainWindow::onClear(FX::FXObject*, unsigned int, void*)':
/cygdrive/d/hidapi-master/testgui/test.cpp:475: undefined reference to `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:475:(.text+0xf8): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:475: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:475:(.text+0x10b): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:475: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:475:(.text+0x123): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXString::~FXString()'
test.o: In function `MainWindow::onConnect(FX::FXObject*, unsigned int, void*)':
/cygdrive/d/hidapi-master/testgui/test.cpp:230: undefined reference to `FX::FXList::getItem(int) const'
/cygdrive/d/hidapi-master/testgui/test.cpp:230:(.text+0x1e6): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXList::getItem(int) const'
/cygdrive/d/hidapi-master/testgui/test.cpp:246: undefined reference to `FX::FXApp::addTimeout(FX::FXObject*, unsigned int, unsigned int, void*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:246:(.text+0x247): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXApp::addTimeout(FX::FXObject*, unsigned int, unsigned int, void*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:249: undefined reference to `FX::FXString::FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:249:(.text+0x24f): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXString::FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:250: undefined reference to `FX::FXString::format(char const*, ...)'
/cygdrive/d/hidapi-master/testgui/test.cpp:250:(.text+0x268): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXString::format(char const*, ...)'
/cygdrive/d/hidapi-master/testgui/test.cpp:251: undefined reference to `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:251:(.text+0x280): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:251: undefined reference to `FX::operator+(FX::FXString const&, wchar_t const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:251:(.text+0x293): additional relocation overflows omitted from the output
/cygdrive/d/hidapi-master/testgui/test.cpp:251: undefined reference to `FX::FXString::operator+=(FX::FXString const&)'
/cygdrive/d/hidapi-master/testgui/test.cpp:251: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:251: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::operator+(FX::FXString const&, wchar_t const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::FXString::operator+=(FX::FXString const&)'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:253: undefined reference to `FX::FXLabel::setText(FX::FXString const&)'
/cygdrive/d/hidapi-master/testgui/test.cpp:259: undefined reference to `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:259: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:249: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:240: undefined reference to `FX::FXMessageBox::error(FX::FXWindow*, unsigned int, char const*, char const*, ...)'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:252: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:249: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:259: undefined reference to `FX::FXString::~FXString()'
test.o: In function `MainWindow::onDisconnect(FX::FXObject*, unsigned int, void*)':
/cygdrive/d/hidapi-master/testgui/test.cpp:270: undefined reference to `FX::FXString::FXString(char const*)'
/cygdrive/d/hidapi-master/testgui/test.cpp:270: undefined reference to `FX::FXLabel::setText(FX::FXString const&)'
/cygdrive/d/hidapi-master/testgui/test.cpp:270: undefined reference to `FX::FXString::~FXString()'
/cygdrive/d/hidapi-master/testgui/test.cpp:277: undefined reference to `FX::FXApp::removeTimeout(FX::FXObject*, unsigned int)'
/cygdrive/d/hidapi-master/testgui/test.cpp:270: undefined reference to `FX::FXString::~FXString()'
test.o: In function `MainWindow::onRescan(FX::FXObject*, unsigned int, void*)':
/cygdrive/d/hidapi-master/testgui/test.cpp:287: undefined reference to `FX::FXList::clearItems(unsigned char)'
/cygdrive/d/hidapi-master/testgui/test.cpp:295: undefined reference to `FX::FXString::FXString()'
[... 231 lines cut ...]
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x150): undefined reference to `FX::FXTopWindow::hide()'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x158): undefined reference to `FX::FXComposite::isComposite() const'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x160): undefined reference to `FX::FXWindow::dropEnable()'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x168): undefined reference to `FX::FXWindow::dropDisable()'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x170): undefined reference to `FX::FXWindow::contains(int, int) const'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x178): undefined reference to `FX::FXWindow::setBackColor(unsigned int)'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x180): undefined reference to `FX::FXWindow::doesSaveUnder() const'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x188): undefined reference to `FX::FXWindow::tr(char const*, char const*) const'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x190): undefined reference to `FX::FXTopWindow::show(unsigned int)'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x198): undefined reference to `FX::FXTopWindow::maximize(unsigned char)'
test.o:test.cpp:(.rdata$_ZTV10MainWindow[_ZTV10MainWindow]+0x1a0): undefined reference to `FX::FXTopWindow::minimize(unsigned char)'
[... cut ...]
test.o:test.cpp:(.rdata$.refptr._ZN2FX12FXMainWindow9metaClassE[.refptr._ZN2FX12FXMainWindow9metaClassE]+0x0): undefined reference to `FX::FXMainWindow::metaClass'
test.o:test.cpp:(.rdata$.refptr._ZTVN2FX10FXListItemE[.refptr._ZTVN2FX10FXListItemE]+0x0): undefined reference to `vtable for FX::FXListItem'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:502: hidapi-testgui.exe] Error 1
make[2]: Leaving directory '/cygdrive/d/hidapi-master/testgui'
make[1]: *** [Makefile:556: all-recursive] Error 1
make[1]: Leaving directory '/cygdrive/d/hidapi-master'
make: *** [Makefile:440: all] Error 2
```

Am I wrong somewhere?

Youw, 2019-08-28:
Do you have Fox toolkit installed locally?
See here: https://github.com/libusb/hidapi/blob/master/configure.ac#L187

gandomi, 2019-08-29:
Yes, I get the `hidapi-externals.zip` package from [here](https://github.com/downloads/signal11/hidapi/hidapi-externals.zip) and extract it just outside of hidapi. Now i have:
```
Parent_Folder
       |
       +hidapi-master
       +hidapi-externals
```

Youw, 2019-08-29:
The error you have is a link time error.
Either `libFOX-1.6.a` not found or not accepted by linker.

Can you share `configure` and `make` output?

Youw, 2019-08-29:
Tried locally on my PC: it appears, that pre-build FOX Toolkit from here: https://github.com/downloads/signal11/hidapi/hidapi-externals.zip was build by a compiler that is incompatible with latest GCC from Cygwin.

Build FOX from sources using same compiler, replace header/binary at hidapi-external, and you should be good (tried that locally, it worked).

Please don't ask how to build FOX using Cygwin - this is a matter of a separate discussion and a bit off-topic (yes I had issues with it and don't have elegant solution for all of them).

gandomi, 2019-08-30:
I tried to build FOX with current installed Cygwin but it has some errors too. Ok I don't ask.
Thank you in advance for your help.

mcuee, 2019-12-12:
Not exactly the same but related. I tried to build the test gui under MSys 2 and it failed as well.

```
ephwy@ephwy-HP MINGW64 /c/work/hidapi/hidapi
$ ./configure --enable-testgui
configure: loading site script /mingw64/etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw64/bin/nm -B
checking the name lister (/mingw64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /mingw64/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... no
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -DDLL_EXPORT -DPIC
[... 66 lines cut ...]
Making all in hidtest
make[2]: Entering directory '/c/work/hidapi/hidapi/hidtest'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/c/work/hidapi/hidapi/hidtest'
Making all in testgui
make[2]: Entering directory '/c/work/hidapi/hidapi/testgui'
  CXX      test.o
test.cpp:18:10: fatal error: fx.h: No such file or directory
   18 | #include <fx.h>
      |          ^~~~~~
compilation terminated.
make[2]: *** [Makefile:511: test.o] Error 1
make[2]: Leaving directory '/c/work/hidapi/hidapi/testgui'
make[1]: *** [Makefile:555: all-recursive] Error 1
make[1]: Leaving directory '/c/work/hidapi/hidapi'
make: *** [Makefile:439: all] Error 2

ephwy@ephwy-HP MINGW64 /c/work/hidapi/hidapi
$ pacman -Ss fox
mingw32/mingw-w64-i686-fox 1.6.57-1 (mingw-w64-i686)
    C++ user interface toolkit (mingw-w64)
mingw64/mingw-w64-x86_64-fox 1.6.57-1 (mingw-w64-x86_64) [installed]
    C++ user interface toolkit (mingw-w64)

```

mcuee, 2019-12-12:
The only thing is that the fox library is hard coded. Probably it is better to change to use pkg-config.

mcuee, 2019-12-12:
Of course you can use the ./configure option to fix the issue, but it is probably better to use pkg-config.

```

```

mcuee, 2021-06-11:
Somehow pkg-config is actually correct. Need to check again.

```
$ pkg-config --cflags fox
-IC:/msys64/mingw64/include/fox-1.6

XFCHEN@1TV4433 MINGW64 /c/work/libusb/hidapi
$ pkg-config --libs fox
-lFOX-1.6
```

mcuee, 2021-06-14:
This one is tough.

```
$ ls /mingw64/include/fox-1.6/fx.h
/mingw64/include/fox-1.6/fx.h

$ ls /mingw64/lib/libFOX-1.6.*
/mingw64/lib/libFOX-1.6.a  /mingw64/lib/libFOX-1.6.dll.a

$ fox_CFLAGS="-I/mingw64/include/fox-1.6" fox_LIBS="-lFox-1.6" ./configure --enable-testgui
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw64/bin/nm -B
checking the name lister (/mingw64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /mingw64/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for dlfcn.h... no
[... 71 lines cut ...]
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hid.c:457:11: note: length computed here
  457 |     len = strlen(str);
      |           ^~~~~~~~~~~
  CCLD     libhidapi.la
make[2]: Leaving directory '/c/work/hid/hidapi/windows'
Making all in hidtest
make[2]: Entering directory '/c/work/hid/hidapi/hidtest'
  CC       test.o
  CCLD     hidtest.exe
make[2]: Leaving directory '/c/work/hid/hidapi/hidtest'
Making all in testgui
make[2]: Entering directory '/c/work/hid/hidapi/testgui'
  CXX      test.o
test.cpp:18:10: fatal error: fx.h: No such file or directory
   18 | #include <fx.h>
      |          ^~~~~~
compilation terminated.
make[2]: *** [Makefile:520: test.o] Error 1
make[2]: Leaving directory '/c/work/hid/hidapi/testgui'
make[1]: *** [Makefile:560: all-recursive] Error 1
make[1]: Leaving directory '/c/work/hid/hidapi'
make: *** [Makefile:444: all] Error 2

```

mcuee, 2022-04-25:
README.md has been updated. So I think we can close this issue.
https://github.com/libusb/hidapi/blob/master/README.md

closed 2022-04-25 by mcuee