libusb/hidapi #71: Are there plans to add vcpkg recipe for hidapi?

kejxu, 2019-09-04:
As mentioned in the title, are there any plans at this point to enable building `hidapi` with [vcpkg](https://github.com/microsoft/vcpkg/)?

Youw, 2019-09-05:
Not by a Core team, but PRs are welcome.

kejxu, 2019-09-10:
@Youw just created https://github.com/microsoft/vcpkg/pull/8113 as a very simple entry point for `vcpkg` to build with the `.sln` file on Windows, it'd be great if `hidapi` could have another release to generate `.vcxproj` project files for modern VS

Youw, 2019-09-10:
Thanks for you effort!

Generating new version of .sln/.vcxproj means having two almost identical build systems (for MSVC2015/MSVC2017 and later for MSVC2019 I guess), which personally I don't find very convenient from maintenance point of view.

In stead we have plans of creating CMake project #12, which can be used to generate any MSVC build project the user likes (and supported by vcpkg naively).

kejxu, 2019-09-10:
> Generating new version of .sln/.vcxproj means having two almost identical build systems (for MSVC2015/MSVC2017 and later for MSVC2019 I guess), which personally I don't find very convenient from maintenance point of view.

good point

> In stead we have plans of creating CMake project #12, which can be used to generate any MSVC build project the user likes (and supported by vcpkg natively).

that sounds like the best approach! I did not have adequate knowledge on the project to create a proper `CMakeLists.txt`, so using the `.sln` file was a simple workaround. Once the CMake project is created, it'd be a lot easier to integrate into `vcpkg` in a cross-platform way.

Stoccarda, 2019-09-16:
Just as a comment, a bit off topic : hidapi can be directly compiled with C++ Builder Rio, using internal CLANG as well as classical Borland compiler

Youw, 2019-09-23:
Since microsoft/vcpkg#8113 is merged, I guess this is no longer a question.

closed 2019-09-23 by Youw