libusb/hidapi #103: HID connection status?

JacoFourie, 2019-10-01:
Hi, this is more a question than an issue. I am new to HID over USB. I found the lib. I use it via Lazarus / free pascal binding. All seems to be working but I have some questions. How do I detect the connection status? Let's say you pull out the USB device and push it back in. Or you lose connection to the device and want to retry the connection. I only see an Open and Close API.

todbot, 2019-10-02:
Not directly with hidapi.  You can try sending a report and if it does not error out, the OS still believes the device is connected.  With your device, there may be a "get status" or "get version" command you can periodically send to determine device plugged in.  

To determine device insertion, you can periodically use `hid_open()` or `hid_enumerate()` and check for any devices returned.  Be careful doing this too often though, as USB enumeration can be expensive on some OSes.

Alternatively, there may be a "usb detection" library for your language.  For instance, in Node.js, there's the `node-hid` wrapper on `hidapi` but there's also `node-usb-detection` that is specifically for providing plug/unplug events.

Youw, 2019-10-02:
On Linux I'd suggest using udev library to monitor plug/unplug events.

JacoFourie, 2019-10-02:
Can I use libusb with hidapi. libusb has hotplug events. So my plan is to use libusb to monitor the status of the usb device and use hidapi to send and receive data.

z3ntu, 2019-10-02:
I've implemented something like this (with Qt signals/slots) in my project: https://github.com/z3ntu/razer_test/tree/master/src/devnotifier

But I agree it would be nice if there was a proper easy cross platform solution for that.

todbot, 2019-10-02:
Yes, I believe you can use libusb with hidapi, but the preferred method is udev.

Youw, 2019-10-02:
libusb is a great choice, if you target linux only (it uses udev under the hood), since hoplug support is not yet implemented on Windows (last time I check it wasn't).

@z3ntu 's implementation is very practical - I did the same way, whan I needed this in one on my closed projects.

Jihadist, 2020-08-05:
Is there any way for windows?

Youw, 2020-08-10:
Yes, this is possible, but you need a Windows even loop, and a window handle (can be a dummy window).
I don't have a complete sample for HID devices notification, but there is a similar sample on MSDN: https://docs.microsoft.com/en-us/windows/win32/devio/registering-for-device-notification , where you probably need to use [GUID_DEVINTERFACE_HID](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/guid-devinterface-hid).

closed 2021-06-10 by mcuee