libusb/hidapi #105: Using the example listed in README breaks my keyboard

amosbird, 2019-10-17:
Hello. I'm trying to test hidapi and possibly use it to set fn_lock of my keyboard. I grabbed the example code and tried to see if it runs.  https://github.com/libusb/hidapi#what-does-the-api-look-like 

Then suddenly my keyboard stops responding, `lsusb` doesn't list it anymore. The id was `0x17ef, 0x60b5` now becomes `0x04f2, 0x1670`  (I tested it while unplug and plug). 

Is there any way of restoring my keyboard? Thank you!

amosbird, 2019-10-17:
The device is very much broken

```
Bus 003 Device 012: ID 04f2:1670 Chicony Electronics Co., Ltd
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x04f2 Chicony Electronics Co., Ltd
  idProduct          0x1670
  bcdDevice            4.00
  iManufacturer           0
  iProduct                0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      31
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

Youw, 2019-10-17:
What backend are you using, hidraw or libusb?

amosbird, 2019-10-17:
libusb   `-lhidapi-libusb`

Youw, 2019-10-17:
1) Is there any way of restoring my keyboard?

So re-plug doesn't help?

2) Have tried hidraw backend?

amosbird, 2019-10-17:
> So re-plug doesn't help?

It doesn't

> Have tried hidraw backend?

But what should I try? Rerun the example doesn't seem to be able to fix the keyboard

Youw, 2019-10-17:
Is it appears broken wne you plug it to a different PC / OS ?
(To identify is it host or dev issue.)

amosbird, 2019-10-17:
I tried plugging into windows 10 machine, doesn't work either. It's dev issue

Youw, 2019-10-17:
So far I haven't heard of devices broken by `hidapi`, and it (hidapi) definitely doesn't do anything that is out of spec from HID (or USB) perspective.

What you experience is look like device's internal state (flash memory or smth.) has been changed in some weird way, and it is likely a vendor bug.

I'd suggest trying to contact your keyboard vendor support to identify what might went wrong.

amosbird, 2019-10-17:
> I'd suggest trying to contact your keyboard vendor support to identify what might went wrong.

Then I'm out of luck.

Youw, 2019-10-17:
BTW: example from README uses device's specific control command, and it is not advisable to try it with the unknown device.
Looks like it worth explicitly adding it to the readme.

amosbird, 2019-10-17:
Exactly! I wish there is a WARNING SIGN.

z3ntu, 2019-10-17:
So it seems that `17ef:60b5` is `Lenovo ThinkPad X1 Tablet Thin Keyboard Gen 3` , is that correct?

amosbird, 2019-10-17:
Yes

Qbicz, 2019-10-18:
Looks like these 2 writes (0x80, 0x81) changed state of your keyboard in a way that the firmware authors have not intercepted. Well, you should never run code from Internet if you don't fully understand it, but we can definitely put a warning before the code sample. @amosbird I hope you will be able to contact the vendor and fix the issue.

Qbicz, 2019-10-18:
Opened https://github.com/libusb/hidapi/pull/106 to prevent problems like this.

amosbird, 2019-10-18:
Is it possible to extract the firmware from another working keyboard and flash it to this broken one?

Youw, 2019-10-18:
That is a good question to a keyboard vendor.
There is no universal solution.

amosbird, 2019-10-18:
OK. Anyway, the tuition is 100$ and I've learned it hard.

Qbicz, 2019-11-05:
Merged #106.

closed 2019-11-05 by Qbicz