libusb/hidapi #108: hidtest-libusb and hidtest-hidraw do not compile

piciu67, 2019-10-22:
hidtest.cpp was renamed to hidtest.c and example makefiles "Makefile-manual" in hidapi/libusb and hidapi/linux stopped working.

Youw, 2019-10-22:
Thanks for reporting this!

Would you be kind to prepare a PR and provide a fix?

It looks like it is a matter of removing `CPPOBJS` variable in `Makefile.linux`.

piciu67, 2019-10-22:
Simplest way fix this problem is just to rename hidtest.c back to hidtest.cpp.

Youw, 2019-10-22:
Maybe it will fix this particular issue, but it creates a bigger one: _requirement to have a C++ compiler_, which is a big deal sometime, and not really required to actually compile and run the example (see #22).

z3ntu, 2019-10-22:
A start for the patch would be
```diff
diff --git a/linux/Makefile-manual b/linux/Makefile-manual
index 04ce1de..c8aab27 100644
--- a/linux/Makefile-manual
+++ b/linux/Makefile-manual
@@ -19,16 +19,15 @@ CXXFLAGS ?= -Wall -g -fpic
 LDFLAGS  ?= -Wall -g
 
 
-COBJS     = hid.o
-CPPOBJS   = ../hidtest/hidtest.o
-OBJS      = $(COBJS) $(CPPOBJS)
+COBJS     = hid.o ../hidtest/hidtest.o
+OBJS      = $(COBJS)
 LIBS_UDEV = `pkg-config libudev --libs` -lrt
 LIBS      = $(LIBS_UDEV)
 INCLUDES ?= -I../hidapi `pkg-config libusb-1.0 --cflags`
 
 
 # Console Test Program
-hidtest-hidraw: $(COBJS) $(CPPOBJS)
+hidtest-hidraw: $(COBJS)
 	$(CXX) $(LDFLAGS) $^ $(LIBS_UDEV) -o $@
 
 # Shared Libs
@@ -39,9 +38,6 @@ libhidapi-hidraw.so: $(COBJS)
 $(COBJS): %.o: %.c
 	$(CC) $(CFLAGS) -c $(INCLUDES) $< -o $@
 
-$(CPPOBJS): %.o: %.cpp
-	$(CXX) $(CXXFLAGS) -c $(INCLUDES) $< -o $@
-
 
 clean:
 	rm -f $(OBJS) hidtest-hidraw libhidapi-hidraw.so ../hidtest/hidtest.o

```
but the following files would have to be (probably) fixed as well:
libusb/Makefile.freebsd
libusb/Makefile.linux
mac/Makefile-manual
testgui/Makefile.freebsd
testgui/Makefile.linux
testgui/Makefile.mac
testgui/Makefile.mingw
windows/Makefile.mingw

piciu67, 2019-10-22:
Right. I'll try to prepare a PR, but I'm quite new to github and it'll take me some time to me to learn how to do it :)

Youw, 2019-10-22:
Even better: you contribute to open-source AND improve your own skills!
Have fun :)

piciu67, 2019-10-22:
> Even better: you contribute to open-source AND improve your own skills!
> Have fun :)

I deleted my promise to prepare PR as irrelevant, you guys are so much faster than me :)

Youw, 2019-10-22:
No body in rush (we aren't), no body makes any demands or obligations.
Suit yourself.

mcuee, 2019-12-14:
The issue seems to be gone with #115.

z3ntu, 2019-12-14:
Should be resolved completely with #131

PR #131 (2020-01-12): Fix manual Makefiles and add CI for them
Fixes #108
closed 2020-01-12 by Youw