libusb/hidapi #112: hiadpi venv python modue does not build on FreeBSD via pip

cederom, 2019-10-27:
Hello world :-)

`hidapi` does not build on FreeBSD via python 3.6 venv pip. Looks like `-lusb-1.0` problem as FreeBSD has its own libusb implementation that should use simply `-lusb`.

Also noted warning related to `iconv` type casting:
```
    hidapi/libusb/hid.c:458:18: warning: passing 'const char **' to parameter of type 'char **' discards qualifiers in nested pointer types [-Wincompatible-pointer-types-discards-qualifiers]
            res = iconv(ic, &inptr, &inbytes, &outptr, &outbytes);
                            ^~~~~~
    /usr/include/iconv.h:58:41: note: passing argument to parameter here
    size_t  iconv(iconv_t, char ** __restrict,
                                             ^
    1 warning generated.
```

Details below:
```
    Running setup.py install for hidapi ... error
    ERROR: Command errored out with exit status 1:
     command: /(..)/venv36bsd/bin/python3.6 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rhbcnebq/hidapi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rhbcnebq/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-e8k1rl3a/install-record.txt --single-version-externally-managed --compile --install-headers /(..)/venv36bsd/include/site/python3.6/hidapi
         cwd: /tmp/pip-install-rhbcnebq/hidapi/
    Complete output (25 lines):
    running install
    running build
    running build_ext
    cythoning hid.pyx to hid.c
    /tmp/pip-install-rhbcnebq/hidapi/.eggs/Cython-0.29.13-py3.6-freebsd-12.0-RELEASE-p10-amd64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-rhbcnebq/hidapi/hid.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    building 'hid' extension
    creating build
    creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6
    creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi
    creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi/libusb
    cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/(..)/venv36bsd/include -I/usr/local/include/python3.6m -c hid.c -o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.o
    cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/(..)/venv36bsd/include -I/usr/local/include/python3.6m -c hidapi/libusb/hid.c -o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi/libusb/hid.o
    hidapi/libusb/hid.c:458:18: warning: passing 'const char **' to parameter of type 'char **' discards qualifiers in nested pointer types [-Wincompatible-pointer-types-discards-qualifiers]
            res = iconv(ic, &inptr, &inbytes, &outptr, &outbytes);
                            ^~~~~~
    /usr/include/iconv.h:58:41: note: passing argument to parameter here
    size_t  iconv(iconv_t, char ** __restrict,
                                             ^
    1 warning generated.
    creating build/lib.freebsd-12.0-RELEASE-p10-amd64-3.6
    cc -shared -lpthread -L/usr/local/lib -fstack-protector-strong build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi/libusb/hid.o -L/usr/local/lib -lusb-1.0 -lpython3.6m -o build/lib.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.so
    /usr/bin/ld: error: unable to find library -lusb-1.0
    cc: error: linker command failed with exit code 1 (use -v to see invocation)
    error: command 'cc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /(..)/venv36bsd/bin/python3.6 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rhbcnebq/hidapi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rhbcnebq/hidapi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-e8k1rl3a/install-record.txt --single-version-externally-managed --compile --install-headers /(..)/venv36bsd/include/site/python3.6/hidapi Check the logs for full command output.
```

cederom, 2019-10-27:
Update:
* in git master head the `iconv()` problem seems to be fixed (but is present in 0.9.0 release).
* git master head builds fine standalone (not from pip).
* 0.9.0 release builds fine standalone (not from pip).
* problem seems to be related with python module build only?

Youw, 2019-10-28:
> problem seems to be related with python module build only?

Yes, it appears so.
Can you specify which module/how exactly have you tried to build (the command line)?

cederom, 2019-11-01:
Hello @Youw and thank you for you reply :-) I am working with ARM MBED OS [mbed] that uses HIDAPI as dependency. Environment is python based. I am working inside dedicated VENV.

Here is the environment setup:
```
(venv36bsd) 0xCFMX4% uname -a
FreeBSD 0xCFMX4 12.0-RELEASE-p10 FreeBSD 12.0-RELEASE-p10 GENERIC  amd64
(venv36bsd) 0xCFMX4% python -V
Python 3.6.9
(venv36bsd) 0xCFMX4% which python
/tmp/venv36bsd/bin/python
```

Here goes the `pip` installation:
```
(venv36bsd) 0xCFMX4% pip install hidapi | tee hidapi1.txt
Collecting hidapi
  Using cached https://files.pythonhosted.org/packages/c1/86/89df0e8890f96eeb5fb68d4ccb14cb38e2c2d2cfd7601ba972206acd9015/hidapi-0.7.99.post21.tar.gz
Requirement already satisfied: setuptools>=19.0 in ./venv36bsd/lib/python3.6/site-packages (from hidapi) (40.6.2)
Installing collected packages: hidapi
  Running setup.py install for hidapi: started
    Running setup.py install for hidapi: finished with status 'error'
    Complete output from command /tmp/venv36bsd/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rycmltsn/hidapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-o5elomlu/install-record.txt --single-version-externally-managed --compile --install-headers /tmp/venv36bsd/include/site/python3.6/hidapi:
    running install
    running build
    running build_ext
    cythoning hid.pyx to hid.c
    /tmp/pip-install-rycmltsn/hidapi/.eggs/Cython-0.29.13-py3.6-freebsd-12.0-RELEASE-p10-amd64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /tmp/pip-install-rycmltsn/hidapi/hid.pyx
      tree = Parsing.p_module(s, pxd, full_module_name)
    building 'hid' extension
    creating build
    creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6
    creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi
    creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi/libusb
    cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/tmp/venv36bsd/include -I/usr/local/include/python3.6m -c hid.c -o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.o
    cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -fPIC -Ihidapi/hidapi -I/usr/include/libusb-1.0 -I/tmp/venv36bsd/include -I/usr/local/include/python3.6m -c hidapi/libusb/hid.c -o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi/libusb/hid.o
    hidapi/libusb/hid.c:458:18: warning: passing 'const char **' to parameter of type 'char **' discards qualifiers in nested pointer types [-Wincompatible-pointer-types-discards-qualifiers]
            res = iconv(ic, &inptr, &inbytes, &outptr, &outbytes);
                            ^~~~~~
    /usr/include/iconv.h:58:41: note: passing argument to parameter here
    size_t  iconv(iconv_t, char ** __restrict,
                                             ^
    1 warning generated.
    creating build/lib.freebsd-12.0-RELEASE-p10-amd64-3.6
    cc -shared -lpthread -L/usr/local/lib -fstack-protector-strong build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hidapi/libusb/hid.o -L/usr/local/lib -lusb-1.0 -lpython3.6m -o build/lib.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.so
    /usr/bin/ld: error: unable to find library -lusb-1.0
    cc: error: linker command failed with exit code 1 (use -v to see invocation)
    error: command 'cc' failed with exit status 1

    ----------------------------------------
Command "/tmp/venv36bsd/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rycmltsn/hidapi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-o5elomlu/install-record.txt --single-version-externally-managed --compile --install-headers /tmp/venv36bsd/include/site/python3.6/hidapi" failed with error code 1 in /tmp/pip-install-rycmltsn/hidapi/
You are using pip version 18.1, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

```

[mbed] https://github.com/ARMmbed/mbed-os

Qbicz, 2019-11-01:
Thanks for the report. It looks like you are building cython-hidapi python library. I think the details for the BSD build are handled here:
https://github.com/trezor/cython-hidapi/blob/master/setup.py#L82

This is probably a special case that is not correctly handled for FreeBSD. You can try to download the cython-hidapi project to your target environment and tweak the `setup.py` script until it works correctly.

FYI @prusnak

cederom, 2019-11-01:
Thank you @Qbicz and @Youw :-)

I have already provided a pull request with a fix (https://github.com/trezor/cython-hidapi/pull/67), build is fine now on FreeBSD. When would that get into a `pip` repository?

```
(venv36bsd) 0xCFMX4% uname -a
FreeBSD 0xCFMX4 12.0-RELEASE-p10 FreeBSD 12.0-RELEASE-p10 GENERIC  amd64
(venv36bsd) 0xCFMX4% python -V
Python 3.6.9
(venv36bsd) 0xCFMX4% python setup.py build
running build
running build_ext
skipping 'hid.c' Cython extension (up-to-date)
building 'hid' extension
creating build
creating build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6
cc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -O2 -pipe -fstack-protector-strong -fno-strict-aliasing -fPIC -I/usr/local/include/hidapi -I/(..)/venv36bsd/include -I/usr/local/include/python3.6m -c hid.c -o build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.o
creating build/lib.freebsd-12.0-RELEASE-p10-amd64-3.6
cc -shared -lpthread -L/usr/local/lib -fstack-protector-strong build/temp.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.o -L/usr/local/lib -lusb -lhidapi -lpython3.6m -o build/lib.freebsd-12.0-RELEASE-p10-amd64-3.6/hid.so
```

Qbicz, 2019-11-02:
I'm glad you were able to fix this. Your fix will be available on pip once your PR is merged and the maintainers create a new release to PyPI: https://pypi.org/project/hidapi/#data

prusnak, 2019-11-02:
I think this one should be closed in favor of https://github.com/trezor/cython-hidapi/pull/67

cederom, 2019-11-02:
Sure thing! Thank you everyone for your help! :-)

closed 2019-11-02 by cederom