libusb/hidapi #116: macOS cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/HELLOKEY-8F3A4DEAFF1A0D6A@14310000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14310000,0

GanZhiXiong, 2019-11-08:
Using 'node-hid' can open the device on Windows and Linux, but it cannot be opened on macOS. 

I've submitted in 'node - hid issue, and discussed this matter todbot, specific can look here: https://github.com/node-hid/node-hid/issues/348

Youw, 2019-11-08:
As far as I can see in the discussion, your device is a keyboard-like device, and as @todbot already mentioned, you will not be able to open it as a general HID device since macOS won't let you.

If the device is your own development, I'd suggest adding another USB-HID interface to that device with a custom (vendor) HID protocol (or use one of the standard ones, but not a Keyboard), and use it for communication you need.

Without the device itself there is not much we can help you with.

z3ntu, 2019-11-08:
Have you maybe tried running your application as root?

GanZhiXiong, 2019-11-11:
> Have you maybe tried running your application as root?

Even if I run with the 'sudo' command, I can't turn on the device

GanZhiXiong, 2019-11-15:
@Youw  @z3ntu 
In macOS calls HID.Devices(),  MacOS there's only one usagePage and one usage for the same device, macOS Device cannot be open.

But you can get two usagePage and two usage on the same device under Windows, and can open device.
 ```
This is the device information that you got under Windows

{
    vendorId: 6790,
    productId: 57568,
    path: '\\\\?\\hid#vid_1a86&pid_e0e0&col02#7&83dd5c8&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',      
    serialNumber: '30A31F8ED8F3BFCE',
    manufacturer: 'TJT-USB',
    product: 'test-30A31F8ED8F3BFCE',
    release: 0,
    interface: -1,
    usagePage: 1,
    usage: 6
  },
  {
    vendorId: 6790,
    productId: 57568,
    path: '\\\\?\\hid#vid_1a86&pid_e0e0&col01#7&83dd5c8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: '30A31F8ED8F3BFCE',
    manufacturer: 'TJT-USB',
    product: 'test-30A31F8ED8F3BFCE',
    release: 0,
    interface: -1,
    usagePage: 65280,
    usage: 1
  },
```

jdk, 2019-11-15:
Don’t believe one of my patches has been merged yet. Can you try my fork and let me know if that works?

https://github.com/flirc/hidapi

Very Sincerely,
Jason

Youw, 2019-11-15:
@GanZhiXiong macOS merges all usage pages into a single device, if endpoints don't overlap over usage pages, so, as a workaround for now, you may just skip filtering usage page on macOS and open a device by VID/PID/SN only, on macOS.

Once #65 is done - hidapi will report separate devices for different usage pages on macOS. Technically it will be same native device.

GanZhiXiong, 2019-11-18:
@jdk   Thanks. I use 'node - hid' https://github.com/node-hid/node-hid, I know 'node - hid' use 'hidapi', but I don't know how to use your fork in 'node - hid'.

todbot, 2019-11-18:
I’m in the process of updating node-hid to use this updated version. The new version of node-his should be out in a few days. Thanks for your patience! (and thanks to hidapi team for applying all these fixes)

GanZhiXiong, 2019-11-18:
@todbot  @Youw   
Thank you very much. I've been tracking this problem for a few months, and I'm looking forward to the new version of 'node-hid' that solves my problem! Thank you again! 😀 👍

todbot, 2019-11-18:
Also, @jdk, can you check libusb/hidapi versus your signal11/hidapi changes?  It appears hidapi/mac/hid.c supports interface numbers now.

jdk, 2019-11-18:
Yes, will do this tomorrow. I need a series of hubs to help repro this. The original version had interface numbers as well, but the issue happened when we had two many parent hubs and reading our string failed. Will report back.

Very Sincerely,
Jason

GanZhiXiong, 2019-11-25:
> Don’t believe one of my patches has been merged yet. Can you try my fork and let me know if that works? https://github.com/flirc/hidapi Very Sincerely, Jason
> […](#)

@jdk   I use macOS, can you tell me how to operate? What IDE should I use? Commands to execute?

jdk, 2019-11-25:
I can confirm that the latest libusb/hidapi code still doesn’t solve the issue. My patch is still needed. I’ll rebase and do a pull request.

todbot, 2019-11-25:
Hi @jdk, the hidapi PR is https://github.com/libusb/hidapi/pull/65 if you haven't seen it already. I'm not sure why it was closed, I was getting ready to make the changes requested but I won't be able to get to it until Weds.  If you're quick with git, I think one could knock it out quickly.

Youw, 2019-11-25:
Wow, it appears I closed it 2 day ago, but I definitely didn't mean to.
Maybe I pressed "Close and comment" instead of "Comment" by accident, when giving you an answer.

The patch is still welcome anyway.

jdk, 2019-11-25:
For clarification, #65 may solve another issue. The patch in my branch wasn't specifically for a usage table, but may be related if a device interface can not be found. The original issue for my patch was: signal11/hidapi#301

signal11/hidapi#301 is long standing and patches to address were:
https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5
and
https://github.com/flirc/hidapi/commit/a4dd0f33ecae939cdad08c00934cac994cda3202

Issue signal11/hidapi#301 is still present on our current top of tree. I will rebase these patches this week and submit a pull request.

Youw, 2019-11-25:
it seems that a variation of flirc/hidapi@a4dd0f3 is already in master

as for flirc/hidapi@8d251c3 - I think it is not related to #65
and worth a separate discussion (e.g., how to reproduce the original issue it tries to solve)

epegzz, 2020-03-24:
I've been tracking this issue for a while now, as I'm having the same issue. 

Trying to use hidapi to talk to Elgato Stream Deck 
```
[ { model: 'xl',
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG2@1,2/IOPP/UPSB@0/IOPP/DSB1@1/IOPP/UPS0@0/IOPP/pci-bridge@0/IOPP/pci1b73,1100@0/AppleUSBXHCIFL1100@02000000/AppleUSB20XHCIPort@02200000/Stream Deck XL@02200000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice',
    serialNumber: 'CL18I1A01810' } ]
```

I lost a bit track of the status though. Is there any pending PR that may solve the issue? :)

Youw, 2020-03-24:
Not right now.
If your device is a keyboard-like device - there might even not be a solution for it, if it is under a limitation of the OS security policies.

epegzz, 2020-03-26:
Yeah, I guess the Stream Deck is keyboard like, as it has mainly keyUp and keyDown events. But I'd also assume you would be able to grant those (accessibility?) MacOS permissions to the program using the hidapi though, no?

Youw, 2020-03-26:
Not sure regarding accessibility, never tried.

Not having access a keyboard input events is a general security concern, since if any app could just open a keyboard and read all input, it could simply the data, like your password/card pin etc.

For key Up/Down events more reliable approach is to use native input event hooks, but those work only when your application is active/have a window.

What exactly are you trying to achieve? Is this your device (you developing it) or do you want to use an existing device?

epegzz, 2020-03-26:
I am using node-hid, so I tried adding `sh`, `zsh`, `node` to this list here, but without success:

[image]

> What exactly are you trying to achieve? Is this your device (you developing it) or do you want to use an existing device?

I want to handle keyUp and keyDown inside a custom NodeJS app instead of using the proprietary app that comes with the device.

However, I gave up on HIDAPI yesterday and am now hooking into their plugin system and send all events from via WebSocket to my node app from there. It was a bit more pain to code, but since it's done now, I'll not pursue the original approach further.

However, I'll be happy to test and debug if that helps! :)

Youw, 2020-03-26:
> the proprietary app that comes with the device

that is likely an easier solution to actually get it done

bridgewaterrobbie, 2020-04-07:
Is there a way to capture a keyboard device with HIDAPI if I am fine with the restriction that the app be focused?

I would prefer to stick with this library rather than native hooks since it is easily cross platform, and I'd prefer to use raw HID packets (My plan is to forward them to a second device and send them into a virtual HID device, so that it is "extended")

If this deserves another issue due to the focus on the Stream deck, I can split mine out, but this came up in my searching. Thanks.

todbot, 2020-04-07:
It doesn't matter the state of the app, the OS "owns" keyboards and mice for security reasons.
If your app is focussed and you're just looking for keypresses, use standard key reading techniques of either Node or Electron.

If you're trying to control the non-keyboard parts of a keyboard-like device, it's almost guaranteed the device exposes an alternate usage/usagePage, interface number, or entire interface.  You should be able to see the first two `HID.devices()` as devices with the same VID/PID but different usages/usagePages or interface numbers.  If you're curious about usage pages, check out https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf

bridgewaterrobbie, 2020-04-07:
Thank you Tod. I am wanting the keyboard parts, so in that case I think my best bet is to reconstruct HID-like packets from the standard key reading techniques.

fedecaccia, 2021-02-25:
Have you got a solution? Im facing the same problem using @ledgerhq/hw-transport-node-hid lib to connect with nano ledger device

mcuee, 2021-06-19:
I see that https://github.com/node-hid/node-hid/issues/348 has been closed. What about this #116?

mcuee, 2021-08-31:
@GanZhiXiong Can you give latest git a try?

mcuee, 2021-08-31:
@fedecaccia Have you tried plain hidapi?

closed 2022-05-16 by Youw