libusb/hidapi #118: how to build hidapi on Android project

Chakib-Temal, 2019-11-14:
hi

what's the steps to build and work with the library in android os ? 

before talking about android.mk ? there is no other operation ? 

download the ndk or what exactly ? thank's

z3ntu, 2019-11-14:
Duplicate of #24

Youw, 2019-11-14:
Start with https://developer.android.com/studio/projects/add-native-code

closed 2019-11-14 by Youw

Chakib-Temal, 2019-11-14:
> Start with [Android: Add C and C++ code to your project](https://developer.android.com/studio/projects/add-native-code)

i have already do this (i can execute my c++ in android )

before asking : i used this link https://github.com/libusb/libusb/tree/master/android

for your exemple : what's the next step after adding the c++ to my project
here we are only the android.mk ?
i don't realy understand [24](https://github.com/libusb/hidapi/issues/24)

new project in android use the cmakeList.txt and not Android.mk !

Youw, 2019-11-14:
This is general Android/build system related question.

If you have issues/questions related directly to `hidapi` - we'd be happy to help you as much as we could.

Personally I am not an Android expert, and I wouldn't expect you to search general Android-related help here.

Chakib-Temal, 2019-11-14:
> This is general Android/build system related question.
> 
> If you have issues/questions related directly to `hidapi` - we'd be happy to help you as much as we could.
> 
> Personally I am not an Android expert, and I wouldn't expect you to search general Android-related help here.

no problem thank's 

i tried to do it in my linux
now i have a problem in execution 

`if (hid_init())
        return -1;`

undefined reference to `hid_init'

the include of the file is fine , but i can't use any functions 
did you have an idea ? thank you for yours answers