libusb/hidapi #126: Windows 7 32 bit Problem: Carriage Return every 64 chars?

maxwell130631, 2019-11-28:
I am using hid_read_timeout  read measurement data from a HID device. This works fine under Linux and Windows 10 64 bit. Usually, I do

hid_read_timeout( _handle, readBuf, MAX_STR, static_cast<int>(_recvTimeout->value()*1000) );

and scan the buffer until I find a Carriage Return. In Linux and Windows 64 bit the length of the buffer until carriage return is about 180. However, My software was used on a Windows 7 32 bit machine and there my buffers contain a Carriage Return already after 64 characters. What could be a possible reason?

Youw, 2019-11-28:
Hi @maxwell130631,

Unfortunately you didn't provide enough info for us to even try help you.
- What kind of device you're using?
- What is `MAX_STR`?
- What size is `readBuf` (I'd assume it is `MAX_STR` size, but who knows)?
- What is the nature of the data you're getting from the device? Does make sense, when `\n` appears early?
- How do you search for `\n`?
- Are you sure the device sends data you're expect (and not the one you're actually getting)?

---
A code sample of you read (and `\n` search) routine should also help identify where the problem might be.

mcuee, 2021-06-10:
Close the issue since the OP does not come back. 
To OP: please reopen if provide more info.

closed 2021-06-10 by mcuee