libusb/hidapi #127: macOS: Bluetooth HID devices have the same path

erikmwerner, 2019-12-01:
hidapi returns the same path IOService:/IOResources/IOBluetoothHCIController/AppleBroadcomBluetoothHostController/IOBluetoothDevice/IOBluetoothL2CAPChannel/IOBluetoothHIDDriver

For every Bluetooth HID device connected. Tested with:
Nintendo Switch JoyCon (L)
Nintendo Switch JoyCon (R)
Nintendo Switch ProController
Microsoft Bluetooth Mouse 5000

(all devices have unique product IDs and serial numbers)

As a result, hid_open(vendor_id, product_id, serial_number) does not open the correct device if there is more than one bluetooth device paired since it eventually calls hid_open_path, and all the paths are identical. It looks like it will always try to open the device that was paired first.

I can open the correct device if I store a copy of the IOHIDDeviceRef in hid_device_info and create an IOHIDDevice with the reference rather than looking up the IORegistry entry from the path.

This is on a MacBookPro11,1 running macOS 10.14.6.

Youw, 2019-12-01:
It looks like the same issue as https://github.com/signal11/hidapi/issues/379

@erikmwerner try if you can apply https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5 locally and check if it fixes your issue

It seems like patch by @flirc preserves backward compatibility and adds stability/reliability to `hid_open_path` on macOS, but before submitting it to master, I'd love to head some alternative opinions

erikmwerner, 2019-12-04:
I can confirm that https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5 fixes the issue for me. Thanks for pointing me in the right direction.

As an alternative, the solution that I described above also fixed the issue. I uploaded it at https://github.com/erikmwerner/hidapi/commit/1d65b88edb3c40653c1f06f32f0ffeff4a175f11

Youw, 2019-12-04:
Thanks.

Yes, it seems like erikmwerner@1d65b88 would fix your issue, but having same API for all 4 backends is more a bit more preferable to me.

@jdk do you think you could prepare a PR to master with flirc/hidapi@8d251c3 ?

mcuee, 2021-06-19:
So #127 and #236 have the same fix in https://github.com/flirc/hidapi/commit/8d251c3 but that fix needs some modification to be accepted, right?

Youw, 2021-06-19:
Right. flirc/hidapi@8d251c3 has a leak in the implementation, that's for sure, but the idea is fine.

commit 180d331b (2021-08-14): macOS: use IORegistryEntryGetRegistryEntryID to resolve paths

With some device connection configurations,
the device paths become over 512 bytes (io_string_t max length)
which makes them unusable with current implementation.

Rather than using ServiceRegistry string path, use its ID,
which is uint64_t and easily serializable/deserializable into a string.

Implementation idea by felix.schwarz@iospirit.com
https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5

Fixes: #127, #236.
PR #322 (2021-08-23): macOS: use IORegistryEntryGetRegistryEntryID to resolve paths
With some device connection configurations,
the device paths become over 512 bytes (io_string_t max length)
which makes them unusable with current implementation.

Rather than using ServiceRegistry string path, use its ID,
which is uint64_t and easily serializable/deserializable into a string.

Implementation idea by felix.schwarz@iospirit.com
https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5

Fixes: #127, #236.
commit 4d63a508 (2021-08-23): macOS: use IORegistryEntryGetRegistryEntryID to resolve paths (#322)

With some device connection configurations,
the device paths become over 512 bytes (io_string_t max length)
which makes them unusable with current implementation.

Rather than using ServiceRegistry string path, use its ID,
which is uint64_t and easily serializable/deserializable into a string.

Implementation idea by felix.schwarz@iospirit.com
https://github.com/flirc/hidapi/commit/8d251c3854c3b1877509ab07a623dafc8e803db5

Fixes: #127, #236.
closed 2021-08-23 by Youw