libusb/hidapi #128: Handle GetOverlappedResult blocking forever on read/write

CristianGlavana, 2019-12-03:
Hello,

We were investigating a blocking issue when write command GetOverlappedResult is blocking forever. It is a follow up from https://github.com/signal11/hidapi/issues/88 

This occurs when maximal speed of USB HID is reached, e.g. 1000 frames per second. If target is just sending data at this rate, it is ok. Some frames are still lost due to computer speed, windows management and other USB frames requested by windows checking target status.
Problem cames when communication goes both ways. When sending commands from the computer if line IO_PENDING flag is true, there is not enought bandwidth and GetOverlappedResult is blocking forever. 
We experienced with 64bytes data in/out.

Solution from https://github.com/signal11/hidapi/issues/88 was implemented in hid.c file and sucessfully tested. When ERROR_IO_PENDING is detected write function waits until previous transaction is complete. Also limit maximal transfer frames rate to 1000 (1000 read or 1000 write or 500 read/write)

Here is attached hid.c file: 
[hid.zip](https://github.com/libusb/hidapi/files/3916910/hid.zip)

Youw, 2019-12-03:
Hi @CristianGlavana,

Thank's for opening this ticket.
It seems like signal11/hidapi#88 does suggest a solution (developed by [Alan Ott](https://github.com/signal11)) itself.

Do you feel like preparing a PR to fix this issue?

Youw, 2020-10-14:
Fix applied, currently in master

closed 2020-10-14 by Youw