libusb/hidapi #134: Unable to open device (access denied)

akiller, 2020-01-05:
Hi,

I'm attempting to access my keyboard to configure the RGB lights etc using [DuckyAPI](https://github.com/Latedi/DuckyAPI) that makes use of the original hidapi library.

I have One2 TKL which is different model to the one this project was targeted to but the same manufacturer. I've updated to the correct PID for my device and and am hoping they use the same protocol for all their devices. 

When I try and open my device hid_read_timeout() in hid.c executes `res = ReadFile(dev->device_handle, dev->read_buf, dev->input_report_length, &bytes_read, &dev->ol)` which is returning 0. Checking GetLastError() returns 5 (ERROR_ACCESS_DENIED). I'm running in an elevated administrator process.

Just to confirm I'm getting this same result when I compile the master branch of this project directly, running entirely separate from the DuckyAPI project - jut using the builtin hidtest.exe having updated the vid / pid to mine.

Hopefully you can point me in the right direction as this is far beyond my knowledge set!

This is the output from hidtest, I've filtered it to my device:

```
Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col04#7&1ca74182&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2
  Usage (page): 0x6 (0x1)

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col02#7&1ca74182&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2
  Usage (page): 0x1 (0xc)

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_00#7&28baf2ff&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    0
  Usage (page): 0x6 (0x1)

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_02&col03#7&1ca74182&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      Ducky Keyboard
  Release:      120
  Interface:    2
  Usage (page): 0x2 (0x1)

Device Found
  type: 04d9 0356
  path: \\?\hid#vid_04d9&pid_0356&mi_01#7&4e3b53d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
  serial_number: (null)
  Manufacturer: DuckyChannel International Co., Ltd.
  Product:      CY040356014C0102 EU
  Release:      120
  Interface:    1
  Usage (page): 0x1 (0xff00)

Manufacturer String: DuckyChannel International Co., Ltd.
Product String: Ducky Keyboard
Unable to read serial number string
Serial Number String: (0)
Indexed String 1: Ducky Keyboard
```

Youw, 2020-01-05:
Hi @akiller,

There was a similar discussion earlier: https://github.com/signal11/hidapi/issues/313
Is it your case?

akiller, 2020-01-05:
Hi, thanks for the reply,

No these weren't me. I did see those when I was trying to diagnose the problem but they didn't seem to have any solutions.

Youw, 2020-01-05:
Going over your description:

1.
> just using the hidtest.exe after having updated the vid / pid to my device

I really hope, you do know what you're doing:
Check our readme:
https://github.com/libusb/hidapi#what-does-the-api-look-like
Warning: Only run the code you understand, and only when it conforms to the device spec. **Writing data at random to your HID devices can break them**.

There is no guarantee that writing random data would work at all.

2.
Your device is identified as 5 different logical devices, each has different usage/usage_page.
Simply using `hid_open(vid, pid, NULL);` would end up opening any of them.
Are you sure you have opened the one that should support the API you're going to use?
NOTE: to open a logical device with a specific usage_page, you need to enumerate all of them, find your usage_page, and then open open device by path.

akiller, 2020-01-05:
> I really hope, you do know what you're doing:
> Check our readme:
> https://github.com/libusb/hidapi#what-does-the-api-look-like
> Warning: Only run the code you understand, and only when it conforms to [...]

I like to live dangerously, but seriously, thanks for the warning 🙂.

> Your device is identified as 5 different logical devices, each has different usage/usage_page.
> Simply using `hid_open(vid, pid, NULL);` would end up opening any of them.
> Are you sure you have opened the one that should support the API you're going to use?
> NOTE: to open a logical device with a specific usage_page, you need to enumerate all of them, find your usage_page, and then open the device by path.

Inside hid_open() I hacked in a basic and nasty test to try and open each of the device paths:

```
while (cur_dev) {
	if (cur_dev->vendor_id == vendor_id &&
		cur_dev->product_id == product_id) {
		if (serial_number) {
			if (wcscmp(serial_number, cur_dev->serial_number) == 0) {
				path_to_open = cur_dev->path;
				break;
			}
		}
		else {
			path_to_open = cur_dev->path;
			
			// Hack
			unsigned char bufIn[256];
			memset(bufIn, 0, sizeof(bufIn));
			handle = hid_open_path(path_to_open);
			uint32_t result = hid_read_timeout(handle, bufIn, 65, 100);
			printf_s("%s result=%d\n", cur_dev->path, result);
		}
	}
	cur_dev = cur_dev->next;
}

```

These were the results.

> \\?\hid#vid_04d9&pid_0356&mi_02&col04#7&1ca74182&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd result=-1
> \\?\hid#vid_04d9&pid_0356&mi_02&col01#7&1ca74182&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} result=0
> \\?\hid#vid_04d9&pid_0356&mi_02&col02#7&1ca74182&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030} result=0
> \\?\hid#vid_04d9&pid_0356&mi_00#7&28baf2ff&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd result=-1
> \\?\hid#vid_04d9&pid_0356&mi_02&col03#7&1ca74182&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030} result=-1
> \\?\hid#vid_04d9&pid_0356&mi_01#7&4e3b53d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} result=0

So it looks like either it won't let me open them (result -1) or it times out (0).

Youw, 2020-01-05:
> thanks for the warning

Yeah, there is at least one proven example of broken device: https://github.com/libusb/hidapi/issues/105

> basic and nasty test

even though I'm sure it wouldn't mess with the test you want to perform, but you better not to forget to close the device anyway

> or it times out 

Are you sure there is any data available for read? Device may simply do not send any, and it might be an expected behaviour.

akiller, 2020-01-08:
> even though I'm sure it wouldn't mess with the test you want to perform, but you better not to forget to close the device anyway

Good spot. I tried that but still no luck.

> Are you sure there is any data available for read? Device may simply do not send any, and it might be an expected behaviour.

Hmm possibly, it also seems to fail if I only request a write handle and comment out the read handle.

I just tried opening it with a different library [(Device.net)](https://github.com/MelbourneDeveloper/Device.Net) which uses ReadFile from kernel32.dll looping through all of them as before, this also doesn't seem to work in either read or write mode, so it must be something weird with my device.

I appreciate this is now likely outside the scope of this library but if you have any further ideas that would be great. The software that comes with the keyboard is able to write to it so it clearly must be possible somehow. I did wonder if it might be locking it somehow but I made sure it and its services were closed too.

Here's an output of it from [USBView](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview) if it's of any use:

```Device Descriptor:
bcdUSB:             0x0200
bDeviceClass:         0x00
bDeviceSubClass:      0x00
bDeviceProtocol:      0x00
bMaxPacketSize0:      0x40 (64)
idVendor:           0x04D9 (Holtek Semiconductor, Inc.)
idProduct:          0x0356
bcdDevice:          0x0120
iManufacturer:        0x03
0x0409: "DuckyChannel International Co., Ltd."
iProduct:             0x01
0x0409: "Ducky Keyboard"
iSerialNumber:        0x00
bNumConfigurations:   0x01

ConnectionStatus: DeviceConnected
Current Config Value: 0x01
Device Bus Speed:     Full
Device Address:       0x02
Open Pipes:              4

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0008 (8)
bInterval:            0x01

Endpoint Descriptor:
bEndpointAddress:     0x83  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0040 (64)
bInterval:            0x01

Endpoint Descriptor:
bEndpointAddress:     0x04  OUT
Transfer Type:   Interrupt
wMaxPacketSize:     0x0040 (64)
bInterval:            0x01

Endpoint Descriptor:
bEndpointAddress:     0x82  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0040 (64)
bInterval:            0x01

Configuration Descriptor:
wTotalLength:       0x005B
bNumInterfaces:       0x03
bConfigurationValue:  0x01
iConfiguration:       0x00
bmAttributes:         0xA0 (Bus Powered Remote Wakeup)
MaxPower:             0x32 (100 Ma)

Interface Descriptor:
bInterfaceNumber:     0x00
bAlternateSetting:    0x00
bNumEndpoints:        0x01
bInterfaceClass:      0x03 (HID)
bInterfaceSubClass:   0x01
bInterfaceProtocol:   0x01
iInterface:           0x00

HID Descriptor:
bcdHID:             0x0111
bCountryCode:         0x00
bNumDescriptors:      0x01
bDescriptorType:      0x22
wDescriptorLength:  0x0040

Endpoint Descriptor:
bEndpointAddress:     0x81  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0008 (8)
bInterval:            0x01

Interface Descriptor:
bInterfaceNumber:     0x01
bAlternateSetting:    0x00
bNumEndpoints:        0x02
bInterfaceClass:      0x03 (HID)
bInterfaceSubClass:   0x00
bInterfaceProtocol:   0x00
iInterface:           0x02
0x0409: "CY040356014C0102 EU"

HID Descriptor:
bcdHID:             0x0111
bCountryCode:         0x00
bNumDescriptors:      0x01
bDescriptorType:      0x22
wDescriptorLength:  0x0022

Endpoint Descriptor:
bEndpointAddress:     0x83  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0040 (64)
bInterval:            0x01

Endpoint Descriptor:
bEndpointAddress:     0x04  OUT
Transfer Type:   Interrupt
wMaxPacketSize:     0x0040 (64)
bInterval:            0x01

Interface Descriptor:
bInterfaceNumber:     0x02
bAlternateSetting:    0x00
bNumEndpoints:        0x01
bInterfaceClass:      0x03 (HID)
bInterfaceSubClass:   0x00
bInterfaceProtocol:   0x00
iInterface:           0x00

HID Descriptor:
bcdHID:             0x0111
bCountryCode:         0x00
bNumDescriptors:      0x01
bDescriptorType:      0x22
wDescriptorLength:  0x00CB

Endpoint Descriptor:
bEndpointAddress:     0x82  IN
Transfer Type:   Interrupt
wMaxPacketSize:     0x0040 (64)
bInterval:            0x01
```

mcuee, 2021-01-26:
Basically you have to reverse engineering the protocol. A USB Analyzer can help.

closed 2021-06-10 by mcuee