libusb/hidapi #135: Open non-zero interface of HID keyboard/mouse device on Windows

CalcProgrammer1, 2020-01-12:
I want to use hidapi to open a Corsair K70 RGB keyboard device on Windows and Linux.  The keyboard's latest firmware causes it to expose two interfaces - interface 0 and interface 1.  Both interfaces use the HidUsb driver according to Zadig.  The keyboard interface seems to be interface 0, as if I override the driver in Zadig the keyboard stops sending input to the PC.  Interface 1 seems to be for the RGB control.  I tried using libusb with the WinUSB filter driver and I could control the RGB on interface 1 but it prevented the keyboard from operating as a keyboard.

I also had some code from an old project that used the Windows USB API directly and it had an option to select the interface when opening the device by VID/PID.  The code can be found here:

https://gitlab.com/CalcProgrammer1/KeyboardVisualizer/blob/master/KeyboardVisualizerCommon/UsbDevice.cpp#L43

I changed the GetDeviceHandle arguments to GetDeviceHandle(vendor, product, MI)

If I use this code, I can access the keyboard with this line:

dev.OpenDevice(CORSAIR_RGB_KEYBOARD_VID, CORSAIR_RGB_KEYBOARD_K70_PID, 1)

If I use hidapi, it fails to open the device.  It looks like hidapi's hid_open function isn't finding the second interface or something.  I would prefer using hidapi for both Windows and Linux instead of the Windows-only API directly.

Youw, 2020-01-12:
Hidapi (on Windows) operates on a higher level by using native HID API and knows nothing about USB or its interfaces, so the approach you've used cannot be applied as is with hidapi.

Since the devices on different interfaces are two different logical devices, and those would appear as separate devices if you call `hid_enumerate`.
`hid_device_info->interface_number` is what you're looking for.

Once you've found your device, you can open it with `hid_open_path`.

Youw, 2020-01-12:
P.S.: as for me, `hid_open` is unusable for most (production) cases, except simple examples. But it is great for small POC applications, learning, etc.

CalcProgrammer1, 2020-01-12:
Great!  Thank you!  I hadn't seen any examples using hid_enumerate before and it looks like it does exactly what I wanted it to do.

The new code works:

```
    hid_init();

    info = hid_enumerate(CORSAIR_RGB_KEYBOARD_VID, CORSAIR_RGB_KEYBOARD_K70_PID);

    //Look for Corsair RGB Keyboard, interface 1
    while(info)
    {
        if((info->vendor_id == CORSAIR_RGB_KEYBOARD_VID)
         &&(info->product_id == CORSAIR_RGB_KEYBOARD_K70_PID)
         &&(info->interface_number == 1))
        {
            dev = hid_open_path(info->path);
            break;
        }
        else
        {
            info = info->next;
        }
    }
```

closed 2020-01-12 by CalcProgrammer1