libusb/hidapi #136: 3dConnexion Universal Receiver - Linux OK, Windows and MacOS not

smartavionics, 2020-01-19:
Hi, I'm using hidapi (via cython-hidapi) to access a 3dConnexion Spacemouse. The mouse is supplied with a USB Universal Receiver. On Linux, I can open this device and read from it quite happily. But on both Windows and MacOS, although I can open the device, no data is ever returned from a read.
I am opening using the path returned from enumeration.
Also, on both Windows and MacOS, if I plug the same mouse into the computers using a USB cable, they both get mouse data OK (of course, they are then using a different vid:pid).
I don't really know that much about USB, HID, etc. so I am hoping someone can suggest where the problem lies.

Here's what lsusb -v on Linux says about the device...

Thanks.

```
Bus 001 Device 077: ID 256f:c652  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        32
  idVendor           0x256f 
  idProduct          0xc652 
  bcdDevice            1.02
  iManufacturer           1 3Dconnexion
  iProduct                2 3Dconnexion Universal Receiver
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          134
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 3DX01.02_B0003
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               98mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     448
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     129
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     129
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     129
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        4
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     129
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

Youw, 2020-01-19:
Can you confirm, that you're opening the same interface (`hid_device_info->interface_number`) on each platform?

Also, please check is it related to some known limitations (at least on Windows): https://github.com/libusb/hidapi/issues/61.

smartavionics, 2020-01-19:
> Can you confirm, that you're opening the same interface (hid_device_info->interface_number) on each platform?

I believe I did. In fact, I tried explicitly specifying the interface number and none worked.

> Also, please check is it related to some known limitations (at least on Windows): #61.

Given that the same spacemouse works quite happily when plugged in using the direct USB cable instead of the Universal Receiver, I don't think it's the same problem. But, perhaps it is?

Youw, 2020-01-19:
> I believe I did. In fact, I tried explicitly specifying the interface number and none worked.

I'd suggest you check what is "correct" interface number on Linux, where you can get it to work, so you know exactly which one.

> I don't think it's the same problem. But, perhaps it is?

Maybe not. Maybe it is related, but not directly.
Windows/macOS native HID API has its limitations (for security reasons) when it comes to Keyboard/Mouse devices, you're not the first one who has issues with it.

Unfortunately I can't help you much more. Maybe someone else from the community will.

smartavionics, 2020-01-19:
> I'd suggest you check what is "correct" interface number on Linux, where you can get it to work, so you know exactly which one.

Good idea, I will do that.

> Unfortunately I can't help you much more. Maybe someone else from the community will.

Thanks anyway for your assistance. Every little helps!

smartavionics, 2020-01-20:
I found the problem, on Windows (and probably MacOS too), I needed to match the usage page and usage to 1 and 8 respectively.

closed 2020-01-20 by smartavionics