libusb/hidapi #137: Unable to use hid_open in android

xmutzlq, 2020-01-20:
Hi,
thank you for you solution to https://github.com/libusb/hidapi/issues/24， I have build android .so lib。 
first，I have use "hid_init()"，and the result is 0;
second，I have got "hid_device_info" from the method "hid_enumerate(0x0, 0x0)" , and the result is (
  info = HidDeviceInfo{path='0003:0003:00', vendor_id=D197, product_id=E0A097}
  info = HidDeviceInfo{path='0003:000b:00', vendor_id=D1B1, product_id=E78083}
  info = HidDeviceInfo{path='0003:000b:01', vendor_id=D1B1, product_id=E78083}
) 
third，When i use hid_open(vendor_id, product_id, NULL) from hidapi.h（vendor_id and product_id from hid_device_info），but it is not working, and log result is "unable to open device"。
then，i was tested "hid_open" in hid.c，flag "handle = hid_open_path(path_to_open);"，go into this method，i do not know why "hid_open_path" return NULL。
can you help me?  thanks

tuxun, 2020-01-20:
Hi,
Does android ask you the permission to use the device when you start your app?
You can dig in your system logs running "logcat" as admin in a terminal, that may help you to understand whats wrong.

xmutzlq, 2020-01-21:
Hi @tuxun 
thank you for your answer.

i have added usb permission, but hid_open is still no working
and i hava a question about if vendor_id and product_id those two params are get from hid_enumerate(0x0, 0x0), and put those params to hid_open(vendor_id, product_id, NULL) ?

because the method hid_open in hid.c is use while condition and check "cur_dev->vendor_id == vendor_id && cur_dev->product_id == product_id". 
```
hid_device * hid_open(unsigned short vendor_id, unsigned short product_id, const wchar_t *serial_number)
{
	struct hid_device_info *devs, *cur_dev;
	const char *path_to_open = NULL;
	hid_device *handle = NULL;
	devs = hid_enumerate(vendor_id, product_id);
	cur_dev = devs;
	**while (cur_dev) {
		if (cur_dev->vendor_id == vendor_id &&
		    cur_dev->product_id == product_id) {**
			if (serial_number) {
				if (cur_dev->serial_number &&
				    wcscmp(serial_number, cur_dev->serial_number) == 0) {
					path_to_open = cur_dev->path;
					break;
				}
			}
			else {
				path_to_open = cur_dev->path;
				break;
			}
		}
		cur_dev = cur_dev->next;
	}
        if (path_to_open) {
		/* Open the device */
		handle = hid_open_path(path_to_open);
	}
	hid_free_enumeration(devs);
	return handle;
}
```
i thank the problem is hid_open_path(const char *path) , but i have no idea how to test this method,  can you help me ? thanks.

tuxun, 2020-01-21:
Hi,

I think i understand. your path is wrongly returned by enum_device(). 
/dev/bus/usb/003/003 is a path, but "0003:0003:00" isnt a path. you need to build a string leading to usb device file descriptor. 

You should check the file exist before open it :)

Youw, 2020-01-21:
Hi @tuxun,
Thanks for helping out. Please note, that your last answer isn't correct for this case and makes a confusion:
On Android hidapi is using libusb backend, and it doesn't operate with file-paths, but instead with USB entities, like "`bus_number`:`device_address`:`interface_number`", which makes a path, and it is a correct representation.

@xmutzlq 
1) I assume you're using a build from latest master. Can you confirm?

2) 
> log result is "unable to open device"

Is it from hidtest here: https://github.com/libusb/hidapi/blob/ca1a2d6efae8d372587f4c13f60632916681d408/hidtest/test.c#L75 ?
If so, it doesn't exactly describe what went wrong.

> i do not know why "hid_open_path" return NULL

Would you be able to trace with a debugger, to catch where exactly `hid_open_path` fails?

What would be helpful to know: is it goes as far as here: https://github.com/libusb/hidapi/blob/ca1a2d6efae8d372587f4c13f60632916681d408/libusb/hid.c#L906
or not.

tuxun, 2020-07-26:
coud we have an hid_open_path function using file path and another one using the other kind of path? IMHO its confusing ATM.
The hid_open_path function both take filepath and USB path?

Youw, 2020-07-26:
`hid_open_path` takes a platform-specific path to a device (which happens to be a file-path only for linux/hidraw backend)

strictly speaking, it is intended to receive _only_ a `path` field of `hid_device_info`

it is not designed to be human-readable, or manually constructible

mcuee, 2021-01-26:
https://github.com/libusb/libusb/pull/830
This example might help.

closed 2021-06-10 by mcuee