libusb/hidapi #138: macOS Catalina testgui segfault libdyld.dylib stack_not_16_byte_aligned_error

mtnbrit, 2020-01-22:
I built the hidapi and testgui on macOS Catalina, when I launch the TestGUI app, it halts with the following debug info

Process:               hidapi-testgui [99735]
Path:                  /Users/USER/Downloads/*/TestGUI.app/Contents/MacOS/hidapi-testgui
Identifier:            us.signal11.hidtestgui
Version:               1.0
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           hidapi-testgui [99735]
User ID:               501

Date/Time:             2020-01-21 16:29:47.519 -0800
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Bridge OS Version:     3.0 (14Y906)
Anonymous UUID:        F53BEB53-80E0-55AC-99A8-F1D3BFFB00E4

Sleep/Wake UUID:       CDAFBF1B-7E33-46E1-962F-31426BDCCAC8

Time Awake Since Boot: 2800000 seconds
Time Since Wake:       190000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [99735]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libdyld.dylib                 	0x00007fff691872be stack_not_16_byte_aligned_error + 0
1   ???                           	0x0000600003598a50 0 + 105553172466256

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000001438  rbx: 0x000000000000008c  rcx: 0x0000000000000070  rdx: 0x0000000000000001
  rdi: 0x0000600003598a50  rsi: 0x000000000000000c  rbp: 0x0000600003598a50  rsp: 0x00007ffee5da11a8
   r8: 0x0000000000000195   r9: 0x0000000000000070  r10: 0x00007fbd948026e8  r11: 0x000000010a044030
  r12: 0x0000000000000006  r13: 0x0000000000000070  r14: 0x0000000000000018  r15: 0x0000000000000400
  rip: 0x00007fff691872be  rfl: 0x0000000000010202  cr2: 0x000000010b28b908
  
Logical CPU:     4
Error Code:      0x00000000
Trap Number:     13

I found some other folks having the similar issue in other projects and they suggest that adding    -fno-stack-check to CFLAGS is a workaround, but I dont know where to add it.

Youw, 2020-01-22:
> but I dont know where to add it

Try here: https://github.com/libusb/hidapi/blob/master/testgui/Makefile.mac#L17

mtnbrit, 2020-01-22:
Yep, I had actually tried that. The line now looks like

CFLAGS=-I../hidapi -Wall -g -c `fox-config --cflags` -fno-stack-check

And I did make clean make make install in that directory. It still fails with the same error. Am I applying it correctly? How to verify what CLFAGs are used by the compiler?

Youw, 2020-01-22:
have you been explicitly specifying the makefile, i.e.: `make -f Makefile.mac` ?

mtnbrit, 2020-01-22:
Ah I was not, now I have done it and I see the g++ cflags are correct. Unfortunately it still hangs up in the same way.

Youw, 2020-01-22:
It likely means that you need to rebuild Fox Toolkit with a specified flag, which is out of this conversation scope

instead, I'd suggest try using `hidtest` console app, or https://github.com/todbot/hidapitester

hstriepe, 2020-08-02:
I built with Xcode 10 under Mojave and all is fine with the result under Catalina.

todbot, 2020-08-02:
Hi @hstriepe, would you mind sharing your Xcode project? (and the process by which you're aquiring/compiling fox-toolkit)  Perhaps we could add it to the `mac` directory to help Mac folk out.

hstriepe, 2020-08-02:
@todbot I built from command line per instructions:
1. Boot into a Mojave VM
2. Xcode10 was installed. I used Xcode-select to switch to that.
3. install xquartsx and fox via (home)brew
4. Clone hidapi
5. cd to hidapi/Mac
6. make -f Makefile-manual
7. cd to hidapi testgui
8. make -f Makefile-manual
Copy the testgui app back to Catalina, which also has xquartzx installed. Launch quartz before testgui

todbot, 2020-08-02:
Ah okay so you are compiling in Mojave, not Catalina.

Youw, 2020-08-02:
one day I'll find time and replace FOX-based UI with Qt-based one, it will make it so much easier

hstriepe, 2020-08-03:
We’re using qt, not always a panacea.

Cheers,
-- Harald
+ ❤️

mcuee, 2020-12-04:
I am trying out the test gui with Mac Mini M1 (Apple Silicon, Big Sur) and the app seems to run fine. I built using Homebrew (x86-64,not native M1). Install fox under Homebrew before the build, and then install xquartz before running the test-gui app.

However, sending invalid requests will cause the system to hang (macOS crash and automatically reboot).

mcuee, 2021-06-13:
Build log under my Mac Mini M1.

```
hidapi/buildarm64 on  master [?] took 39s ❯ ../configure --enable-testgui
checking for a BSD-compatible install... /opt/homebrew/bin/ginstall -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /opt/homebrew/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... aarch64-apple-darwin20.5.0
checking host system type... aarch64-apple-darwin20.5.0
checking how to print strings... printf
checking for a sed that does not truncate output... /opt/homebrew/bin/gsed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert aarch64-apple-darwin20.5.0 file names to aarch64-apple-darwin20.5.0 format... func_convert_file_noop
checking how to convert aarch64-apple-darwin20.5.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
[... 60 lines cut ...]
config.status: creating testgui/Makefile
config.status: creating windows/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
hidapi/buildarm64 on  master [?] took 3s ❯ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in mac
make[2]: Nothing to be done for `all'.
Making all in hidtest
make[2]: Nothing to be done for `all'.
Making all in testgui
  CXX      test.o
  OBJC     mac_support_cocoa.o
  CXXLD    hidapi-testgui
ld: warning: ignoring file /usr/local/lib/libjpeg.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/lib/libtiff.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
../../testgui/copy_to_bundle.sh
libtool: warning: '../mac/libhidapi.la' has not been installed in '/usr/local/lib'
libtool: install: cp .libs/hidapi-testgui /Users/mcuee/build/hidapi/hidapi/buildarm64/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui
bundling /opt/homebrew/opt/fox/lib/libFOX-1.6.0.dylib because it's in the list.
/Library/Developer/CommandLineTools/usr/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/mcuee/build/hidapi/hidapi/buildarm64/testgui/TestGUI.app/Contents/MacOS/libFOX-1.6.0.dylib
bundling @executable_path/libFOX-1.6.0.dylib because it's in the list.
```

mcuee, 2021-06-13:
It runs fine. I will close the issue for now. Please reopen if you can reproduce the issue with latest homebrew version of fox and hidapi.

mcuee, 2021-06-13:
x86_64 build is also fine.

```
hidapi/buildx64 on  master [?] ❯ ../configure --enable-testgui
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... ../install-sh -c -d
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... x86_64-apple-darwin20.5.0
checking host system type... x86_64-apple-darwin20.5.0
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert x86_64-apple-darwin20.5.0 file names to x86_64-apple-darwin20.5.0 format... func_convert_file_noop
checking how to convert x86_64-apple-darwin20.5.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
[... 62 lines cut ...]
config.status: creating mac/Makefile
config.status: creating testgui/Makefile
config.status: creating windows/Makefile
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
hidapi/buildx64 on  master [?] took 11s ❯ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in mac
  CC       hid.lo
  CCLD     libhidapi.la
Making all in hidtest
  CC       test.o
  CCLD     hidtest
Making all in testgui
  CXX      test.o
  OBJC     mac_support_cocoa.o
  CXXLD    hidapi-testgui
../../testgui/copy_to_bundle.sh
libtool: warning: '../mac/libhidapi.la' has not been installed in '/usr/local/lib'
libtool: install: cp .libs/hidapi-testgui /Users/mcuee/build/hidapi/hidapi/buildx64/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui
bundling /usr/local/opt/fox/lib/libFOX-1.6.0.dylib because it's in the list.
bundling @executable_path/libFOX-1.6.0.dylib because it's in the list.
make[2]: Nothing to be done for `all-am'.
```

closed 2021-06-13 by mcuee

mcuee, 2021-08-28:
https://github.com/mcuee/hidapi/releases
x64 build binary, tested under my Mac Mini M1.

mcuee, 2021-08-29:
The above method will build a working hidapi-testgui under ARM64 but the TestGUI.app will not work under ARM64.

x86_64 version of TestGUI.app will work fine.

More discussions here:
https://github.com/libusb/hidapi/issues/151

mcuee, 2021-08-29:
[image]