libusb/hidapi #140: Add official support for the Haiku operating system

alwayslivid, 2020-01-29:
The [Haiku](https://haiku-os.org) operating system already has a [port](https://github.com/haikuports/haikuports/tree/0d782eba6738395cc7a73e300ef29f1cc27cbb0f/dev-libs/hidapi) that makes `hidapi` successfully compile on the operating system (some software depends on it as well), but the way it gets compiled is through additional patches that add the necessary configurations in the build system.

I think that it would be a better idea if the source got integrated into the upstream tree. I attempted to [update the recipe from a version from 2016 to 0.9.5](https://github.com/haikuports/haikuports/pull/4660), but it was later brought up that another contributor had already made a [tree with the necessary changes](https://github.com/Begasus/hidapi/tree/haiku).

Before proceeding with a pull request (after getting some fine details done), I'd like to ask whether it would be okay before doing so.

z3ntu, 2020-01-29:
I do have a pull request open (#30) but last time I've tried to get hidapi working on Haiku, there was some issue with libusb on Haiku - as described in the PR.

Youw, 2020-01-29:
LD;DR: PRs are welcome.

---
You're not the first one to make some effort in this direction, see https://github.com/libusb/hidapi/pull/30.

Although @z3ntu haven't finished it yet, or not confident to suggest to merge the PR as is, I think you may combine the efforts or suggest your own PR.

Personally I think supporting Haiku of the box isn't a bad thing, especially if there already is a community of users/someone willing to help with its support.

z3ntu, 2020-02-13:
Merged now.

closed 2020-02-13 by Youw