libusb/hidapi #142: Linux: Hanging inside hid_close()

vozhyk-com, 2020-02-01:
For catching this hanging need just open device by hid_open() or hid_open_path() and close it by hid_close(), see simple example:

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include "hidapi.h"

int main(int argc, char* argv[])
{
     hid_device *handle;
     struct hid_device_info *devs, *cur_dev;
   
     if (hid_init())
          return -1; 
     int i;
     for (i =0 ; i < 10000; i++) { 
          handle = hid_open(0x22c9, 0x0901, NULL);
          if (!handle) {
               printf("unable to open device\n");
               return 1;
          }
          printf("iteration: %d, going to call hid_close() \n", i); 
          hid_close(handle);
          printf("iteration: %d, hid_close() done \n", i); 
     }   
     /* Free static HIDAPI objects. */
     hid_exit();
     return 0;
}

Youw, 2020-09-20:
I wasn't able to reproduce the original issue, presumably because I don't have a device that would allow me to do so.
I believe there is a race condition, that when hit, may lead to deadlock, when trying to close the device, in case if device submits transfers/interruptions frequently, and device is being closed exactly during successful handling of such transfer.

The fix suggested in #143 kind-of fixes the issue. Kind-of, because it constantly tries not to hit a deadlock, but a race condition is still there, it is a root cause, and we better fix a root cause, not symptoms.

@vozhyk-com would you be able to check a patch I prepared, with your device? https://github.com/libusb/hidapi/tree/libusb-hang

vozhyk-com, 2020-09-26:
I confirm that the patch from branch libusb-hang solves #142 issue. I can not reproduce it with the patch, but easy can reproduce without.

Youw, 2020-09-26:
That's awesome news. As I hoped, the root issue found and fixed.

PR #189 (2020-10-14): libusb: fix race condition on device close
Read callback may fire itself on its own even after its been
requested to stop and exactly before the calling code
waits for its completion in indefinite loop.

Explicitly preventing re-fireing the submission loop,
fixes the issue.

Fixes: #142.
commit d2c3a986 (2020-10-14): libusb: fix race condition on device close (#189)

Read callback may fire itself on its own even after its been
requested to stop and exactly before the calling code
waits for its completion in indefinite loop.

Explicitly preventing re-fireing the submission loop,
fixes the issue.

Fixes: #142.
closed 2020-10-14 by Youw