libusb/hidapi #144: crash on device unplug on macOS Catalina (10.15)

Youw, 2020-02-04:
Typical usage scenario (simplified):
```c
hid_init();
dev = hid_open(...);
// use the device
// ...

// device physically disconnected somewhere here

// ...
// all further hid_* funcitons are gracefully failing

hid_close(dev); // ok here

hid_exit(); // <-- macOS Catalina - crash here
```

The cause (in `hid_close`):
https://github.com/libusb/hidapi/blob/65d22a9c3124d9fad02cc064d734fef84a575202/mac/hid.c#L1122-L1124
+
https://github.com/libusb/hidapi/blob/65d22a9c3124d9fad02cc064d734fef84a575202/mac/hid.c#L1097-L1104

The original comment from Alan:
https://github.com/libusb/hidapi/blob/65d22a9c3124d9fad02cc064d734fef84a575202/mac/hid.c#L1119-L1121

> it's been unplugged, then calling 
 	   IOHIDDeviceClose() will crash

Apparently that was the case for the version of macOS, where this was originally developed.
In Catalina (10.15) update - it is the opposite case: if we skip closing the device, the app crashes on `hid_exit()` (`IOHIDManagerClose`).

---
Ideally, of course, we should always close all handles, and avoid any resource leaks. Unfortunately, cannot do it unconditionally as some users on older systems might experience crashes as the original author.

---
Currently I have access only to macOS Catalina and cannot check the behaviour on other systems.
Would be great to check how it behaves on older systems, at least up(down) to 10.10.

vitalys, 2020-09-17:
@Youw I'd like to help you with this issue, but I can't reproduce it. 

What I tried:
- [Native Instruments Kontrol F1](https://www.native-instruments.com/en/products/traktor/dj-controllers/traktor-kontrol-f1/) device 
- Test app that writes to device (to highlight buttons/segment indicators)
- My test app doesn't read for the device (is this an issue?)
- I used hidapi from `master` branch

What my app does:
1) `hid_init`
2) `hid_open`
3) `hid_write`, until Ctrl-C is pressed (during this step I log if device is active or unplugged)
4) `hid_close`
5) `hid_exit`
6) Additional `printf` before exiting 

As result I can see final `printf` output, so my app doesn't crash
I tested using: 
- macOS 10.15.6 (Macbook Pro)
- macOS 10.15.1 (VirtualBox)
- macOS 11.0 Beta 5 (VirtualBox)

Later today I'm going to try with macOS 11.0 Beta 6 using Macbook Pro

Do you have any specific scenario to reproduce this crash?

Youw, 2020-09-17:
Try next:
1) `hid_init`
2) `hid_open`
3) **Physically unplug the device**
4) `hid_close`
5) `hid_exit`

vitalys, 2020-09-17:
I've unplugged my device during write cycle, but it doesn't crash at the end
Let me simplify my test app and share it with you

vitalys, 2020-09-17:
There is simplified test code:
```
    hid_device *handle;
    hid_init();
    handle = hid_open(0x17cc, 0x1120, NULL);

    printf("\nUnplug your device and press any key\n");
    getchar();

    hid_close(handle);
    hid_exit();

    printf("\nBye!\n");
```

It doesn't crash, I can see `Bye!` in my console

vitalys, 2020-09-18:
@Youw Is my test correct? Should I try other macOS versions?

Youw, 2020-09-18:
It looks correct.
Its been a while since I tested this last time. I may have a different environment, when I did the test, like run all this in a separate thread, and had a UI windows in a main thread.

vitalys, 2020-09-18:
I've just tried test UI app and made all calls to hidapi from separate queue (and separate thread)
It doesn't crash.

I tried to force call `IOHIDDeviceClose` in `hid_close`:
```
// if (!dev->disconnected) { 
 	IOHIDDeviceClose(dev->device_handle, kIOHIDOptionsTypeSeizeDevice); 
// } 
```

In this case app crashed in `hid_exit`

Youw, 2020-09-18:
This is interesting, let me check on my side.
Would you be able to check on older version of macOS? (10.10-10-14) ?

vitalys, 2020-09-18:
> Would you be able to check on older version of macOS? (10.10-10-14) ?
Yes. That's why I proposed my help

Youw, 2020-09-19:
> I tried to force call IOHIDDeviceClose in hid_close

NOTE: if you do so, you also need to force call `IOHIDDeviceRegisterInputReportCallback`
```cpp
	/*if (!dev->disconnected)*/ {
		IOHIDDeviceRegisterInputReportCallback( /* ... */ );
        }
```
a few lines before that

Youw, 2020-09-19:
Apparently, I cannot reproduce the original crash in simple environments (e.g. a console application, or an app with just a working thread.
The project, where I can reproduce it, uses a lot of stuff: it opens hid device (hidapi) + raw usb device (libusb) + UVC device (IOKit Framework) -those are parts of the single composite device.
Maybe a combination of all of them causes the issue I see.

commit 02a57ae5 (2020-09-20): Don't leak device handle in macOS 10.10 or newer

In one of the early versions of macOS, when you try to close the device
with IOHIDDeviceClose() that is being physically disconnected.
Starting with some version of macOS, this crash bug was fixed,
and starting with macSO 10.15 the opposite effect took place:
in some environments crash happenes if IOHIDDeviceClose() is _not_ called.

This patch is to keep a workaround for old versions of macOS,
and don't have a leak in new/tested environments.

Fixes: #144.
commit 060f39a2 (2020-09-20): Don't leak device handle in macOS 10.10 or newer

In one of the early versions of macOS, when you try to close the device
with IOHIDDeviceClose() that is being physically disconnected.
Starting with some version of macOS, this crash bug was fixed,
and starting with macSO 10.15 the opposite effect took place:
in some environments crash happens if IOHIDDeviceClose() is _not_ called.

This patch is to keep a workaround for old versions of macOS,
and don't have a leak in new/tested environments.

Fixes: #144.
commit 8b37a8be (2020-09-20): Don't leak device handle in macOS 10.11 or newer

In one of the early versions of macOS, when you try to close the device
with IOHIDDeviceClose() that is being physically disconnected.
Starting with some version of macOS, this crash bug was fixed,
and starting with macSO 10.15 the opposite effect took place:
in some environments crash happens if IOHIDDeviceClose() is _not_ called.

This patch is to keep a workaround for old versions of macOS,
and don't have a leak in new/tested environments.

Fixes: #144.
commit 914817a0 (2020-09-26): Don't leak device handle in macOS 10.10 or newer

In one of the early versions of macOS, when you try to close the device
with IOHIDDeviceClose() that is being physically disconnected.
Starting with some version of macOS, this crash bug was fixed,
and starting with macSO 10.15 the opposite effect took place:
in some environments crash happens if IOHIDDeviceClose() is _not_ called.

This patch is to keep a workaround for old versions of macOS,
and don't have a leak in new/tested environments.

Fixes: #144.
closed 2020-09-29 by Youw