libusb/hidapi #147: Linux : hidraw -> 64 bytes report

giellamoswhard, 2020-02-11:
I'm trying to use hid api to interface a custom board with the a x64 Ubuntu 19.10 Linux System. I'm using hidapi-0.9.0 as per distro repository.

In my device I've configured a report descriptor of 64 bytes.
What I've seen is that if from the board I send up 63 bytes ( or more than 65) hid_read() works flawlessly,  but if a send 64 bytes it still doesn't read anything until I send almost another byte then I get all the 65 bytes correct.

If I use the libusb backend i always get the 64 bytes (splitted in 16 bytes chunks).

giellamoswhard, 2020-02-11:
It works the same way at 16 - 32 - 48 bytes

Youw, 2020-02-11:
Could you provide a (simplified) piece of code? The part where you're trying to read/write data, exact buffer sizes, your expectations and the actual results step-by-step?

Just so we don't misunderstand your description.

giellamoswhard, 2020-02-12:
Hi, My code for reading is

```
	unsigned char buf[256] = {};
	while (1) {
                int  res = 0;
		while (res == 0) {
			res = hid_read(handle, buf, 128);
			if (res < 0) {
				printf("Unable to read()\n");
				continue;
			}
			printf("Data read [%d]:\n", res);
			// Print out the returned buffer.
			for (i = 0; i < res; i++)
				printf("%02hhx ", buf[i]);
			printf("\n");
		}
	}
```
I sent 2 bytes via a report my test program prints (as expected)

> Data read [2]:
> 01 09
> 

THen i send a 64 byte full frame and the test program doesn't print anything

The i send other two bytes and i get (the data inside are all the 66 I've sent)

> Data read [66]:
> 01 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38
> 39 3a 3b 3c 3d 3e 3f 40 41 42 01 09

giellamoswhard, 2020-02-12:
If i build it with the **libusb** and send 64 bytes backend i get 

> Data read [16]:
> 01 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 10 11 12
> Data read [16]:
> 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22
> Data read [16]:
> 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32
> Data read [16]:
> 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42

But if get it  I need to put framing inside to reconstruct it, I think

Youw, 2020-02-12:
as any other pipe-like entity - I don't think `hid_read` guarantees you to receive the data in the same chunks size as it was sent

so I'd suggest you should expect both 16 or 66 bytes of actual data, and handle it on the application side

---
as for hang: what happens, when you try: `hid_read(handle, buf, 64);` (pass buffer size same as the actual sent data) ?

the reason is likely the fact, the actual USB data chunk size is 16 bytes, and when you send data in full chunks, the OS receives them like: [full chunk][full chunk][full chunk][full chunk]...

when not, OS eventually gets: [15 bytes chunk][end data chunk]
hid_read would return in several conditions:
- all the data that was requested has been received;
- no more data to read (and return actual data read);

it appears, that hidraw backend (implemented as part of a linux core) won't recognize "no more data" in case, if there was no [end data chunk]

Youw, 2020-02-12:
my point is - I don't think thin it is a hidapi issue (it cannot be - the implementation is trivial: https://github.com/libusb/hidapi/blob/master/linux/hid.c#L744)

instead, try to adapt you application to the specifics of the underlying environment

giellamoswhard, 2020-02-12:
@Youw `hid_read(handle, buf, 64);` it's the same.

So ok, I was on the impression that hid_read was not a "pipe like interface" given all the burden of the building the descriptor but that said it's ok for me (I started thinking of use a CDC profile)

closed 2020-02-12 by giellamoswhard

Youw, 2020-02-12:
> I was on the impression that hid_read was not a "pipe like interface"

Take a look at `hid_get_input_report` (requires the device to support input reports / descriptors)
it is not available for `hidraw` (only macOS, Windows and libusb), but more reliable in terms of receiving fixed size data chunks