libusb/hidapi #149: Accept new maintainer

Youw, 2020-02-24:
I suggest we add one more team member to libusb/hidapi team: @z3ntu
He has been one of the most active contributors since the beginning of libusb/hidapi fork. The project itself and entire community should benefit from having him as an official maintainer.

The team has already discussed this internally, and there was no votes against.

@z3ntu if you willing to join the team - I'll be happy to send you the invitation.

z3ntu, 2020-02-24:
While I do feel flattered that I get offered this position, I'm afraid I already have too many responsibilities on my shoulders with different projects and I already feel that I don't spend enough time on some projects (e.g. my PR review queue in openrazer counts 10 at the moment). 

The great thing about contributing to hidapi for me is that I don't have any commitment to respond to any comments where I don't have the expertise to add something - but as you might have noticed (and because I have the repository watched) I respond to quite a few comments and I definitely plan to do so in the future as I think the project is really great and useful!

Another reason is that my project using hidapi (https://github.com/z3ntu/razer_test) is a bit stale at the moment, so I don't really actively use hidapi at the moment :/

And now because I already write some text, I want to quickly link a hidapi wrapper for Kotlin Native that I've created around two months ago: https://git.sr.ht/~z3ntu/kotlin-native-hidapi (I'll probably move it to GitHub and make it public (not unlisted) when once it gets a bit more fleshed out)

Youw, 2020-02-24:
That's more than fair.

In any case - your opinion and help is always welcome :star2:

I think we'll keep an offer implicitly active, in case if you change your mind :slightly_smiling_face:.

closed 2020-02-24 by Youw