libusb/hidapi #151: MacOS testgui make error

84Park, 2020-03-02:
Hardware: MacBook Pro, MacOS 10.15
Software: fox-toolkit installed with MacPorts
("./configure && make && make install" of fox toolkit on Mac fails with several files not found)

testgui % **gmake -f Makefile-manual**

g++ -I../hidapi -Wall -g -c `fox-config --cflags` test.cpp -o test.o
g++ -I../hidapi -Wall -g -c `fox-config --cflags` -x objective-c++ mac_support_cocoa.m -o mac_support_cocoa.o
rm -Rf TestGUI.app
mkdir -p TestGUI.app
cp -R TestGUI.app.in/ TestGUI.app
g++ -Wall -g ../mac/hid.o test.o mac_support_cocoa.o `fox-config --libs` -framework IOKit -framework CoreFoundation -framework Cocoa -L/usr/X11R6/lib -o hidapi-testgui
**ld: warning: directory not found for option '-L/usr/X11R6/lib'**
./copy_to_bundle.sh
application/x-mach-binary; charset=binary
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/objdump: 
**error: '/Users/xxx/Documents/Arduino/hidapi/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui': Not a directory**
#cp TestGUI.app/Contents/MacOS/hidapi-testgui  TestGUI.app/Contents/MacOS/tg
#cp start.sh TestGUI.app/Contents/MacOS/hidapi-testgui

84Park, 2020-03-02:
The unix executable file that is built, "hidapi-testgui", does run by double click on file in Finder or by using terminal command line "./hidapi-testgui"

hidapi-testgui will find and open a device and send and receive reports.

Youw, 2020-03-03:
I think you're the first user of this GUI app for a long time (on macOS)

Can't promise me or anyone from the team is going to check on it any time soon

Feel free to fixing it yourself and opening a PR - help is welcome

z3ntu, 2020-03-05:
[image]
Works fine for me with the autotools build process (`fox` is installed via homebrew as far as I can tell)

But it looks like the manual Makefile is indeed broken

Youw, 2020-03-05:
thanks @z3ntu, that's awesome you've checked it

we're ready to merge #152 

@84Park would you care to check is it fixes the issue in your environment?

84Park, 2020-03-08:
Thank you. I modified copy_to_bundle.sh with "mkdir -p $EXEPATH" and that removed the error reported in #151 (error: '/Users/xxx/Documents/Arduino/hidapi/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui': Not a directory)

Good job!

closed 2020-03-08 by Youw

mcuee, 2021-08-28:
https://github.com/mcuee/hidapi/releases
Here is a x64 binary snapshot tested under my Mac Mini M1 (with Rosetta 2). I am not so sure how to build the native ARM64 binary as the bundle seems to be a bit off.

todbot, 2021-08-28:
The way I've been building for M1 (arm64) is to build a "fat binary" that contains both executables. It's pretty easy by passing in two `-arch` flags to CFLAGS. Here's how `hidapitester` does it:
https://github.com/todbot/hidapitester/commit/eb8bae29a44e2f38563df6164f0019e868d6fa3a

The result is checkable with `file`:
```sh
% file hidapitester
hidapitester: Mach-O universal binary with 2 architectures: [x86_64:Mach-O 64-bit executable x86_64] [arm64:Mach-O 64-bit executable arm64]
hidapitester (for architecture x86_64):	Mach-O 64-bit executable x86_64
hidapitester (for architecture arm64):	Mach-O 64-bit executable arm64
```

(This was just added. I'm using this technique on a few other projects with success. I clearly need to fix the hidapitester CI now :-)

mcuee, 2021-08-29:
Nice tip. I am using Homebrew and the auto-tools to build test-gui (it is not suppored by CMake). The issue is in the bundling portion that it does not know to bundle the right ARM64 version (/opt/homebrew/lib) of libjpeg.dylib and libtiff.dylib.

```
hidapi/build_autotool on  master [?] ❯ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in mac
  CC       hid.lo
  CCLD     libhidapi.la
Making all in hidtest
  CC       test.o
  CCLD     hidtest
Making all in testgui
  CXX      test.o
  OBJC     mac_support_cocoa.o
  CXXLD    hidapi-testgui
ld: warning: ignoring file /usr/local/lib/libjpeg.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/lib/libtiff.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
../../testgui/copy_to_bundle.sh
libtool: warning: '../mac/libhidapi.la' has not been installed in '/usr/local/lib'
libtool: install: cp .libs/hidapi-testgui /Users/mcuee/build/hidapi/hidapi/build_autotool/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui
bundling /opt/homebrew/opt/fox/lib/libFOX-1.6.0.dylib because it's in the list.
/Library/Developer/CommandLineTools/usr/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/mcuee/build/hidapi/hidapi/build_autotool/testgui/TestGUI.app/Contents/MacOS/libFOX-1.6.0.dylib
bundling @executable_path/libFOX-1.6.0.dylib because it's in the list.
make[2]: Nothing to be done for `all-am'.
```

mcuee, 2021-08-29:
Even if I copy the necessary dylib files to the TestGUI.app/Contents/MacOS folder to be in the same directory with hidapi-testgui, TestGui.app will still crash as it still wants to find /usr/local/lib files which are of x64 arch.

```
Process:               hidapi-testgui [2795]
Path:                  /Users/USER/*/TestGUI.app/Contents/MacOS/hidapi-testgui
Identifier:            us.signal11.hidtestgui
Version:               1.0
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           hidapi-testgui [2795]
User ID:               501

Date/Time:             2021-08-29 09:51:08.947 +0800
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Anonymous UUID:        42E4176B-6AC7-D4CA-E9DC-A4B9DEA90EE3

Time Awake Since Boot: 1300 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x2] Wrong architecture

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /usr/local/lib/libhidapi.0.dylib
  Referenced from: /Users/USER/*/TestGUI.app/Contents/MacOS/hidapi-testgui
  Reason: no suitable image found.  Did find:
	/usr/local/lib/libhidapi.0.dylib: mach-o, but wrong architecture
	/usr/local/lib/libhidapi.0.dylib: stat() failed with errno=1
	/usr/local/lib/libhidapi.0.dylib: mach-o, but wrong architecture
	/usr/local/Cellar/hidapi/HEAD-4d63a50/lib/libhidapi.0.dylib: mach-o, but wrong architecture
	/usr/local/Cellar/hidapi/HEAD-4d63a50/lib/libhidapi.0.dylib: mach-o, but wrong architecture
...

```

mcuee, 2021-08-29:
So my question is how to specify the dylib directory. In the command line, I can do that. But I do not know how to do that in the app bundle.

```
TestGUI.app/Contents/MacOS on  master [?] ❯ ./hidapi-testgui 
dyld: Library not loaded: /usr/local/lib/libhidapi.0.dylib
  Referenced from: /Users/mcuee/build/hidapi/hidapi/build_autotool/testgui/TestGUI.app/Contents/MacOS/./hidapi-testgui
  Reason: no suitable image found.  Did find:
	/usr/local/lib/libhidapi.0.dylib: mach-o, but wrong architecture
	/usr/local/lib/libhidapi.0.dylib: stat() failed with errno=1
	/usr/local/lib/libhidapi.0.dylib: mach-o, but wrong architecture
	/usr/local/Cellar/hidapi/HEAD-4d63a50/lib/libhidapi.0.dylib: mach-o, but wrong architecture
	/usr/local/Cellar/hidapi/HEAD-4d63a50/lib/libhidapi.0.dylib: mach-o, but wrong architecture
zsh: abort      ./hidapi-testgui

TestGUI.app/Contents/MacOS on  master [?] ❯ DYLD_LIBRARY_PATH=/opt/homebrew/lib ./hidapi-testgui     
Unable to revert mtime: /Library/Fonts
```

mcuee, 2021-08-29:
Full build-log for ARM64. "/Users/mcuee/bin/bin/hidapi-testgui" works just fine. But the "TestGUI.app" is not working.
```

hidapi/build_autotool on  master [?] ❯ ../configure --enable-testgui --prefix=/Users/mcuee/bin 
checking for a BSD-compatible install... /opt/homebrew/bin/ginstall -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /opt/homebrew/bin/gmkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for ar... ar
checking the archiver (ar) interface... ar
checking build system type... aarch64-apple-darwin20.6.0
checking host system type... aarch64-apple-darwin20.6.0
checking how to print strings... printf
checking for a sed that does not truncate output... /opt/homebrew/bin/gsed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /Library/Developer/CommandLineTools/usr/bin/ld
checking if the linker (/Library/Developer/CommandLineTools/usr/bin/ld) is GNU ld... no
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert aarch64-apple-darwin20.6.0 file names to aarch64-apple-darwin20.6.0 format... func_convert_file_noop
checking how to convert aarch64-apple-darwin20.6.0 file names to toolchain format... func_convert_file_noop
checking for /Library/Developer/CommandLineTools/usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for archiver @FILE support... no
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dsymutil... dsymutil
checking for nmedit... nmedit
checking for lipo... lipo
checking for otool... otool
checking for otool64... no
checking for -single_module linker flag... yes
checking for -exported_symbols_list linker flag... yes
checking for -force_load linker flag... yes
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
[... 86 lines cut ...]
Making install in mac
 /opt/homebrew/bin/gmkdir -p '/Users/mcuee/bin/lib'
 /bin/sh ../libtool   --mode=install /opt/homebrew/bin/ginstall -c   libhidapi.la '/Users/mcuee/bin/lib'
libtool: install: /opt/homebrew/bin/ginstall -c .libs/libhidapi.0.dylib /Users/mcuee/bin/lib/libhidapi.0.dylib
libtool: install: (cd /Users/mcuee/bin/lib && { ln -s -f libhidapi.0.dylib libhidapi.dylib || { rm -f libhidapi.dylib && ln -s libhidapi.0.dylib libhidapi.dylib; }; })
libtool: install: /opt/homebrew/bin/ginstall -c .libs/libhidapi.lai /Users/mcuee/bin/lib/libhidapi.la
libtool: install: /opt/homebrew/bin/ginstall -c .libs/libhidapi.a /Users/mcuee/bin/lib/libhidapi.a
libtool: install: chmod 644 /Users/mcuee/bin/lib/libhidapi.a
libtool: install: ranlib /Users/mcuee/bin/lib/libhidapi.a
 /opt/homebrew/bin/gmkdir -p '/Users/mcuee/bin/include/hidapi'
 /opt/homebrew/bin/ginstall -c -m 644 ../../hidapi/hidapi.h '/Users/mcuee/bin/include/hidapi'
Making install in hidtest
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
Making install in testgui
 /opt/homebrew/bin/gmkdir -p '/Users/mcuee/bin/bin'
  /bin/sh ../libtool   --mode=install /opt/homebrew/bin/ginstall -c hidapi-testgui '/Users/mcuee/bin/bin'
libtool: install: /opt/homebrew/bin/ginstall -c .libs/hidapi-testgui /Users/mcuee/bin/bin/hidapi-testgui
make[2]: Nothing to be done for `install-data-am'.
make[2]: Nothing to be done for `install-exec-am'.
 /opt/homebrew/bin/gmkdir -p '/Users/mcuee/bin/share/doc/hidapi'
 /opt/homebrew/bin/ginstall -c -m 644 ../README.md ../AUTHORS.txt ../LICENSE-bsd.txt ../LICENSE-gpl3.txt ../LICENSE-orig.txt ../LICENSE.txt '/Users/mcuee/bin/share/doc/hidapi'
 /opt/homebrew/bin/gmkdir -p '/Users/mcuee/bin/lib/pkgconfig'
 /opt/homebrew/bin/ginstall -c -m 644 pc/hidapi.pc '/Users/mcuee/bin/lib/pkgconfig'
```

mcuee, 2021-08-29:
[image]

mcuee, 2021-08-29:
Looks the warning about signature is fatal. The others may not be a real issue.

```
bundling /opt/homebrew/opt/fox/lib/libFOX-1.6.0.dylib because it's in the list.
/Library/Developer/CommandLineTools/usr/bin/install_name_tool: warning: changes being made to the file will invalidate the code signature in: /Users/mcuee/build/hidapi/hidapi/build_autotool/testgui/TestGUI.app/Contents/MacOS/libFOX-1.6.0.dylib
```

Crash log:
```
Process:               hidapi-testgui [13427]
Path:                  /Users/USER/*/TestGUI.app/Contents/MacOS/hidapi-testgui
Identifier:            us.signal11.hidtestgui
Version:               1.0
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           hidapi-testgui [13427]
User ID:               501

Date/Time:             2021-08-29 12:49:52.892 +0800
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Anonymous UUID:        42E4176B-6AC7-D4CA-E9DC-A4B9DEA90EE3

Sleep/Wake UUID:       D9BE8A4A-53DE-4DD8-830F-7C0182A495BB

Time Awake Since Boot: 5800 seconds
Time Since Wake:       690 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (Code Signature Invalid)
Exception Codes:       0x0000000000000032, 0x0000000104b64000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace CODESIGNING, Code 0x2

kernel messages:

VM Regions Near 0x104b64000:
    __LINKEDIT                  104b5c000-104b64000    [   32K] r--/r-- SM=COW  /Users/*/TestGUI.app/Contents/MacOS/hidapi-testgui
--> mapped file                 104b64000-104b68000    [   16K] r--/r-x SM=PRV  Object_id=25c655a3
    __TEXT                      104b80000-104b84000    [   16K] r-x/rwx SM=COW  /Users/*/*.dylib

Application Specific Information:
dyld: launch, loading dependent libraries
@executable_path/libFOX-1.6.0.dylib
```

mcuee, 2021-08-29:
No such signature warning under x64 build.

```
hidapi/build_auto_x64 on  master [?] took 11s ❯ make
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in mac
  CC       hid.lo
  CCLD     libhidapi.la
Making all in hidtest
  CC       test.o
  CCLD     hidtest
Making all in testgui
  CXX      test.o
  OBJC     mac_support_cocoa.o
  CXXLD    hidapi-testgui
../../testgui/copy_to_bundle.sh
libtool: warning: '../mac/libhidapi.la' has not been installed in '/Users/mcuee/binx64/lib'
libtool: install: cp .libs/hidapi-testgui /Users/mcuee/build/hidapi/hidapi/build_auto_x64/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui
bundling /usr/local/opt/fox/lib/libFOX-1.6.0.dylib because it's in the list.
bundling @executable_path/libFOX-1.6.0.dylib because it's in the list.
make[2]: Nothing to be done for `all-am'.
```

mcuee, 2021-08-29:
There is apparently a differnce between an ARM64 build and x86_64 build.

```
hidapi/build_autotool on  master [?] took 2s ❯ codesign --verify --deep --verbose=2 ../build_auto_x64/testgui/TestGUI.app
../build_auto_x64/testgui/TestGUI.app: code object is not signed at all
In architecture: x86_64

hidapi/build_autotool on  master [?] ❯ codesign --verify --deep --verbose=2 ./testgui/TestGUI.app
./testgui/TestGUI.app: code has no resources but signature indicates they must be present
```

todbot, 2021-08-29:
> Nice tip. But I do not know how to apply the same tip to GUI applications which relies on multiple libraries.

Apologies, I only really read your last post before I replied. The dual `-arch` CFLAGS must be done for all binaries being created (including shared libs). If you're using libraries you don't fully control or can specify the build style for, then you'll have to create two different executables.

mcuee, 2021-08-29:
No problem. It is okay to use two different executables in this case.

Just wondering who knows how to fix the invalid signature problem on the ARM64 (Apple Silicon) side.

Youw, 2021-09-04:
> Even if I copy the necessary dylib files to the TestGUI.app/Contents/MacOS folder to be in the same directory with hidapi-testgui, TestGui.app will still crash as it still wants to find /usr/local/lib files which are of x64 arch.

Try: `otool -l <your binary/executable>`
You'll get a list of all kind of things, including the list of dynamic libraries your binary links too statically (`LC_LOAD_DYLIB` section):
- Some of them will be listed as an absolute paths, e.g.: `/usr/lib/libSystem.B.dylib` - there is no way to hack/inject the usage of those, unless replacing the file itself;
- Some of them will be listed as a relative path, e.g.: `@rpath/libhidapi.0.dylib`; those are searched relatively to [RPATH](https://developer.apple.com/library/archive/documentation/DeveloperTools/Conceptual/DynamicLibraries/100-Articles/RunpathDependentLibraries.html) (or RUNPATH);
- List of `RPATH`s or `RUNPATH`s is always hardcoded into the binary itself (unlike in linux, where you configure ldconfig) - `LC_RPATH` section;
 
`DYLD_LIBRARY_PATH` is probably the only way to extend the list of `RPATH`s (and Apple has deprecated its usage, and it is really hard to use it properly nowadays; it is a separate topic - I won't start it now), but even it won't work for dependencies that are listed as an absolute paths;

The only way to tweak you binary to use propper set of libraries (after it was built/linked) - is to use [`install_name_tool`](https://www.unix.com/man-page/osx/1/install_name_tool/), either by `-change <old> <new>` the path to dependent library or by `-delete_rpath <old>/-add_rpath <new>` list or `RPATH` where to search for the dependent libraries.

---

@mcuee I haven't tried to describe the full story of how this all things works on macOS, but this _should_ be enough for you to play around and try to make it work.

mcuee, 2021-09-05:
Thanks. I will try the tip to see if I can get it work or not.

mcuee, 2021-09-05:
This diff fixed the issue for me under my Mac Mini M1.
```
hidapi on  master [!?] via △ v3.21.2 ❯ git diff
diff --git a/testgui/copy_to_bundle.sh b/testgui/copy_to_bundle.sh
index ddff64f..7085bcc 100755
--- a/testgui/copy_to_bundle.sh
+++ b/testgui/copy_to_bundle.sh
@@ -11,7 +11,8 @@ EXEPATH=$PWD/TestGUI.app/Contents/MacOS
 # may not be in /opt/local. One per line. These
 # are used with grep, so only a portion of the name
 # is required. eg: libFOX, libz, etc.
-LIBS_TO_BUNDLE=libFOX
+#LIBS_TO_BUNDLE=libFOX
+LIBS_TO_BUNDLE=
 
 
 function copydeps {
```

mcuee, 2021-09-05:
That being said, I think it is not properly done as the libhidapi.0.dylib is hardcoded to "/Users/mcuee/bin/lib/libhidapi.0.dylib" which I installed to as part of the build process.

```

/Applications🔒 ❯ otool -l ./TestGUI_m1.app/Contents/MacOS/hidapi-testgui 
./TestGUI_m1.app/Contents/MacOS/hidapi-testgui:
Load command 0
      cmd LC_SEGMENT_64
  cmdsize 72
  segname __PAGEZERO
   vmaddr 0x0000000000000000
   vmsize 0x0000000100000000
  fileoff 0
 filesize 0
  maxprot 0x00000000
 initprot 0x00000000
   nsects 0
    flags 0x0
Load command 1
      cmd LC_SEGMENT_64
  cmdsize 1032
  segname __TEXT
   vmaddr 0x0000000100000000
   vmsize 0x0000000000008000
  fileoff 0
 filesize 32768
  maxprot 0x00000005
 initprot 0x00000005
   nsects 12
    flags 0x0
Section
  sectname __text
   segname __TEXT
      addr 0x0000000100003c58
      size 0x0000000000002400
    offset 15448
     align 2^2 (4)
    reloff 0
    nreloc 0
     flags 0x80000400
 reserved1 0
 reserved2 0
Section
  sectname __stubs
   segname __TEXT
      addr 0x0000000100006058
      size 0x0000000000000384
    offset 24664
     align 2^2 (4)
    reloff 0
    nreloc 0
     flags 0x80000408
 reserved1 0 (index into indirect symbol table)
 reserved2 12 (size of stubs)
Section
  sectname __stub_helper
   segname __TEXT
      addr 0x00000001000063dc
      size 0x0000000000000384
    offset 25564
     align 2^2 (4)
    reloff 0
    nreloc 0
     flags 0x80000400
 reserved1 0
 reserved2 0
Section
  sectname __objc_methlist
   segname __TEXT
      addr 0x0000000100006760
[... 542 lines cut ...]
compatibility version 300.0.0
Load command 37
          cmd LC_LOAD_DYLIB
      cmdsize 56
         name /usr/lib/libobjc.A.dylib (offset 24)
   time stamp 2 Thu Jan  1 07:30:02 1970
      current version 228.0.0
compatibility version 1.0.0
Load command 38
      cmd LC_FUNCTION_STARTS
  cmdsize 16
  dataoff 71592
 datasize 48
Load command 39
      cmd LC_DATA_IN_CODE
  cmdsize 16
  dataoff 71640
 datasize 0
Load command 40
      cmd LC_CODE_SIGNATURE
  cmdsize 16
  dataoff 87248
 datasize 827

```

Youw, 2021-09-05:
Yeah, it isn't.
To build it properly - a lot of tweaking need to be done for autotools scripts, which I'd rather not touch.

CMake build off-the-box likely won't properly work too, but CMake has a lot of convetient machanism to configure it easily.
I will do so if/when we have Qt-based GUI.

mcuee, 2021-09-13:
I will still keep this issue closed for now. Ad the testgui build itself is okay and runs properly under macOS (x64 and ARM64), only the application bundle is not generated properly.

JustJoostNL, 2023-01-14:
I still have this issue, is there already a solution?

Lewiscowles1986, 2023-07-29:
So I'm trying to use python ctypes to load the built obj file.

```
import ctypes

ctypes.cdll.LoadLibrary('hid.o')
```

this then gives me a number of errors; but an important (I feel) error reads

> unloadable mach-o file type 1 'hid.o'

I built using the guide to clone, cd mac, `make -f Makefile-manual`

Is this some weirdo OSX security issue, or something else?

The sample program builds and runs enumerating devices, so I don't doubt that hidapi works if I call it from C

mcuee, 2026-05-02:
Now it failed to build with latest macOS 26.4.1.

```
mcuee in ~/build/libusb/hidapi/build_autotools on master % sw_vers 
ProductName:		macOS
ProductVersion:		26.4.1
BuildVersion:		25E253

mcuee in ~/build/libusb/hidapi/build_autotools on master % make -i 
/Library/Developer/CommandLineTools/usr/bin/make  all-recursive
Making all in mac
make[2]: Nothing to be done for `all'.
Making all in hidtest
make[2]: Nothing to be done for `all'.
Making all in testgui
  OBJC     mac_support_cocoa.o
../../testgui/mac_support_cocoa.m:47:9: error: cannot initialize return object of type
      'NSApplicationTerminateReply' with an rvalue of type 'BOOL' (aka 'bool')
   47 |         return YES;
      |                ^~~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/objc/objc.h:96:13: note: expanded
      from macro 'YES'
   96 | #define YES __objc_yes
      |             ^~~~~~~~~~
1 error generated.
make[2]: [mac_support_cocoa.o] Error 1 (ignored)
  CXXLD    hidapi-testgui
clang++: error: no such file or directory: 'mac_support_cocoa.o'
make[2]: [hidapi-testgui] Error 1 (ignored)
../../testgui/copy_to_bundle.sh
libtool: install: cp hidapi-testgui /Users/mcuee/build/libusb/hidapi/build_autotools/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui
cp: hidapi-testgui: No such file or directory
error: /Library/Developer/CommandLineTools/usr/bin/otool-classic: can't open file: /Users/mcuee/build/libusb/hidapi/build_autotools/testgui/TestGUI.app/Contents/MacOS/hidapi-testgui (No such file or directory)
```

PR #152 (2020-03-08): testgui: ensure $EXEPATH is a directory
Otherwise hidapi-testgui will be put at TestGUI.app/Contents/MacOS which
results in a malformed .app structure.

The following build combinations have been tested:
* 'make -f Makefile-manual' with libtool
* 'make -f Makefile-manual' without libtool
* './configure --enable-testgui && make'

Fixes #151
commit 30c46ad8 (2020-03-08): testgui: ensure $EXEPATH is a directory

Otherwise hidapi-testgui will be put at TestGUI.app/Contents/MacOS which
results in a malformed .app structure.

The following build combinations have been tested:
* 'make -f Makefile-manual' with libtool
* 'make -f Makefile-manual' without libtool
* './configure --enable-testgui && make'

Fixes #151