libusb/hidapi #156: Crosscompiling libudev

jano42, 2020-04-14:
in cross compilation readme (https://github.com/libusb/hidapi#prerequisites-1)
it is said to build libudev (Build libudev. From the libudev source directory, run:....)

But where are libudev sources ? I find it only in systemd, with strange and very obscur meson build system...
Is there another sources ?

Youw, 2020-04-14:
This part of the README was written [8 years ago](https://github.com/libusb/hidapi/commit/6d50f8fa0f889566cdcc4affe9244ce0ae28d8fc) by the original author.
I guess not much users needed to build libudev by themselves, since they switched to meson, so you have to excuse us that the readme is not up-to-date.

If/when you figure out how to build it properly - PR/readme update is welcome.

mcuee, 2021-06-13:
Indeed README file needs some improvement.

Youw, 2021-06-13:
https://github.com/libusb/hidapi/blob/cmake-support/README.md is an (unfinished) improvement.

mcuee, 2021-07-13:
New README is probably good enough.

closed 2021-07-13 by mcuee