libusb/hidapi #158: hid_get_manufacturer_string(handle, wstr, MAX_STR);   Getting Exception

BarisUygun, 2020-04-21:
`#ifdef WIN32
#include <windows.h>
#endif
#include <stdio.h>
#include <stdlib.h>
#include "hidapi.h"

#define MAX_STR 255

int main(int argc, char* argv[])
{
	int res;
	unsigned char buf[65];
	wchar_t wstr[MAX_STR];
	hid_device *handle;
	int i;

	// Initialize the hidapi library
	res = hid_init();

	// Open the device using the VID, PID,
	// and optionally the Serial number.
	handle = hid_open(0x4d8, 0x3f, NULL);

	// Read the Manufacturer String
	res = hid_get_manufacturer_string(handle, wstr, MAX_STR);
	wprintf(L"Manufacturer String: %s\n", wstr);

	// Read the Product String
	res = hid_get_product_string(handle, wstr, MAX_STR);
	wprintf(L"Product String: %s\n", wstr);

	// Read the Serial Number String
	res = hid_get_serial_number_string(handle, wstr, MAX_STR);
	wprintf(L"Serial Number String: (%d) %s\n", wstr[0], wstr);

	// Read Indexed String 1
	res = hid_get_indexed_string(handle, 1, wstr, MAX_STR);
	wprintf(L"Indexed String 1: %s\n", wstr);

	// Toggle LED (cmd 0x80). The first byte is the report number (0x0).
	buf[0] = 0x0;
	buf[1] = 0x80;
	res = hid_write(handle, buf, 65);

	// Request state (cmd 0x81). The first byte is the report number (0x0).
	buf[0] = 0x0;
	buf[1] = 0x81;
	res = hid_write(handle, buf, 65);

	// Read requested state
	res = hid_read(handle, buf, 65);

	// Print out the returned buffer.
	for (i = 0; i < 4; i++)
		printf("buf[%d]: %d\n", i, buf[i]);

	// Close the device
	hid_close(handle);

	// Finalize the hidapi library
	res = hid_exit();

	return 0;
}`

Youw, 2020-04-21:
What kind of exception?

BarisUygun, 2020-04-21:
Exception thrown at 0x77631CB9 (hidapi.dll) in dene.exe: 0xC0000005: Access violation reading location 0x00000000.

If there is a handler for this exception, the program may be safely continued.

Also handle = hid_open(0x4d8, 0x3f, NULL); returns null

Youw, 2020-04-21:
`handle = hid_open(0x4d8, 0x3f, NULL); returns null`
so here is your issue
do you have a device VID=0x4d8, PID=0x3f connected to your system?

BarisUygun, 2020-04-21:
Yeah i tought that this was the main issue i do not understand what VID and PID are suppose to be.

Youw, 2020-04-21:
Those are the VendorID and DeviceID of your device
those are up to you to figure out what are they (check device's specification/documentation)

P.S.: make sure you read **Warning** part here: https://github.com/libusb/hidapi#what-does-the-api-look-like

closed 2020-04-21 by Youw