libusb/hidapi #160: Can the Device Capabilities Add FeatureReportLength

PaulDeignan, 2020-04-30:
Many devices require fixed lengths for getting and setting feature reports.  This info is available from the caps structure but it is not exposed.
ex.
dev->feature_report_length = caps.FeatureReportByteLength;

Youw, 2020-04-30:
I guess that's possible

PRs are welcome

per-nn, 2020-06-10:
I found this issue while investigating the same problem for my own project. I even made a [patch](https://github.com/per-nn/hidapi/commit/ed3c82553cabfe33f2602a3a78c4c9b6d8daff3b) ready for pull request, when I realized this is not the correct way to do it.

I use get/set feature to get serial communication over HID with a device I build. I typically only write part of the buffer feature report, and this causes the API call to fail. I knew the WDK API had the size of the buffer, so I hoped I wouldn't have to fill every feature up with zeros to this exact size. This was further motivated by that variations of my product used different buffer feature report sizes.

I am not sure if the driver will attempt to set or get a feature with a different size than inferred from the report descriptor. Regardless, this commit will at least allocate a buffer of the size of the largest feature report for every set feature call. If, for instance, the device has a feature that is a 10 kB calibration table, and a 1 byte feature setting the update frequency, every call to set the frequency will allocate the full 10 kB. 

If this commit also affects the size sent by the driver, which I doubt, it could break compatibility with devices relying upon this size.

So if anyone wants this you are welcome to try, but I advice against it.

crashniels, 2020-12-24:
Any updates on this? I have a device that is identified by it's FeatureReportLength. It shows up 3x via the usage page 0xFF00 and is separated by the feature report lengths of 520 and 6. The 520 one being for writes and the 6 one for reads.

todbot, 2020-12-24:
This is difficult to do in a cross-platform way, so I expect there won't be much movement on this. It requires reading and parsing the HID Report Descriptor for each device, which means opening each device on Linux and MacOS I think.

@crashniels, in your case, you should be able to disambiguate your devices by looking at the HID Report Descriptor for each device and finding which reportIds you should send on.

crashniels, 2020-12-24:
Thanks for the quick reply. I already made a quick hack that is Windows only. Linux does not need this as I can just claim interface 1 and it is done. Windows is special and the descriptor was made by some random Chinese company which does not combine all usage pages under 0xFF00 but rather splits them into separate devices. The report ids that I need are 4 and 5. One 0xFF00 device accepts 4 and the other 5. The de-compiled software does the same in its detector function.

todbot, 2020-12-24:
Wow what a mess.  Good luck. :)

The `hid_device_info` struct does return `interface_number` so can you `hid_enumerate(vid,pid)` and then pick which device as the `interface_number`, `usage_page` and `usage` you are looking for?

crashniels, 2020-12-24:
Thanks and yes I can. On mine however it lists three. That is why I have to filter further by going after the feature report length sizes.

Youw, 2020-12-25:
> That is why I have to filter further by going after the feature report length sizes

Aren't usage + usage_page `combination` isn't enough, to identify it on Win?

k1-801, 2020-12-25:
It is _expected to_, but somehow it isn't. A single physical device shows up as 6 different HID devices, 3 of which have the same interface no (1), the same usage page and the same usage. But they behave _differently_. If there was any other way to distinguish them, it would have been used already.

It looks like this:

```
\\?\hid#vid_258a&pid_0033&mi_01&col04#9&3d7401&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030} < ff00:1
\\?\hid#vid_258a&pid_0033&mi_01&col02#9&3d7401&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030} < c:1
\\?\hid#vid_258a&pid_0033&mi_01&col05#9&3d7401&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030} < ff00:1
\\?\hid#vid_258a&pid_0033&mi_00#9&136d903f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} < 1:2
\\?\hid#vid_258a&pid_0033&mi_01&col03#9&3d7401&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030} < ff00:1
\\?\hid#vid_258a&pid_0033&mi_01&col01#9&3d7401&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd < 1:6
```

Numbers after the '<' are the page and usage respectively.

crashniels, 2020-12-25:
Usage page and usage is usually enough looking at other devices in OpenRGB. This device however is weird. The report descriptor from the USB dump lists 3 devices with the usage page 0xFF00 and usage 0x00. The only difference is as mentioned the size it accepts. The Corsair Keyboard I have only lists one device with the usage page 0xFFC2. 
The parsed report descriptor of the device: <details>0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0x0C, 0x00,  // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x25, 0x01,        //   Logical Maximum (1)
0x15, 0x00,        //   Logical Minimum (0)
0x75, 0x01,        //   Report Size (1)
0x0A, 0xB5, 0x00,  //   Usage (Scan Next Track)
0x0A, 0xB6, 0x00,  //   Usage (Scan Previous Track)
0x0A, 0xB7, 0x00,  //   Usage (Stop)
0x0A, 0xCD, 0x00,  //   Usage (Play/Pause)
0x0A, 0xE2, 0x00,  //   Usage (Mute)
0x0A, 0xA2, 0x00,  //   Usage (Daily)
0x0A, 0xE9, 0x00,  //   Usage (Volume Increment)
0x0A, 0xEA, 0x00,  //   Usage (Volume Decrement)
0x95, 0x08,        //   Report Count (8)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x0A, 0x83, 0x01,  //   Usage (AL Consumer Control Configuration)
0x0A, 0x94, 0x01,  //   Usage (AL Local Machine Browser)
0x0A, 0x86, 0x01,  //   Usage (AL Spreadsheet)
0x0A, 0x88, 0x01,  //   Usage (AL Presentation App)
0x0A, 0x8A, 0x01,  //   Usage (AL Email Reader)
0x0A, 0x92, 0x01,  //   Usage (AL Calculator)
0x0A, 0xA8, 0x02,  //   Usage (0x02A8)
0x0A, 0x84, 0x01,  //   Usage (AL Word Processor)
0x95, 0x08,        //   Report Count (8)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x0A, 0x21, 0x02,  //   Usage (AC Search)
0x0A, 0x23, 0x02,  //   Usage (AC Home)
0x0A, 0x24, 0x02,  //   Usage (AC Back)
0x0A, 0x25, 0x02,  //   Usage (AC Forward)
0x0A, 0x26, 0x02,  //   Usage (AC Stop)
0x0A, 0x27, 0x02,  //   Usage (AC Refresh)
0x0A, 0x2A, 0x02,  //   Usage (AC Bookmarks)
0x0A, 0xB1, 0x02,  //   Usage (0x02B1)
0x95, 0x08,        //   Report Count (8)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x04,        //   Report ID (4)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x09, 0x00,        //   Usage (0x00)
0x75, 0x08,        //   Report Size (8)
0x96, 0x07, 0x02,  //   Report Count (519)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x07,        //   Report ID (7)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x09, 0x00,        //   Usage (0x00)
0x75, 0x08,        //   Report Size (8)
0x95, 0x07,        //   Report Count (7)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x05,        //   Report ID (5)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x09, 0x00,        //   Usage (0x00)
0x95, 0x05,        //   Report Count (5)
0x75, 0x08,        //   Report Size (8)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 213 bytes
</details>

todbot, 2020-12-25:
Judging from that HID Report Descriptor, I think you can do `hid_open_path()` on any of those devices with usage_page/usage:0xFF00/0x0001 and send on reportIds 4, 5, or  7 and it should behave as you expect.

PaulDeignan, 2020-12-25:
The problem comes when trying to call SendFeatureReport.  If you don’t know the length and send an entire report then an error is thrown.

Paul Deignan

Youw, 2020-12-25:
> If you don’t know the length and send an entire report then an error is thrown.

That's [fixed](https://github.com/libusb/hidapi/commit/fb4135c7a87e87fb9f3f2fb95c3389677eb35ac9) in current master. Can you try it out?

> It looks like this:

Weird, this is the first time I see a device like that.

PaulDeignan, 2020-12-25:
Is that only fixed on windows or is it also fixed on Linux?

Paul Deignan

todbot, 2020-12-25:
> The problem comes when trying to call SendFeatureReport. If you don’t know the length and send an entire report then an error is thrown. Paul Deignan

That's how USB HID works. Both the device and the application have to heed the USB HID Report Descriptor. I know there's a fix for Windows, but if you send a 8-byte buffer to a reportId expecting a 519-byte payload, you're going to get undefined behavior.

Youw, 2020-12-25:
That's fixed for Windows only, because Windows API requires it.

@PaulDeignan are you having a different issue than @crashniels has?

hidapi originally designed for applications that specifically know all the device they are going to work with in advance, including the report size. So identifying a device is a one thing, and sending a correct-size reports - that's a "must have".

crashniels, 2021-01-02:
> Judging from that HID Report Descriptor, I think you can do `hid_open_path()` on any of those devices with usage_page/usage:0xFF00/0x0001 and send on reportIds 4, 5, or 7 and it should behave as you expect.

No as writing with a different report id as what the device accepts results in hid_send_feature_report returning -1. 

> hidapi originally designed for applications that specifically know all the device they are going to work with in advance, including the report size. So identifying a device is a one thing, and sending a correct-size reports - that's a "must have".

I know the report sizes I need to send. I just need to know which device accepts them.

On Windows I have a quick hack implemented in [my fork](https://github.com/crashniels/hidapi) of hidapi. I have it working in [this fork](https://gitlab.com/crashniels/OpenRGB/-/blob/SinowealthController/Controllers/SinowealthController/SinowealthControllerDetect.cpp) of OpenRGB. If Linux is now also using the usage page changes I am not quite sure how I should go from there. There is no accessible function to return the maximum report length.

todbot, 2021-01-02:
@crashniels do you have an Amazon ASIN or similar link to the device in question? I'd like to get one of these myself to experiment on.

crashniels, 2021-01-02:
Its the [Glorious Model O](https://www.amazon.de/Glorious-PC-Gaming-Race-Gaming-Maus/dp/B07LDH3TWM/ref=sr_1_2?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=model+o&qid=1609629963&sr=8-2).

todbot, 2021-01-02:
aha, awesome name and looks like something out of a movie set. sold. 👍

Youw, 2023-03-12:
I believe this is fixed with https://github.com/libusb/hidapi/pull/224 as much as possible.

closed 2023-03-12 by Youw