libusb/hidapi #162: Windiws 10: hid_read() never returns or return 0 if used with timeout

bearsh, 2020-05-08:
I have a custom hid device (stm micro). Using linux, reading/writing works perfectly but on Win10, hid_read() just blocks, no data arrives. sending data with hid_write works.

Youw, 2020-05-08:
Are you sure you're opening same device? (Same VID/PID/Serial Number/USAGE/USAGE_PAGE?

bearsh, 2020-05-08:
I open the device once, then I write/read, so I'm pretty sure. 

Yeah, I'm sure the device send data, as it works with linux and I also played around with USBlyzer which also displays the `in` data

Youw, 2020-05-08:
> I open the device once, then I write/read, so I'm pretty sure.

Sure of what?
Did you check the USAGE/USAGE_PAGE of the logical device before/after you open it?
Windows separates "physical" HID device with several USAGE/USAGE_PAGE pages into separate _logical_ devices and separates the access to those on driver level.

libusb/macOS implementation don't do such separations, and that's an OS-specific behavior which cannot be controlled by any user application/library.

>  I'm sure the device send data, as it works with linux

based on my experience, I wouldn't count this as a proof

> I also played around with USBlyzer which also displays the in data

great, that's a good proof

---
The most important question I should've ask at the beginning:
What version of hidapi? Is it master/specific release? Build by yourself or from version control system?
What backend on Linux works for you (hidraw or libusb)?

bearsh, 2020-05-08:
> Sure of what?
> Did you check the USAGE/USAGE_PAGE of the logical device before/after you open it?
> Windows separates "physical" HID device with several USAGE/USAGE_PAGE pages into separate _logical_ [...]

is there an easy way I can check this?

> What version of hidapi? Is it master/specific release? Build by yourself or from version control system?
> What backend on Linux works for you (hidraw or libusb)?

I tried it with master and a ~three year old version bundled with https://github.com/karalabe/hid (both from go). I also tried it with cython-hidapi-0.9.0.post2 without success

On linux I use it with the libusb backend (hidraw I have to test)

Youw, 2020-05-08:
`hid_enumerate`, and check usage/usage page of the device, before opening it: https://github.com/libusb/hidapi/blob/master/hidapi/hidapi.h#L67

bearsh, 2020-05-08:
hid_enumerate returns one entry with
```
Path: "\\\\?\\hid#vid_0483&pid_a364#6&30db5a96&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}"
UsagePage: 0xff00
Usage: 0x1
Interface: -1
```

Youw, 2020-05-08:
Unfortunately I can't say that I can help you right now with all the info I see.

One more note: try `hid_get_input_report` instead of `hid_read`

bearsh, 2020-05-08:
ok, it seems I figured it out, my device used to send a variable number of bytes whereas I called `hid_head_timeout(..., ..., 64, ...)` with the max possible length. this works fine on linux, it just returns the actual bytes received but not on windows, where it seems a hid transfer has to be either 64byte long or at least the report length long (don't know which one applies as my report length is 64bytes).

In the end, is this the expected behavior? reading the function doc it might be (but then, why does it work differently on linux) or just another stupid windows misfeature?

todbot, 2020-05-08:
Yes, this is the expected behavior.  On Windows, the buffer to write or the buffer to read must be exactly the size of the HID report you are sending / receiving.  On Linux and MacOS it is not this way.

Youw, 2020-05-08:
All the devices I had a chance to work with - always send the <report length> chunks of data, with reports of 4 or 8 bytes.
Practically speaking - it is very expected from Windows.
As for "is this behavior defined by HID specification?" - don't know for sure, but looks like so. Meaning - if you have a "report length", you'd expect each report to be of specified "length".
HID specification definitely doesn't define a "variable length" reports.

bearsh, 2020-05-09:
@Youw @todbot **thanks** for your fast responses!

As I have a already defined protocol, which uses variable package length, which I also want to expose via hid, I think I have to live with that... 

thinking about it, it may be possible to define a couple of reports (e.g. 16, 24, 48, 64 bytes) and use them but I doubt that will work easily and requires more work on embedded device and pc for no real value...

as the main problem is solved, I'm closing the issue...

closed 2020-05-09 by bearsh