libusb/hidapi #170: hid_read hang in linux, but hid_write and hid_read work fine in window10

Rockyhoon, 2020-06-13:
I've tried both libusb/hid.c and linux/hid.c, when I run hidtest.cpp, it hangs on at hid_read(), if I hid_set_nonblocking(handle, 0). And I've set device permission by the method provided in /udev directory. The whole system works fine in window10. Now I am desperate.
I've also post on signal11 before I found this repo.

Any suggestion is appreciated!

todbot, 2020-06-13:
What is `hidtest.cpp`?  Do you mean `hidtest/test.c`?  If so, you have the VID/PID of your device and you know the reports being sent & received by your device?  `hidtest/test.c` is a compilable code example for a non-existent device, not a real one.

Rockyhoon, 2020-06-14:
Yes, I meant hidtest/test.c. I've modified VID/PID for my USB hid device, and the example code can print Manufacturer String, Product String... properly. However, while I execute hid_read(), it is stuck there.

Youw, 2020-06-15:
I assume that you're reading less than a "report size" of bytes. On Linux/Windows the behavior would be exactly as you described.

Rockyhoon, 2020-06-15:
Thanks for answering.
Although I don't know "report size" exactly, I've still tried 
	res = hid_send_feature_report(handle, buf, 17); in the test.
It returns "Unable to send a feature report.\n".

Youw, 2020-06-15:
Right now hidapi doesn't provide an API to get a report size (see #160).

As a general rule - you should refer to the documentation for the device you're using.

As an alternative, you can check it on Windows, using the debugger: https://github.com/libusb/hidapi/blob/master/windows/hid.c#L616

Rockyhoon, 2020-06-15:
Thanks for the advice.
I've tried that in win10, and get both output_report_length and input_report_length are 0x100.
Unfortunately, I read/write 100 bytes in Linux, it still hangs there.
Is there anything to do with the cpu architecture, cause mine is not typical x86.

Youw, 2020-06-15:
> Unfortunately, I read/write 100 bytes in Linux, it still hangs there.

Depending either you're using indexed reports you may try to read 99/101 bytes.
You could also try to use `get/set_input_report` instead of `read/write`

> Is there anything to do with the cpu architecture, cause mine is not typical x86.

I don't believe that should matter. Although I guess you could try using smth like Ubuntu x86_64 to be sure.

jonasmalacofilho, 2020-06-24:
Just noticed something: 0x100 is 256 decimal (also try 257 to account for numbered reports).

mcuee, 2021-01-26:
@Rockyhoon Please check your code to use 0x100 (256) or 0x101 (257).

mcuee, 2021-06-16:
@Rockyhoon I will close this for now. Please reopen if you come back.

Please try out hidapitester as well.
https://github.com/todbot/hidapitester

closed 2021-06-16 by mcuee