libusb/hidapi #171: Reopening closed USB device causes crash on Mac unless the IOHIDManager is reinitialised

victorlewis, 2020-06-16:
On MacOS systems, when a USB device handle is closed via hid_close and then reopened via hid_open, hid_enumerate will cause a hard crash. If however, hid_init is changed from:

```
int HID_API_EXPORT hid_init(void)
{
	if (!hid_mgr) {
		return init_hid_manager();
	}

	/* Already initialized. */
	return 0;
}
```

to 

```
int HID_API_EXPORT hid_init(void)
{
	return init_hid_manager();
}
```

The USB device can then be opened and closed an arbitrary number of times with no instability.

MacOS only (tested on High Sierra), no comparable issue on Windows. All hidapi functions are called from a single thread in my application. However I am using hidapi in the context of a VST plugin, and the issue also occurs when the first instance of the plugin is deleted/readded (but not,  however, if multiple instances are created and those later instances are deleted).

#https://github.com/signal11/hidapi/issues/205

Youw, 2020-06-16:
I believe the crash happens only of you physically detach your device.

This doesn't fix an original issue, just hides it.
Since macOS High Sierra, `hid_close` is broken (it always was broken, but now it causes a crash) - if a device was disconnected, HIDAPI leaks a handle: https://github.com/libusb/hidapi/blob/24a822c80f95ae1b46a7a3c16008858dc4d8aec8/mac/hid.c#L1122
The history behind this workaround ancient, caused by a bug in OSX (which apparently is fixed long ago).

What you suggest - is to leak the entire (broken) `hid_mgr`, and use a new one - another workaround.

If you remove that `if` statement on line [1122](https://github.com/libusb/hidapi/blob/24a822c80f95ae1b46a7a3c16008858dc4d8aec8/mac/hid.c#L1122), it will fix the issues as well.

Caused by: https://github.com/libusb/hidapi/issues/144

Youw, 2020-06-16:
Also note: https://github.com/signal11/hidapi/issues/205#issuecomment-527654560

victorlewis, 2020-06-16:
I actually had no problem physically detaching and reconnecting the device, the crash only occurs when the device stays connected but the usb device handle is closed (in my case this is an important functionality as users add/delete plugins arbitrarily). Maybe this is because

`if (!dev->disconnected) `

does not get called when it is disconnected but occurs when the handle is closed while the device stays connected, although commenting out what's inside failed to remove the crash.

Youw, 2020-06-17:
I have a bunch of projects of my own, and open-source, that would open -> use -> close HID device with hidapi, and you're that first one experiencing a crash in this scenario.

Maybe the workaround fixes something for this particular scenario, but it definitely introduces a resource leak, and even for you it has to be a problem.

Can you prepare a small code sample with usage, similar to you real life scenario?

Youw, 2020-06-17:
> as users add/delete plugins arbitrarily

Can you describe more? Does hidapi loaded dynamically, with one of your plugins?

Youw, 2020-10-14:
@victorlewis try if current master still has issue you described

Youw, 2020-11-24:
Presumably fixed by the latest master.
To be reopened by the author, if the issue is still there.

closed 2020-11-24 by Youw