libusb/hidapi #172: I can't see a bluetooth device (Joy-Con) using cython-hidapi library nor I can connect to it with hid python library

soswow, 2020-06-17:
I am trying to use https://github.com/tokoroten-lab/joycon-python this library to work with my Switch Joy-Con controller. First I tried to install `cython-hidapi` both with libusb and without it. In both cases I couldn't make joy-con Bluetooth device to show up in enumeration.
I've reported this over here, https://github.com/trezor/cython-hidapi/issues/89 but author redirected me over here.

Then I noticed that author of `joycon-python` is suggesting using another lib if first didn't work - https://github.com/apmorton/pyhidapi this one. I tried that one. I got the device listed as part of enumeration (first one:
```
[
{'path': b'/dev/hidraw2', 'vendor_id': 1406, 'product_id': 8199, 'serial_number': 'b8:78:26:19:c1:8c', 'release_number': 0, 'manufacturer_string': '', 'product_string': 'Joy-Con (R)', 'usage_page': 11825, 'usage': 12080, 'interface_number': -1},
{'path': b'/dev/hidraw0', 'vendor_id': 1008, 'product_id': 36, 'serial_number': '', 'release_number': 304, 'manufacturer_string': 'CHICONY', 'product_string': 'HP Basic USB Keyboard', 'usage_page': 11825, 'usage': 12080, 'interface_number': 0},
{'path': b'/dev/hidraw1', 'vendor_id': 1118, 'product_id': 203, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'PixArt', 'product_string': 'Microsoft USB Optical Mouse', 'usage_page': 11825, 'usage': 12080, 'interface_number': 0}
]
```
But when I am trying to access it, I get 
```
Traceback (most recent call last):
  File "joy_con_test.py", line 4, in <module>
    device = hid.Device(path=b'/dev/hidraw2')
  File "/home/soswow/env/lib/python3.6/site-packages/hid/__init__.py", line 132, in __init__
    raise HIDException('unable to open device')
hid.HIDException: unable to open device
```
It doesn't matter if i am using path or vendor and product id.

I am not fluent at Bluetooth, HID and Kernels. I am not sure where to look and if my OS is to blame for taking control over device for example and so I can't access it? I am not sure.

Any Idea?

Youw, 2020-06-17:
Apparently you're on Linux, and you're using hidraw backend.

Do you know what version of hidapi do you have? What version on Linux?
There may be plenty of reasons why you can't open a device.

soswow, 2020-06-17:
Thank you, @Youw 
It's `Linux 4.9.140-tegra aarch64` , `Ubuntu 18.04.4 LTS`
I am not sure how to see version of hidapi. I've installed `cython-hidapi` from the source, which has you as a git-module thingy, so I guess It pulled latest from master? 🤷 

@todbot over another issue has [pointed me](https://github.com/trezor/cython-hidapi/issues/89#issuecomment-645462065) to the direction of `Udev rules` which I will not go and learn about. See if that's what I was missing - this is what I hoping to get help about.

Youw, 2020-06-17:
> so I guess It pulled latest from master? 

It uses a specific commit from Jan 22.
It is fairly fresh, that's the only thing I wanted to check.

> `Udev rules`

Ah, right, I keep forgetting of that too.
You can quick check is it the case by running your test app with `sudo`.

soswow, 2020-06-17:
Alright. udev rules was what I was missing. I'll close the issue, since this is not a bug or an issue and @todbot helped me resolve it. udev rules I used are [from here](https://www.reddit.com/r/Stadia/comments/egcvpq/using_nintendo_switch_pro_controller_on_linux/).

closed 2020-06-17 by soswow