libusb/hidapi #175: Consider merging changes from SuperCollider fork

brianlheim, 2020-06-30:
Hey maintainers!

we at [SuperCollider](https://github.com/supercollider/supercollider) have our [own fork](https://github.com/supercollider/hidapi) of hidapi which has diverged quite a bit from the signal11 library:

<details>

```
[bheim@mandible hidapi]$ git diff libusb/master --stat
 .appveyor.yml                                |   33 -
 .builds/alpine.yml                           |   25 -
 .builds/archlinux.yml                        |   18 -
 .builds/fedora-mingw.yml                     |   22 -
 .builds/freebsd.yml                          |   23 -
 .cirrus.yml                                  |   33 -
 .gitignore                                   |    5 +-
 .travis.yml                                  |   13 -
 AUTHORS.txt                                  |    4 +-
 CMakeLists.txt                               |  127 ++++
 Makefile.am                                  |   15 +-
 README.md                                    |  380 ----------
 README.txt                                   |  371 ++++++++++
 cmake_modules/FindIConv.cmake                |   43 ++
 cmake_modules/FindPthreads.cmake             |  104 +++
 cmake_modules/FindUDev.cmake                 |   53 ++
 cmake_modules/Findlibusb-1.0.cmake           |  110 +++
 configure.ac                                 |   64 +-
 dist/hidapi.podspec                          |   31 -
 doxygen/Doxyfile                             | 2522 +++++++++++++++++++++------------------------------------------
 hidapi/CMakeLists.txt                        |    1 +
 hidapi/hidapi.h                              |  160 ++--
 hidapi2osc/CMakeLists.txt                    |   36 +
 hidapi2osc/Makefile.am                       |   20 +
 hidapi2osc/hidapi2osc.cpp                    |  648 ++++++++++++++++
 hidapi2osc/supercollider/hidapi2osc-test.scd |   39 +
 hidapi_parser/CMakeLists.txt                 |   14 +
 hidapi_parser/Makefile.am                    |   20 +
 hidapi_parser/hidapi_parser.c                | 2029 ++++++++++++++++++++++++++++++++++++++++++++++++++
 hidapi_parser/hidapi_parser.h                |  195 +++++
 hidapi_parser/main.c                         |  202 +++++
 hidparsertest/CMakeLists.txt                 |   13 +
 hidparsertest/Makefile.am                    |   20 +
 hidparsertest/hidparsertest.c                |  282 +++++++
 hidtest/CMakeLists.txt                       |   18 +
 hidtest/Makefile.am                          |    6 +-
 hidtest/hidtest.cpp                          |  680 +++++++++++++++++
 hidtest/test.c                               |  195 -----
 hidtestosx/CMakeLists.txt                    |   18 +
 hidtestosx/hidtestosx.cpp                    |   81 ++
 hidtestwindows/CMakeLists.txt                |   18 +
 hidtestwindows/hidtestwindows.cpp            |  311 ++++++++
 hut/hut_11_telephony.yaml                    |  133 ++++
 hut/hut_12_consumer.yaml                     |  464 ++++++++++++
 hut/hut_13_digitizers.yaml                   |  122 +++
 hut/hut_1_generic_desktop.yaml               |  209 ++++++
 hut/hut_20_alphanumeric_display.yaml         |   76 ++
 hut/hut_2_simulation_controls.yaml           |  158 ++++
 hut/hut_3_vr_control.yaml                    |   40 +
 hut/hut_4_sport_controls.yaml                |  107 +++
 hut/hut_5_game_controls.yaml                 |   88 +++
 hut/hut_64_medical_instrument.yaml           |    4 +
 hut/hut_6_generic_device.yaml                |   12 +
 hut/hut_7_keyboard_keypad.yaml               |  659 +++++++++++++++++
 hut/hut_8_ledpage.yaml                       |  234 ++++++
 libusb/CMakeLists.txt                        |   32 +
 libusb/Makefile-manual                       |    4 -
 libusb/Makefile.am                           |    7 -
 libusb/Makefile.freebsd                      |   13 +-
 libusb/Makefile.haiku                        |   39 -
 libusb/Makefile.linux                        |   15 +-
 libusb/hid.c                                 |  210 +++---
 linux/CMakeLists.txt                         |   20 +
 linux/Makefile-manual                        |   17 +-
 linux/README.txt                             |   63 ++
 linux/hid.c                                  |  215 +++---
 mac/CMakeLists.txt                           |   11 +
 mac/Makefile-manual                          |   13 +-
 mac/hid.c                                    |  950 ++++++++++++++----------
 testgui/copy_to_bundle.sh                    |    3 +-
 testgui/mac_support.cpp                      |  134 ++++
 testgui/mac_support_cocoa.m                  |   13 +-
 testgui/start.sh                             |    2 +
 udev/99-hid.rules                            |    7 +-
 windows/.gitignore                           |    2 -
 windows/CMakeLists.txt                       |    5 +
 windows/Makefile.mingw                       |   11 +-
 windows/ddk_build/hidapi.def                 |    1 -
 windows/hid.c                                |  317 ++------
 windows/hidapi.sln                           |   26 +-
 windows/hidapi.vcxproj                       |  160 ----
 windows/hidapi.vcxproj.filters               |   27 -
 windows/hidsdi.h                             |   88 +++
 windows/hidtest.vcproj                       |    6 +-
 windows/hidtest.vcxproj                      |  156 ----
 windows/hidtest.vcxproj.filters              |   22 -
 86 files changed, 9989 insertions(+), 3908 deletions(-)
```

</details>

i got a request awhile ago to consider merging our changes "upstream" (i am going to refer to your library as being upstream now since it clearly gets more traffic than ours and was also endorsed by signal11 maintainers). i initially declined because at the time, i wasn't sure of the benefit, but now i'm reconsidering that since this repo does seem to be getting sustained activity and also probably has several bugfixes that we don't.

is there any possibility of this happening?

from what i understand, the major changes can be summarized as:
- large changes to some code files (this would require careful review, i believe we may both have some bug fixes and features, although the core API appears to be mostly unchanged)
- your fork does not have a CMake build system (unacceptable for our use, although i see you have #12 in progress)
- our fork has some extra components such as hidapi2osc and hidapi_parser (we can pull these into our larger project easily)

a closer inspection should be done for any licensing conflicts on files to be merged to your fork, for instance i have just noticed that although the LICENSE.txt file in our repo still claims the project can be used under several licenses, certain files like `hidapi_parser/hidapi_parser.h` include a GPL3 header.

i should also ping @sensestage who did most of the work by which our fork differs from yours, but who is no longer actively maintaining SuperCollider. they may be able to provide some more info.

Youw, 2020-06-30:
> is there any possibility of this happening?

I didn't have a chance to take a really close look at the changes in your stream, but as far as I see right now, there're a lot of project-related changes, which aren't needed for most users of hidapi.

So if you ask me (other maintainers/contributors may have a different opinion), either we should (at least try) to merge all changes from your upstream to libusb/hidapi - I think we shouldn't. I'm trying to follow a simple ideology, started by the original author: this a simple C library, that does one thing: provides access to HID devices, and do it identically on all supported platforms. At this point it is a (more or less) mature implementation, proven to be working for years, and I intend to keep it that way. To my experience, the easiest way to do it, it to keep changes as small as possible, specially, where everything is works as expected.

Although, I'm not claiming that current implementation is perfect or has no bugs. If you can point in a specific place of code and show that there is a bug (e.g. discovered when you project was developed), or even better - prepare a PR, I think we'll gladly consider merging small fixes.

As for API changes (e.g. new function, etc.), we don't have a good strategy for that right now. As a user of this library (not only a maintainer) I think the first function we're missing - `hid_version`.

As for CMake - I'm personally a big fan of CMake as a build system for C/C++ projects, I'm even trying to get a decent implementation on my own: https://github.com/libusb/hidapi/tree/cmake-support. My goals so far:
- completely repeat Autotools implementation in terms of functionality;
- use modern CMake (code less, more simplicity/flexibility/etc.);
- make it super-easy to use for CMake users;

todbot, 2020-06-30:
Hi, I've looked at the supercollider version of hidapi over the years and there's a lot of good stuff in there, but like @Youw I feel there's a a bunch of separate projects in that repo that should be their own.

As you mention,  the `hidapi_parser` and `hidapi2osc` directories should be their own repos. (Though I could see `hidapi_parser` being very useful to `hidapi` users: one of the biggest issues they have is determining the kind of reports to send/receive) 

Beyond that,  it seems there is not that much difference on a per-platform basis from signal11's repo.  (i.e. cursory inspection of `mac/hid.c`, `windows/hid.c`, and `linux/hid.c` haven't diverged much from signal11)

Would it be possible for you to create a new `hidapi` repo or branch that contains only the per-platform differences from `libusb/hidapi`?

brianlheim, 2020-06-30:
@Youw i think we are basically suggesting the same thing :) perhaps i could have phrased it better.

i'll look and see what bugfixes and features in our fork might be of value given this repo's goals, and start thinking long-term how we could switch our project to use this fork.

> Would it be possible for you to create a new hidapi repo or branch that contains only the per-platform differences from libusb/hidapi? That would make it much easier to do PRs to this repo.

sure, that can be part of my review of our code.

> As you mention, the hidapi_parser and hidapi2osc directories should be their own repos. (Though I could see hidapi_parser being very useful to hidapi users: one of the biggest issues they have is determining the kind of reports to send/receive)

i was planning to move these into our main repo at github.com/supercollider/supercollider rather than into separate small repos, as that is easiest for us to maintain. if you believe there is concrete benefit to having hidapi_parser as a separate repo, and the GPL3+ licensing does not present a problem to you, i could definitely set it up as a separate repo. it would consist of those 3 C files, CMakeLists.txt, readme, and a GPL3+ license. does that sound OK?

> use modern CMake (code less, more simplicity/flexibility/etc.);

would keeping the minimum CMake version to <= 3.5 be a problem? i see you already have 3.1 as the minimum in your branch.

when do yall plan on having CMake support available (roughly)?

Youw, 2020-06-30:
> i'll look and see what bugfixes and features in our fork might be of value given this repo's goals, and start thinking long-term how we could switch our project to use this fork.

Awesome, sounds like a plan.

> would keeping the minimum CMake version to <= 3.5 be a problem? i see you already have 3.1 as the minimum in your branch.

So far I've tried to stick with lowest version possible. So far [windows](https://github.com/libusb/hidapi/blob/cmake-support/windows/CMakeLists.txt) backend has no extra requirements, [mac](https://github.com/libusb/hidapi/blob/cmake-support/mac/CMakeLists.txt) needs _3.4.3_ , but [libusb](https://github.com/libusb/hidapi/blob/cmake-support/libusb/CMakeLists.txt) and [hidraw](https://github.com/libusb/hidapi/blob/cmake-support/linux/CMakeLists.txt) are tight to _3.6.3_, where `pkg_check_modules` supports `IMPORTED_TARGET` options, which comes extremely handy.

So far I was going to leave it that way. Can't say that it is not possible to workaround without `IMPORTED_TARGET` and lower the version, but with it the implementation becomes simple/elegant/modern/you_name_it.

> when do yall plan on having CMake support available (roughly)?

It's not like I have a plan or dedicated time for the project. So as soon as I get my hands to it... Hopefully weeks.

brianlheim, 2020-06-30:
sounds good! 3.6 would probably be fine for us.

since it sounds like you're not on any particular schedule, I'll check back in a month or two to see where CMake support is, since not having that would block us from using this fork. for now my question's been answered so feel free to close this ticket if you'd like. :) thanks for your time!

Youw, 2020-07-02:
Awesome

closed 2020-07-02 by Youw