libusb/hidapi #176: hidapi installation support

ashishsiddharth, 2020-07-01:
hi.. 
I am getting this error? what should i do

>>> import hid
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Anaconda3\lib\site-packages\hid\__init__.py", line 30, in <module>
    raise ImportError(error)
ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll

Youw, 2020-07-02:
At pip [package page](https://pypi.org/project/hid/) you may find a link to the package source [home page](https://github.com/apmorton/pyhidapi) where the first chapter of README is pretty much clear: pyhidapi is dependant upon the hidapi library, _which must be installed separately_.

Since you're on Windows, you need _hidapi.**dll**_ (exactly as described in the error message) which is provided at [releases page](https://github.com/libusb/hidapi/releases).

closed 2020-07-02 by Youw

ashishsiddharth, 2020-07-03:
I already did those. Build hidapi using .sln file as instructed. and also have downloaded the precompiled files. 

In which place or folder should i place those file i.e. hidapi.dll?

Youw, 2020-07-03:
Apparently somewhere, where python can find them.

Please note, that this is an issue tracker, not a support forum. Your question is not directly related to hidapi, but rather to the usage of `hid` pip package on Windows.
Stackoverflow is more appropriate place for such questions.

DesperateForHelpIntern, 2021-08-09:
I am also experiencing the same issue, but with a different context.  I've installed PYHIDAPI and HIDAPI through pip (pip install hid, pip install pyhidapi)and checked to see they were being installed to the python version being used by my API (C:\Users\user\AppData\Local\Microsoft\WindowsApps).  Still, when running the program, the line "import hid" throws the given error: 
ImportError: Unable to load any of the following libraries:libhidapi-hidraw.so libhidapi-hidraw.so.0 libhidapi-libusb.so libhidapi-libusb.so.0 libhidapi-iohidmanager.so libhidapi-iohidmanager.so.0 libhidapi.dylib hidapi.dll libhidapi-0.dll.  I've checked stack overflow for a solution to this as well: https://stackoverflow.com/questions/64753345/how-to-install-hidapi-on-windows-with-cygwin has no solutions posted, but it links the following stack overflow with solutions: https://stackoverflow.com/questions/62620247/how-to-determine-where-to-install-hidapi-dll.  I was not able to implement the solutions given because when I downloaded the repository, it did not contain a .dll or a .lib file to place in the System32 file.  I was also not able to to implement the 2nd solution provided. For some reason, when I was trying to implement the 2nd solution, the system could not recognize that any of the file names existed.  

Anwhoo, it would be preferable if I can solve this by Friday because that's when my internship ends.  I will save the link to this post for the next person who needs to deal with this problem if I can't solve it by then.