libusb/hidapi #183: hidapi doesn't work no more with Python 3

nbenm, 2020-08-13:
I have a Python script that was working well.
Since last upgrade homebrew and/or Python3, it stopped working.
I have the following error:
```
Traceback (most recent call last):
  File "./microlife.py", line 221, in <module>
    en = Enumeration()
  File "/Users/nb/Dropbox/dev/python/sl4a/scripts/apps/diabete/tension/easyhid.py", line 366, in __init__
    self.device_list = _hid_enumerate(vid, pid)
  File "/Users/nb/Dropbox/dev/python/sl4a/scripts/apps/diabete/tension/easyhid.py", line 428, in _hid_enumerate
    start = hidapi.hid_enumerate(vendor_id, product_id)
  File "/usr/local/lib/python3.8/site-packages/cffi/api.py", line 912, in __getattr__
    make_accessor(name)
  File "/usr/local/lib/python3.8/site-packages/cffi/api.py", line 908, in make_accessor
    accessors[name](name)
  File "/usr/local/lib/python3.8/site-packages/cffi/api.py", line 838, in accessor_function
    value = backendlib.load_function(BType, name)
AttributeError: function/symbol 'hid_enumerate' not found in library '<None>': dlsym(RTLD_DEFAULT, hid_enumerate): symbol not found
```
I can reproduce this using:

```
from easyhid import Enumeration
en=Enumeration()
```
What can I do to solve my problem?
I have read that this happens when hidapi is not installed.
But it is installed (hidapi 0.9.0), and it is built from `https://github.com/libusb/hidapi/archive/hidapi-0.9.0.tar.gz`
I'm on MacOS Catalina, and hidapi is installed from homebrew.
easyhid (version 0.0.10) is installed with pip3.

Thanks for any help.

Youw, 2020-08-13:
What do you get from following?
```python
import cffi
hidapi = ffi.dlopen('hidapi')
```

```python
import cffi
hidapi = ffi.dlopen(ctypes.util.find_library('hidapi'))
```

Youw, 2020-08-13:
Actually, I'm ready to close this issue as invalid
There is no issues with hidapi - it does work
As for python packages that uses it - it is up to them / their users to make sure they work / have proper environment

this project/repository has nothing to do with `easyhid` - it is a separate project

nbenm, 2020-08-13:
@Youw 
```
>>> import cffi
>>> hidapi = ffi.dlopen('hidapi')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ffi' is not defined
```
```
>>> import cffi
>>> hidapi = ffi.dlopen(ctypes.util.find_library('hidapi'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ffi' is not defined
```

Youw, 2020-08-13:
Try updated

nbenm, 2020-08-13:
```
>>> import cffi
>>> ffi = cffi.FFI()
>>> hidapi = ffi.dlopen('hidapi')
>>> 
```

```
>>> import cffi
>>> ffi = cffi.FFI()
>>> hidapi = ffi.dlopen(ctypes.util.find_library('hidapi'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'ctypes' is not defined
>>> 
```

Youw, 2020-08-13:
```
>>> import cffi
>>> ffi = cffi.FFI()
>>> hidapi = ffi.dlopen('hidapi')
>>> 
```
and you're saying that in the same console, the code above executes without errors, and the code below:
```python
from easyhid import Enumeration
en=Enumeration()
```
fails?

nbenm, 2020-08-13:
If I type all code in the same console, it works:
```
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import cffi
>>> ffi = cffi.FFI()
>>> hidapi = ffi.dlopen('hidapi')
>>> 
>>> from easyhid import Enumeration
>>> en=Enumeration()
>>>
``` 

But if I type only the second...

```
Python 3.8.5 (default, Jul 21 2020, 10:48:26) 
[Clang 11.0.3 (clang-1103.0.32.62)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> from easyhid import Enumeration
>>> en=Enumeration()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/site-packages/easyhid/easyhid.py", line 366, in __init__
    self.device_list = _hid_enumerate(vid, pid)
  File "/usr/local/lib/python3.8/site-packages/easyhid/easyhid.py", line 428, in _hid_enumerate
    start = hidapi.hid_enumerate(vendor_id, product_id)
  File "/usr/local/lib/python3.8/site-packages/cffi/api.py", line 912, in __getattr__
    make_accessor(name)
  File "/usr/local/lib/python3.8/site-packages/cffi/api.py", line 908, in make_accessor
    accessors[name](name)
  File "/usr/local/lib/python3.8/site-packages/cffi/api.py", line 838, in accessor_function
    value = backendlib.load_function(BType, name)
AttributeError: function/symbol 'hid_enumerate' not found in library '<None>': dlsym(RTLD_DEFAULT, hid_enumerate): symbol not found
>>> 
```

In my code, I have only the second part.

Youw, 2020-08-13:
maybe you need to update `easyhid` or smth
in any case - this is not a hidapi library issue, and this is a bug tracker, not a support forum

closed 2020-08-13 by Youw

nbenm, 2020-08-13:
I did update easyhid with no success.
Anyway, thanks for your help.

nbenm, 2020-08-13:
@Youw I wanted to let you know, the issue is solved.
It was an easyhid one.
The author was checking if platform.platform() was containing "Darwin".
But the output of this function has changed. It is now "macOS-...."
The code you asked me to try helped me very much.