libusb/hidapi #184: Compiling a dynamic library (.dylib) on MacOS?

pierrelavoie, 2020-08-25:
A newbie question probably: How can I compile a ".dylib" library on Mac (Catalina)?

I have cloned hidapi 0.9.0, installed command line tools, Xcode, homebrew, and autotools on Catalina. I ran ./bootstrap, ./configure, make, and "sudo make install", which produces "hid.o" and "hid.lo" files.  Adding the --enable-shared option to configure changes nothing.  My goal is to call hidapi as a library from Matlab which needs the ".dylib" file type.
Thank you!

z3ntu, 2020-08-25:
As far as I can tell, a .dylib is built by our CI: https://travis-ci.org/github/libusb/hidapi/builds/717648513#L249 so I'm not sure why it's not working for you.

pierrelavoie, 2020-08-25:
As it turns out it was working for me too.  I was looking in my hidapi/ directory instead of in /use/local/library for the library file. My bad.  Thank you!

closed 2020-08-25 by pierrelavoie