libusb/hidapi #185: New release

mcuee, 2020-09-10:
Just wondering if it is already a good time for a new release.

Youw, 2020-09-10:
some time ago I figured possible scope for new release:
https://github.com/libusb/hidapi/projects/1

vitalys, 2020-09-26:
@Youw some time ago you wanted to release v0.9.1 that fixes #60 
So what's the current plan? Are we going to release something before v0.10.0?

Youw, 2020-09-26:
I don't see point having anything before 0.10
According to current plan: https://github.com/libusb/hidapi/projects/1 - everything is in review, so no development/fixes is expected before 0.10.
Waiting for feedback regarding #189 from @vozhyk-com, and we good to go with 0.10.0

Youw, 2020-10-14:
#198 will be last commit into master before firing v0.10.0
Since a bunch of commits got into master today, I think it worth having master for few more days as is, just to give at least some time to community to check its stability

if no new issues (introduced by latest commits) are going to be reported for this few days - I'll fire v0.10.0 over the weekend

Youw, 2020-10-25:
https://github.com/libusb/hidapi/releases/tag/hidapi-0.10.0

closed 2020-10-25 by Youw