libusb/hidapi #187: i want to run test-gui in windows,but it doesnot compile successfully

ShiverZm, 2020-09-24:
i read readme carefully, i see maybe i need hidapi-externals,but i dont know where it is ,could you mind tell me this?

Youw, 2020-09-24:
The README is outdated. The best course of action, in case you need GUI - build fox toolking by yourself first

bevanweiss, 2020-09-26:
It seems that building the Fox Toolkit on a modern MSVC compiler is troublesome also.
Is there an alternative recommended test GUI for hidapi (under Windows)?

Youw, 2020-09-26:
No, unfortunately there is none.
Bu there is a console test tool you may try: https://github.com/todbot/hidapitester

bevanweiss, 2020-09-26:
Thanks, and it does appear that 'todbot' also has a Python GUI tester
https://github.com/todbot/hidpytoy

Which will probably work for me right now.

mcuee, 2020-12-04:
Haha, managed to cause kernel panic using hidpytoy for my new Mac Mini M1, when connecting to a Microchip PICkit 2 programmer and playing with hidpytoy.

mcuee, 2020-12-04:
https://github.com/libusb/hidapi/issues/68
Last time I tried to build test-gui under MSYS2 but it is not successful. I will try it again later.