libusb/hidapi #188: Linux cannot open device but Windows can

Jihadist, 2020-09-24:
Hello, i try to open my device with hidapi-hidraw through hid_open and hid_open_path on linux but it doesnt work, i receive always null. But it works with hidapi.dll on windows. I try hidapi-libusb but it doesnt work too.
My device looks like this (test app hidraw)

> Device Found
>   type: c251 3505
>   path: /dev/hidraw3
>   serial_number: 000A20000000
>   Manufacturer: Keil Software
>   Product:      Keil USB Device 0
>   Release:      100
>   Interface:    0
>   Usage (page): 0x0 (0x0)

libusb test app

> 
> Device Found
>   type: 0458 6001
>   path: 0001:0002:02
>   serial_number: (null)
>   Manufacturer: (null)
>   Product:      (null)
>   Release:      1001
>   Interface:    2
>   Usage (page): 0x0 (0x0)

Also i tried to connect with test gui app but it doesnt work too

Youw, 2020-09-24:
To use libusb backend -run the app with sudo
I don't see reasons why wouldn't be able to open a device

Jihadist, 2020-09-25:
> 
> 
> To use libusb backend -run the app with sudo
> I don't see reasons why wouldn't be able to open a device

I didnt know libusb needs sudo

closed 2020-09-25 by Jihadist